KinDR007/VictronMPPT-ESPHOME #1: Please add load current

LuckyTriple7, 2021-05-30:
Hello,

Great work, I will test this next week.
Can you please add load current sensor.
Thx in advance

KinDR007, 2021-05-30:
what do you mean ?  mppt charger does not have this option

Hello,

>
> Great work, I will test this next week.
> Can you please add load current sensor.
> Thx in advance
-- 
*Jiří Bětuňák*

Mobil / WhatsApp                  :  +420 603 73 22 48

______________________________________________________________________________________________________________________________________________________

*Tento mail může obsahovat DŮVĚRNÉ INFORMACE. Jestliže nejste příjemce
nejste oprávnění použít informace obsažené v tomto e- mailu. Prosím
informujte nás ihned že jste ho dostali omylem.*

*This mail may contain CONFIDENTIAL INFORMATION. If you are not the
addresseed you are not authorized to make use of the information contained
in this mail. Please inform us immediately that you have received it by
mistake.*

*Dieses Mail kann VERTRAULICHE INFORMATIONEN enthalten. Wenn Sie nicht der
Adressat sind, sind Sie nicht zur Verwendung der in diesem Mail enthaltenen
Informationen befugt. Bitte benachrichtigen Sie uns sofort über den
irrtümlichen Empfang.*

LuckyTriple7, 2021-05-30:
Hello,
thanks for the fast Response, i mean this:
 
[image]

I changed the Arduino code and it works.

[image]

KinDR007, 2021-05-30:
This is 10Amp output ?

LuckyTriple7, 2021-05-30:
Hello,
it's the same formula as Battery Current:

[image]

KinDR007, 2021-05-30:
Added ("IL" Load current) please test it ... my mppt charger not have this option

LuckyTriple7, 2021-05-31:
Hello,

great Job, works perfect:

[image]

Maybe there is a way to optimize the text sensors:

CHARGER_TEXT
ERROR_TEXT
TRACKER_TEXT
FW_VERSION
PID

The data rarely changes, but is sent every second.
Maybe there is the possibility to send the data only when the value changes.

closed 2021-05-31 by LuckyTriple7

KinDR007, 2021-06-02:
Now i testing better option
[image]