KinDR007/VictronMPPT-ESPHOME #2: Question for monitor a second victron input

kead, 2021-06-12:
Congratulations and thank you for this fantastic integration!! 
I passed the code on esp32 (it has more power) and I would like to use the same to monitor a Victron BMV-712 Battery monitor as well.
Is it possible in your opinion?

KinDR007, 2021-06-12:
Hi for BMV use my other component "SmartShunt esphome " btw bmv and smartshunt has a same properties 

https://github.com/KinDR007/VictronSmartShunt-ESPHOME

kead, 2021-06-12:
i saw the other project, but i didn't understand if they can be used together in the same esp32 (or esp8266)... 
Thank you

KinDR007, 2021-06-12:
Hi,

theoretically they could work on one esp .. but .. mppt and bmv (including smartshunt) have some properties named the same which will cause a problem .. therefore i recommend to use one esp for each victron device

closed 2021-06-12 by KinDR007