KinDR007/VictronMPPT-ESPHOME #5: load current doesnt match battery current during night

danrat94, 2021-08-23:
hi,
shouldnt be the load current be the same as battery current once its night?
it seems like load current has only 1 digit accuracy when bat current got 2. (and my config is set to 2 accuracy overall)
i already checked the ve direct docs but they dont mention the accuracy.

for example:
load current: 0,40a
battery current: 0,45a

any ideas? Or is it just limited by the protocol?
my device is a smartsolar 75/15.

greetings

KinDR007, 2021-08-24:
Hi, mppt returns only one current value, how do you read the second value, or is it the consumption from the battery via the terminals on mppt?  then I would say the difference will be the consumption of the mppt itself

danrat94, 2021-08-24:
Actually it returns battery current AND load current as you can see on page 6 of the ve direct protocol description.
The names are "Main or channel 1 battery current" and "Load current". Right now im using mostly your example.yaml.
i also thought its the victron consumption but they state that its really really low (about 20mA) and not 50+.
Also, i tested it again and even if battery current is about 0,89a then load current will be only 0,8.
so i think its gonna be cut off somewhere or even low precision (reported) by the mppt itself.
i already tried adding more accuracy decimals, doesnt change a thing, just adds more zeros.

KinDR007, 2021-08-24:
Hi, those values ​​are reported by mppt so we don't do anything about it, the component only reads them

closed 2021-08-24 by KinDR007