KinDR007/VictronMPPT-ESPHOME #6: Mppt 100/20 not work

lucasimons, 2021-08-28:
Hi,
i tried to use the custom component with espHome but i have this warnig
[image]
[image]

syssi, 2021-08-29:
Did you try to swap the RX/TX?

lucasimons, 2021-08-29:
Should I reverse the cable on the mppt controller?

syssi, 2021-08-29:
D7 must be connected to RXI. Don't connect TXO because the communication is read-only / receiving-only.

lucasimons, 2021-08-29:
yes but also in the photo he connected the first and third pin which are gnd and tx then obviously the d7 sara rx that is, it reads the data sent by the victron

syssi, 2021-08-29:
I assume the photo is mirrored compared to your picture because the red wire is connected to GND and is the most right position at the connector. Do you own a multimeter to messer the voltage between the outer pins? So you could make sure which one is GND.

lucasimons, 2021-08-29:
yes I checked and the ground in my controller is on the left I do not wish that something had broken because initially I had connected in the wrong position that is gnt to 5v and d7 to rx .... it seems that the connection is unstable I would not have broken something

syssi, 2021-08-29:
Do you see/receive any sensor values of the solar charger? Your debug output contains `uptime` and `wifi quality` measurements only. It looks like you are unable to receive any data of the charger.

lucasimons, 2021-08-29:
i don t  see any solar sensor only the wifi streght
[image]

syssi, 2021-08-29:
Could you post your configuration yaml (without secrets)? You could increase the log level of the uart to monitor the traffic between your ESP and the victron:

```
logger:
  baud_rate: 0
  level: VERY_VERBOSE
  logs:
    uart: VERY_VERBOSE 
    uart_esp8266: VERY_VERBOSE 
    api.service: WARN
    ota: WARN
    sensor: DEBUG
    victron: DEBUG
```

lucasimons, 2021-08-30:
after doing some tests I was able to make it work, now I simply have to fix the stability because I notice constant disconnections to the home assistant

syssi, 2021-08-30:
Could you share your solution/mistakes/findings?

KinDR007, 2021-08-30:
And this wiring not works ?
[](https://community.home-assistant.io/t/victron-energy-vedirect-bluetooth/20726/66?u=kindr)

KinDR007, 2021-08-30:
> D7 must be connected to RXI. Don't connect TXO because the communication is read-only / receiving-only. Take a look at this photo:
> 
> 
> 
> https://community.home-assistant.io/t/victron-energy-vedirect-bluetooth/20726/43?u=syssi

D7 = rx pin on wemos 
Ve.direct tx pin connect to D7 pin its ok

KinDR007, 2021-08-30:
> after doing some tests I was able to make it work, now I simply have to fix the stability because I notice constant disconnections to the home assistant

hi how do you solve the power supply?  it keeps me going and the connection doesn't fall
[image]

lucasimons, 2021-08-30:
I did not make substantial changes simply I checked the connections and it worked then I took a d1 mini and used it to control a relay and an ac dimmer and I also connected the victron to pin d1 I don't know if this creates problems, but I don't think so and now I have also changed the power supply but it seems to disconnect anyway

esphome:
  name: fotovoltaico
  platform: ESP8266
  board: d1_mini
  includes:
      - custom_components/victron/victron.h
substitutions:
    lower_devicename: "victron"
    devicename: "SmartSolar_100_20" 
    config_version: "v2021.05.30"
    wifi_fast_connect: "false"
   # wifi_reboot_timeout: 0s
    accuracy: "2"

wifi:
  ssid: 
  password: 
  power_save_mode: none
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Fotovoltaico Fallback Hotspot"
    password: "R8aNhGpg7EA3"

captive_portal:

# Enable logging
logger:
  baud_rate: 0
  level: VERY_VERBOSE
  logs:
    uart: VERY_VERBOSE 
    uart_esp8266: VERY_VERBOSE 
    api.service: WARN
    ota: WARN
    sensor: DEBUG
    victron: DEBUG

# Enable Home Assistant API
api:

ota:

uart:

- id: uart_bus
  tx_pin: D4 # not connected !! 
  rx_pin: D1 # use olny this pin and GND to MPPT chrager
  baud_rate: 19200 
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_buffer_size: 256
  
victron:
  uart_id: uart_bus  

sensor:  

  - platform: victron
    panel_voltage:
      name: "Pannello Tensione" 
      id: pv
    battery_voltage:
      name: "Batteria Tensione"  
      id: bv
    battery_current:
      name: "Batteria Corrente" 
      id: bc
    load_current:
      name: "Corrente carichi DC" 
      id: load_current
    max_power_yesterday:  
      name: "Massima potenza Ieri" 
      id: max_power_yesterday
    max_power_today:  
      name: "Massima potenza Oggi" 
      id: max_power_today
      accuracy_decimals: ${accuracy}
    yield_total:
      name: "Energia totale pannelli"   
      id: yield_total
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
      unit_of_measurement: kWh
      accuracy_decimals: ${accuracy}
    yield_yesterday:
      name: "Energia pannelli Ieri "  
      id: yield_yesterday
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
      unit_of_measurement: kWh
      accuracy_decimals: ${accuracy}
    yield_today:
      name: "Energia pannelli Oggi"  
      id: yt
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
      unit_of_measurement: kWh
      accuracy_decimals: ${accuracy}
    panel_power:
      name: "Potenza Pannelli"  
      id: panel_power
    day_number:
      name: "Day sequence number"  
      id: day_number
      icon: "mdi:calendar-today"
    charger_text:
      name: "Charger state"
      id: ct
    error_text:
      name: "Error code"  
      id: et
    tracker_text:
      name: "Tracker operation"  
      id: tt
    fw_version:
      name: "Fw version"  
      id: fw
    pid:
      name: "PID"  
      id: pid  
 

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: D7
    zero_cross_pin:
      number: D8
      mode: INPUT
      inverted: yes

light:
  - platform: monochromatic
    output: dimmer1
    name: Resistenza Dimmer  

    
switch:
  - platform: gpio
    pin: D3
    
    name: "Inverter"      # Modificare nome a piacimento da esporre in Home Assistant
  - platform: restart
    name: "Auto Restart"
    id: restart_switch
time:
  - platform: homeassistant
    on_time:
      - seconds: 0
        minutes: 50
        hours: 3
        days_of_week: MON-SUN
        then:
           - switch.toggle: restart_switch
    id: homeassistant_time

KinDR007, 2021-08-30:
Pin D3 = flash change to other pin 

[image]

lucasimons, 2021-08-31:
last night I tried to put the relay on pin d4 but about every 1000 seconds I had a disconnection, now I tried with another wemos d1 mini that I used previously for the dimmer and the relay, I will update you if I have solved the problem

lucasimons, 2021-08-31:
I tried to change the power supply and micro USB cable but I always encountered that problem now I used a step down converter 12v to 5v I see if it improves, I update you if I have solved or found something[image]

KinDR007, 2021-09-01:
> I tried to change the power supply and micro USB cable but I always encountered that problem now I used a step down converter 12v to 5v I see if it improves, I update you if I have solved or found something[image]

powered by battery pack connected to mppt?  if so do not connect gnd from stepdown to wemos

lucasimons, 2021-09-01:
I should have solved the problem by changing the wemos d1 mini with a nodemcu esp8266 now 5000 seconds have already passed and I think I have solved it, but I also changed the connections in this way

- id: uart_bus
  tx_pin: D8 # not connected !! 
  rx_pin: GPIO13 # use olny this pin and GND to MPPT chrager
  baud_rate: 19200 
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_buffer_size: 256
  
    
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: GPIO12
    zero_cross_pin:
      number: GPIO14
      mode: INPUT
      inverted: yes

light:
  - platform: monochromatic
    output: dimmer1
    name: Resistenza Dimmer  

switch:
  - platform: gpio
    pin: D1
    
    name: "Inverter"

today i will also try this https://github.com/syssi/esphome-pipsolar made by @syssi  with an easun inverter after i would miss the bms (now i'm using the serial Bluetooth connection i don't know if it is possible with an esp board) to be integrated via esphome instead of the current raspberry and mqtt

syssi, 2021-09-01:
@lucasimons Which BMS do you use?

lucasimons, 2021-09-01:
i have a bms like this  [BMS](https://it.aliexpress.com/item/32997552090.html?spm=a2g0s.9042311.0.0.27424c4dvSMF6Q) has no serial ports or at least i think and i currently use a script in py to extrapolate the data and publish it via mqtt, but I saw on this  [forum](https://diysolarforum.com/threads/for-those-of-you-looking-to-monitor-your-ant-bms-with-pi3-via-bluetooth.6726/page-3) that if you wanted there would also be the possibility of controlling the charge and discharge mosfets but I could not

BTW, I can control the BMS now. The checksum is just the sum of the 3 bytes (address + data + data).

So.... sending it:

'A5A5FA0000FA' # Turn Off Charge MOSFET
'A5A5FA0001FB' # Turn On Charge MOSFET
'A5A5FE0000FE' # Reboot BMS
'A5A5F90000F9' # Turn Off DisCharge MOSFET
'A5A5F90001FA' # Turn On DisCharge MOSFET
'A5A5FC0000FC' # Toggles balancing

syssi, 2021-09-01:
You could try to open the BMS and solder some wires to the RX/TX pins of the BLE daughterboard:

https://user-images.githubusercontent.com/14097148/67513889-6faa1e00-f6a4-11e9-94c1-70b825f00976.png

lucasimons, 2021-09-01:
Currently I have a bms that I have disassembled and opening it actually has this board ble so should I connect these 4 cables?
[image]

syssi, 2021-09-01:
Don't connect VCC because the module is already powered by the BMS. RX, TX and GND is sufficient. Take a look at the bottom of the circuit board (left and right corner). You could try to trace (measure the resistance) between all of this pins and the RX and TX pin of the BLE module. I assume there is a connection. Just the jack is missing.

lucasimons, 2021-09-01:
I checked the continuity of all the connections that seem to be set up but I only found these two,
[image]

 I don't know if it makes sense to continue writing here or directly open a new repository and document everything there

syssi, 2021-09-01:
Yes. We shouldn't hijack this issue and repository. Feel free to ping me if you need support. I'm trying to build a powerwall using a JK BMS at the moment: https://github.com/syssi/esphome-jk-bms

lucasimons, 2021-09-01:
perfect I had seen your repo in fact I had found that of the inverter that I will test shortly then in case I continue with the bms

closed 2021-09-01 by lucasimons