KinDR007/VictronMPPT-ESPHOME #7: Convert custom component to external component

syssi, 2021-08-29:
As soon this component is an external component it can be used with a single instruction at the yaml configuration without the need to clone the component.

syssi, 2021-08-29:
I will care about this issue as soon #8, #9 and #10 is merged. The PRs are the foundation to a use github actions / a little CI pipeline to enforce code style and making sure this project always builds fine at the latest esphome version. I'm talking about something like this:

https://github.com/syssi/esphome-jnge-mppt-controller/actions/runs/1148178140

KinDR007, 2021-08-30:
``
esphome:

...

external_components:
  - source:
      type: git
      url: https://github.com/KinDR007/VictronMPPT-ESPHOME.git
``
Is that what you meant?

closed 2021-08-30 by KinDR007

syssi, 2021-08-30:
Yes!

KinDR007, 2021-08-30:
Done 😉

-- 
*Jiří Bětuňák*

Mobil / WhatsApp                  :  +420 603 73 22 48

______________________________________________________________________________________________________________________________________________________

*Tento mail může obsahovat DŮVĚRNÉ INFORMACE. Jestliže nejste příjemce
nejste oprávnění použít informace obsažené v tomto e- mailu. Prosím
informujte nás ihned že jste ho dostali omylem.*

*This mail may contain CONFIDENTIAL INFORMATION. If you are not the
addresseed you are not authorized to make use of the information contained
in this mail. Please inform us immediately that you have received it by
mistake.*

*Dieses Mail kann VERTRAULICHE INFORMATIONEN enthalten. Wenn Sie nicht der
Adressat sind, sind Sie nicht zur Verwendung der in diesem Mail enthaltenen
Informationen befugt. Bitte benachrichtigen Sie uns sofort über den
irrtümlichen Empfang.*