KinDR007/VictronMPPT-ESPHOME #18: how to slow the polling of the sensor?

gazzaman2k, 2021-09-05:
is there any way to stop it updating every 2 seconds and set it to scan like once every 10seconds? its too frequent and will soon fill up my databases

but great work got it running on a victron mppt 250/70 and esp32-s

KinDR007, 2021-09-05:
> is there any way to stop it updating every 2 seconds and set it to scan like once every 10seconds? its too frequent and will soon fill up my databases
> 
> 
> 
> but great work got it running on a victron mppt 250/70 and esp32-s 

Hi , use function of esphome !!
Add 

udate_interval: 10s 

to sensor 😉

syssi, 2021-09-05:
This isn't correct. The component doesn't use the `pollig_component_schema` / isn't polling the device. The victron device pushs a new message every two seconds. You should use a `filter` to `throttle` the publishing of the measurements. You could also use a `filter` to calculate a `median`: https://esphome.io/components/sensor/index.html

syssi, 2021-09-05:
@gazzaman2k Off-topic: Do you own the smart version (bluetooth) or the standard version of the mppt 250/70?

gazzaman2k, 2021-09-05:
its the smart version with bluetooth

KinDR007, 2021-09-05:
> This isn't correct. The component doesn't use the `pollig_component_schema` / isn't polling the device. The victron device pushs a new message every two seconds. You should use a `filter` to [...]

Hi,
all data counts/calculate victron, median is in my opinion useless, or am I wrong?

KinDR007, 2021-09-05:
Hi ,is it possible to contact you outside of github?  discord or telegram for example?

gazzaman2k, 2021-09-05:
i used 

filters:
    - median:
        send_every: 10
        
Working great thanks, having it update every 2 seconds you could see the mppt calculating its maximum tracking points and causes it to spike on the graph as its doing its adjustment

KinDR007, 2021-09-05:
bmv and smartshunt is supported https://github.com/KinDR007/VictronSmartShunt-ESPHOME

syssi, 2021-09-05:
> @syssi ,Hi
> is it possible to contact you outside of github? discord or telegram for example?

You could contact me by mail: basti at linkt.de

syssi, 2021-09-05:
> all data counts/calculate victron, median is in my opinion useless, or am I wrong?

The median should be used for selected entities only. Most of the time throttling should be the way to go.

closed 2021-09-05 by KinDR007