KinDR007/VictronMPPT-ESPHOME #21: Some errors

lucasimons, 2021-09-16:
Hi I am using this software with a 100/20 l mppt, I have encountered problems with decimals let me explain better, some times for example the battery voltage is 26v but the sensor sends 0.26v, this also happens to the total energy I already have put a filter that averages but is not enough

syssi, 2021-09-16:
Could you try to provide the raw data of your solar charger?

lucasimons, 2021-09-17:
[image]
[image]

KinDR007, 2021-09-17:
@lucasimons 

may i ask how are you wiring ?

lucasimons, 2021-09-18:
I have connected the tx to D7 e gnd to gnd... I would not want my nodemcu to go into crisis with esp8266 because it actually sends a lot of data it is not possible to decrease the refresh rate, as soon as I get it I will use an esp 32 to see if I solve

KinDR007, 2021-09-18:
@lucasimons 

power supply how did you deal with?  external source?  or stepdown from Battery

lucasimons, 2021-09-19:
Yes I have the 24v battery and a victron orion 24v to 12v converter and then connected a quick charge 3.0 module that powers a raspberry and the nodemcu however I think it's like putting a step down 24 to 5

KinDR007, 2021-09-20:
Ok , 
Please disconect gnd from ve.direct port and try 
(
Ve.direct connected only tx
Power supply vcc and gnd from step down 
)

closed 2021-10-04 by KinDR007