KinDR007/VictronMPPT-ESPHOME #22: How to connect two Victron devices

oliverguj, 2021-09-18:
Great work!

I managed to monitor my MPPT (SmartSolar 75/15) with an ESP32 on UART2.

Now I'd like to add my inverter (Victron Phoenix, also a ve.direct device).

Readme says "The uart_id and victron_id is optional if you use a single UART / victron device."

But how can I configure a second device on a second UART?

syssi, 2021-09-18:
```
uart:
  - id: uart0
    baud_rate: 9600
    tx_pin: GPIO1
    rx_pin: GPIO3
  - id: uart1
    baud_rate: 9600
    tx_pin: GPIO14
    rx_pin: GPIO4
  - id: uart2
    baud_rate: 9600
    tx_pin: GPIO32
    rx_pin: GPIO33

victron:
  - id: victron0
    uart_id: uart0
  - id: victron1
    uart_id: uart1
  - id: victron2
    uart_id: uart2

sensor:
  - platform: victron
    victron_id: victron0
    panel_voltage:
      name: "victron0 panel voltage"
  - platform: victron
    victron_id: victron1
    panel_voltage:
      name: "victron1 panel voltage"
  - platform: victron
    victron_id: victron2
    panel_voltage:
      name: "victron2 panel voltage"
```

It's untested. Please give it a try!

oli101010, 2021-09-22:
Works perfectly, thanks!

oliverguj, 2021-09-22:
Works perfectly, thanks!

closed 2021-09-22 by oliverguj