KinDR007/VictronMPPT-ESPHOME #26: victron mppt 100/50?

drawlusyk, 2021-10-10:
Is support for the 100/50 available? Is there anything i can provide to help you make this work with this version? I have configured as outlined in the docs but receive nan for all items.

KinDR007, 2021-10-10:
hello please describe in detail the connection including power supply for esp

syssi, 2021-10-10:
Please make sure the `logger` doesn't write to the same GPIOs (by default).

drawlusyk, 2021-10-10:
I tried commenting out everything under logger:
I tried UART1 also.
Both do not work.
I checked victron app for tx function and it is set to normal communcation.
Anyway to see if anything is coming out of the tx port of 100/50 into rx port on d1mini?

KinDR007, 2021-10-10:
@drawlusyk please describe the connection between wemos and mppt and how to power the wemos

KinDR007, 2021-10-13:
Please 24v in is the battery connected to mppt?  if so, disconnect gnd from the ve.direct connector

KinDR007, 2021-10-13:
> [image]
> 
> 

if gnd cables are wemos connected?  so is disconnection!  connect ground only with stepdown and with ve.direct port take tx pin only

KinDR007, 2021-10-14:
@drawlusyk any progress?

KinDR007, 2021-10-15:
@drawlusyk it's weird, i run both mppt and smartshunt on wemos d1 and it runs as it should so .probably the wrong piece in your case