KinDR007/VictronMPPT-ESPHOME #29: Serial connection crashing esp32

drawlusyk, 2021-12-27:
Hello all,

I have found that both mppt and shunt connections to the esp32 restart the esp32 between 2 and 5 mins going. 
I am using 2 separate esp32. one for mppt and one for shunt. 
It is configured, i get all data out. I am only using rx and gnd on the esp32. I have tried different 4 different esp32 devices all restart.
I configured the device to show the uptime. As soon as i unplug the connector from controller and/or shunt, the devices does not restart.

Any thoughts on what this could be?

Thanks for any input.

logger:
  baud_rate: 0

uart:
  id: uart_bus
  #tx_pin: GPIO01  #D8 Not connected! The communication is read-only
  rx_pin: GPIO3  #D7 Connect this this GPIO and GND to the MPPT charger
  baud_rate: 19200
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_buffer_size: 256
  
victron:
  id: victron0
  uart_id: uart_bus

drawlusyk, 2021-12-27:
[09:55:46][D][uart_debug:158]: <<< "\x88\x98\xA4\b\xC4\xCF\x1FJ\b\xC4\xCE\x1F\x88\x98\x10\x11\b\xEE\xA4\x8C\x04\x10JJJJ\f\xCE\xA4\f\f\x1D\f\f\x10DH\x88\xC4L\x1A:\x88Z\xBC\xC4L\x1AJ\f\x94\x04\b\xC4\xDC\xDD\xAC\xB5\x8C\xBC9!\xDC"
[09:55:46][W][victron:044]: Last transmission too long ago.
[09:55:47][D][uart_debug:158]: <<< "\f\xD6\xA4\xA6\x8A\xA8\xAC\x8C\xD8\xF3\x84\xA7\x8A\x06\xBE>\xE8\xA4\n"
[09:55:47][D][uart_debug:158]: <<< "\x88\x98\xA4\b\xC4\xCF\x1FzJ:\xC8\xEE\xBC\xC4\b\x14\f\x00\x1F:\xB8\x1E\xCAJJJJ\f\xCE\xA4\f\f\x1D\f\f\x10DH\x88\xC4L\x1A:\x88Z\xBC\xC4L\x1AJ\f\x94\x04\b\xC4\xDC\xDD\xAC\xB5\x8C\xBC9!\xD4"
[09:55:47][W][victron:044]: Last transmission too long ago.
[09:55:48][D][uart_debug:158]: <<< "\f\xD6\xA4\xA6\x8A\xA8\xAC\x8C\xD8\xF3\x84\xA7\x8A\x06\xBE>\xE8\xA4\n"
[09:55:48][D][uart_debug:158]: <<< "\x88\x98\xA4\xA5\xC4\xCF\x1Fzz:\xC8\xEE\xBC\xC4\b\x14\f\x00\x1F:\xB8\x1E\xCAJJJJ\f\xCE\xA4\f\f\x1D\f\f\x10DH\x88\xC4L\x1A:\x88Z\xBC\xC4L\x1AJ\f\x94\x04\b\xC4\xDC\xDD\xAC\xB5\x8C\xBC9!\xC4"
[09:55:48][W][victron:044]: Last transmission too long ago.

KinDR007, 2021-12-27:
Hi, how do you resolve the power supply for esp32? external source or stepdown from the battery?

drawlusyk, 2021-12-27:
Hi. Power is a separate power supply. Step down buck to 5v. 3amp.

KinDR007, 2021-12-27:
OK ... so disconnect GND from the Ve.Direct port and leave only 1 pin connected for communication

drawlusyk, 2021-12-27:
still restarts at 154s

KinDR007, 2021-12-27:
"ve.direct" is set for communication only? in mppt or smart shunt (switch or menu)

KinDR007, 2021-12-30:
> still restarts at 154s

Please send me your sketch

drawlusyk, 2021-12-31:
I have left for the last few days. When i had the web server of the device active, and web page open to the esp32, it seemed to keep restarting. I closed these tabs and used home assistant to get the uptime. It has been up for a couple days. SO, i can only assume that having the esp32 web server open on a browser causes some type of problem and restarts the esp32. 
Anyway looks like things are working ok now. Thanks for all the help!!!

syssi, 2021-12-31:
Please try to avoid the web server. It's the most resource expensive component of esphome.

closed 2022-01-01 by drawlusyk