KinDR007/VictronMPPT-ESPHOME #31: Not getting data from MPPT

CountParadox, 2022-01-24:
Hello! 

Not sure if anything has changed in more recent firmwares? 

I am not getting any data from my 75/15.

[image]

Very simple config;

[image]

connected TX pin of mppt to D7 on the D1 mini, and GND to GND pin.

ESP is powered via USB while testing from a wall brick.

Logs do not appear to show anything helpful;

`INFO Reading configuration /config/esphome/vedirect.yaml...
INFO Starting log output from vedirect.local using esphome API
INFO Successfully connected to vedirect.local
[18:33:56][I][app:102]: ESPHome version 2022.1.1 compiled on Jan 24 2022, 18:02:06
[18:33:56][C][wifi:488]: WiFi:
[18:33:56][C][wifi:350]:   Local MAC: 30:xx.xx.xx.xx:CA
[18:33:56][C][wifi:351]:   SSID: [redacted]
[18:33:56][C][wifi:352]:   IP Address: 172.16.0.62
[18:33:56][C][wifi:354]:   BSSID: [redacted]
[18:33:56][C][wifi:355]:   Hostname: 'vedirect'
[18:33:56][C][wifi:357]:   Signal strength: -66 dB ▂▄▆█
[18:33:56][C][wifi:361]:   Channel: 1
[18:33:56][C][wifi:362]:   Subnet: 255.255.255.0
[18:33:56][C][wifi:363]:   Gateway: 172.16.0.1
[18:33:56][C][wifi:364]:   DNS1: 172.16.0.1
[18:33:56][C][wifi:365]:   DNS2: 8.8.8.8
[18:33:56][C][logger:233]: Logger:
[18:33:56][C][logger:234]:   Level: DEBUG
[18:33:56][C][logger:235]:   Log Baud Rate: 115200
[18:33:56][C][logger:236]:   Hardware UART: UART0
[18:33:56][C][uart.arduino_esp8266:100]: UART Bus:
[18:33:56][C][uart.arduino_esp8266:101]:   TX Pin: GPIO15
[18:33:56][C][uart.arduino_esp8266:102]:   RX Pin: GPIO13
[18:33:56][C][uart.arduino_esp8266:104]:   RX Buffer Size: 256
[18:33:56][C][uart.arduino_esp8266:106]:   Baud Rate: 19200 baud
[18:33:56][C][uart.arduino_esp8266:107]:   Data Bits: 8
[18:33:56][C][uart.arduino_esp8266:108]:   Parity: NONE
[18:33:56][C][uart.arduino_esp8266:109]:   Stop bits: 1
[18:33:56][C][uart.arduino_esp8266:113]:   Using software serial
[18:33:56][C][victron:011]: Victron:
[18:33:56][C][victron:012]:   Max Power Yesterday 'Max Power Yesterday'
[18:33:56][C][victron:012]:     Device Class: 'power'
[18:33:56][C][victron:012]:     State Class: ''
[18:33:56][C][victron:012]:     Unit of Measurement: 'W'
[18:33:56][C][victron:012]:     Accuracy Decimals: 0
[18:33:56][C][victron:012]:     Icon: 'mdi:power'
[18:33:56][C][victron:013]:   Max Power Today 'Max Power Today'
[18:33:56][C][victron:013]:     Device Class: 'power'
[18:33:56][C][victron:013]:     State Class: ''
[18:33:56][C][victron:013]:     Unit of Measurement: 'W'
[18:33:56][C][victron:013]:     Accuracy Decimals: 0
[18:33:56][C][victron:013]:     Icon: 'mdi:power'
[18:33:56][C][victron:014]:   Yield Total 'Yield Total'
[18:33:56][C][victron:014]:     Device Class: 'power'
[18:33:56][C][victron:014]:     State Class: ''
[18:33:56][C][victron:014]:     Unit of Measurement: 'Wh'
[18:33:56][C][victron:014]:     Accuracy Decimals: 0
[18:33:56][C][victron:014]:     Icon: 'mdi:power'
[18:33:56][C][victron:015]:   Yield Yesterday 'Yield Yesterday'
[18:33:56][C][victron:015]:     Device Class: 'power'
[18:33:56][C][victron:015]:     State Class: ''
[18:33:56][C][victron:015]:     Unit of Measurement: 'Wh'
[18:33:56][C][victron:015]:     Accuracy Decimals: 0
[18:33:56][C][victron:015]:     Icon: 'mdi:power'
[18:33:56][C][victron:016]:   Yield Today 'Yield Today'
[18:33:56][C][victron:016]:     Device Class: 'power'
[18:33:56][C][victron:016]:     State Class: ''
[18:33:56][C][victron:016]:     Unit of Measurement: 'Wh'
[18:33:56][C][victron:016]:     Accuracy Decimals: 0
[18:33:56][C][victron:016]:     Icon: 'mdi:power'
[18:33:56][C][victron:017]:   Panel Voltage 'Panel voltage'
[18:33:56][C][victron:017]:     Device Class: 'voltage'
[18:33:56][C][victron:017]:     State Class: ''
[18:33:56][C][victron:017]:     Unit of Measurement: 'V'
[18:33:56][C][victron:017]:     Accuracy Decimals: 3
[18:33:56][C][victron:017]:     Icon: 'mdi:flash'
[18:33:56][C][victron:018]:   Panel Power 'Panel Power'
[18:33:56][C][victron:018]:     Device Class: 'power'
[18:33:56][C][victron:018]:     State Class: ''
[18:33:56][C][victron:018]:     Unit of Measurement: 'W'
[18:33:56][C][victron:018]:     Accuracy Decimals: 0
[18:33:56][C][victron:018]:     Icon: 'mdi:power'
[18:33:56][C][victron:019]:   Battery Voltage 'Battery voltage'
[18:33:56][C][victron:019]:     Device Class: 'voltage'
[18:33:56][C][victron:019]:     State Class: ''
[18:33:56][C][victron:019]:     Unit of Measurement: 'V'
[18:33:56][C][victron:019]:     Accuracy Decimals: 3
[18:33:56][C][victron:019]:     Icon: 'mdi:flash'
[18:33:56][C][victron:023]:     Unit of Measurement: 'A'
[18:33:56][C][victron:023]:     Accuracy Decimals: 3
[18:33:56][C][victron:023]:     Icon: 'mdi:current-ac'
[18:33:56][C][victron:026]:   Error Code 'Error Code'
[18:33:56][C][victron:026]:     State Class: ''
[18:33:56][C][victron:026]:     Unit of Measurement: ''
[18:33:56][C][victron:026]:     Accuracy Decimals: 0
[18:33:56][C][captive_portal:144]: Captive Portal:
[18:33:56][C][ota:082]: Over-The-Air Updates:
[18:33:56][C][ota:083]:   Address: vedirect.local:8266
[18:33:56][C][ota:086]:   Using Password.
[18:33:56][C][api:134]: API Server:
[18:33:56][C][api:135]:   Address: vedirect.local:6053
[18:33:56][C][api:139]:   Using noise encryption: NO
[18:33:56][C][mdns:084]: mDNS:
[18:33:56][C][mdns:085]:   Hostname: vedirect`

any help appreciated !

KinDR007, 2022-01-24:
Hi ,try to recompile and upload fw again

CountParadox, 2022-01-24:
hmm interesting... this was my first install of it, so I dont have a reference of it working before..

should it be showing data within the logs in esphome? I dont see anything there.. 

Maybe I will try using different pins on the wemos...

CountParadox, 2022-01-24:
I have tried using a different cable (a cut in half victron one) and still not working for me... I checked the settings of the mppt and tx port is set to normal communication.
I suppose ill try hooking up the mppt to a putty terminal and see if there is even any data coming out...?

KinDR007, 2022-01-24:
can you send photos of connection "ve.direct" port and wemos?

CountParadox, 2022-01-24:
[image]

Excuse the mess, you can see the original wire still attached to the wemos.

CountParadox, 2022-01-24:
[image]

KinDR007, 2022-01-25:
Wrong wirring 🙉

KinDR007, 2022-01-25:
[image][image]

CountParadox, 2022-01-25:
Oh my god

CountParadox, 2022-01-25:
I'm so sorry I'm sure I tripple checked it....

I see what happened, the diagrams online seem to have the PCB the other way up, see: [image]

So it's reversed...

KinDR007, 2022-01-25:
> I'm so sorry I'm sure I tripple checked it....
> 
> I see what happened, the diagrams online seem to have the PCB the other way up, see: [image]
> 
> So it's reversed...

Yep flipped😉 let me know and it will work for you 😉

closed 2022-02-07 by KinDR007

DHoogland, 2022-04-06:
I'm facing the exact same issue as @CountParadox, but my wiring should be right. The ESP (ESP8266/NodeMCU V3) worked with HA and showed data. At some point the data wasn't updating anymore, after rebooting the ESP I only got "Unknown" in HA. If I look at the logs of the ESP it only outputs the following:

```
INFO Reading configuration /config/esphome/victron-esp.yaml...
INFO Updating https://github.com/KinDR007/VictronMPPT-ESPHOME.git@main
INFO Starting log output from victron/debug
INFO Connected to MQTT broker!
```

I have the feeling my serial/VE.Direct port isn't working anymore, does anyone have advice how to solve and/or test this? I already tried two different ESP's, no luck.

DHoogland, 2022-04-06:
thanks a lot @syssi! That info is really helpful, it's actually giving some data. 

```
[D][uart_debug:114]: <<< 0D:0A:50:49:44:09:30:78:41:30:35:35:0D:0A:46:57:09:31:35:39:0D:0A:53:45:52:23:09:48:51:32:31:30:32:5A:54:35:4D:39:0D:0A:56:09:31:33:33:31:30:0D:0A:49:09:2D:33:30:0D:0A:56:50:56:09:31:30:0D:0A:50:50:56:09:30:0D:0A:43:53:09:30:0D:0A:4D:50:50:54:09:30:0D:0A:4F:52:09:30:78:30:30:30:30:30:30:30:31:0D:0A:45:52:52:09:30:0D:0A:4C:4F:41:44:09:4F:4E:0D:0A:49:4C:09:30:0D:0A:48:31:39:09:35:31:33:0D:0A:48:32:30:09:30:0D:0A:48:32:31:09:30:0D:0A:48:32:32:09:30
[D][sensor:124]: 'victron yield yesterday': Sending state 0.00000 Wh with 0 decimals of accuracy
[D][sensor:124]: 'victron max power yesterday': Sending state 0.00000 W with 0 decimals of accuracy
[D][sensor:124]: 'victron day number': Sending state 39.00000  with 0 decimals of accuracy
[D][uart_debug:114]: <<< 0D:0A:48:32:33:09:30:0D:0A:48:53:44:53:09:33:39:0D:0A:43:68:65:63:6B:73:75:6D:09:40
[D][sensor:124]: 'victron panel current': Sending state -0.03000 A with 3 decimals of accuracy
[D][sensor:124]: 'victron panel voltage': Sending state 0.01000 V with 3 decimals of accuracy
[D][sensor:124]: 'victron panel power': Sending state 0.00000 W with 0 decimals of accuracy
[D][sensor:124]: 'victron charging mode id': Sending state 0.00000  with 0 decimals of accuracy
[D][text_sensor:067]: 'victron charging mode': Sending state 'Off'
[D][sensor:124]: 'victron tracking mode id': Sending state 0.00000  with 0 decimals of accuracy
[D][text_sensor:067]: 'victron tracking mode': Sending state 'Off'
[D][sensor:124]: 'victron error code': Sending state 0.00000  with 0 decimals of accuracy
[D][text_sensor:067]: 'victron error': Sending state 'No error'
[D][sensor:124]: 'victron load current': Sending state 0.00000 A with 3 decimals of accuracy
[D][sensor:124]: 'victron yield total': Sending state 5130.00000 Wh with 0 decimals of accuracy
[D][sensor:124]: 'victron yield today': Sending state 0.00000 Wh with 0 decimals of accuracy
[D][sensor:124]: 'victron max power today': Sending state 0.00000 W with 0 decimals of accuracy
[D][uart_debug:114]: <<< 0D:0A:50:49:44:09:30:78:41:30:35:35:0D:0A:46:57:09:31:35:39:0D:0A:53:45:52:23:09:48:51:32:31:30:32:5A:54:35:4D:39:0D:0A:56:09:31:33:33:31:30:0D:0A:49:09:2D:33:30:0D:0A:56:50:56:09:31:30:0D:0A:50:50:56:09:30:0D:0A:43:53:09:30:0D:0A:4D:50:50:54:09:30:0D:0A:4F:52:09:30:78:30:30:30:30:30:30:30:31:0D:0A:45:52:52:09:30:0D:0A:4C:4F:41:44:09:4F:4E:0D:0A:49:4C:09:30:0D:0A:48:31:39:09:35:31:33:0D:0A:48:32:30:09:30:0D:0A:48:32:31:09:30:0D:0A:48:32:32:09:30
```

Only its not showing up in MQTT, but I guess that's an configuration issue. Strange that it just stopped working at some point though...