KinDR007/VictronMPPT-ESPHOME #46: Update interval

Terrorr, 2022-05-04:
Hello, 

is there an option as "update_interval" like in pipsolar or daly_bms ?

I started with smartsolar 100/30 and it works fine, but I don't data from each second (30s or 1min will be enought) ;-)

P.S. Is there also option to get data from "network system power" ? I have also 75/15 and it will work temporary (if not used at camping).  For me it would be super just collecting data from main mppt and also got info when smallest one connected. 

Thanks for any advices !

syssi, 2022-05-04:
This component is a bit different compared to the mentioned components using an `update_intervall` because the victron device pushes the data periodically (once per second). The pipsolar component polls/requests the data once per update.

You can apply a throttle filter per sensor to reduce the update interval.

syssi, 2022-05-04:
PS. You can attach multiple victron devices to a single ESP (using this component multiple times) and add a template sensor to sum all power sensors to get the total power.

Terrorr, 2022-05-04:
> This component is a bit different compared to the mentioned components using an `update_intervall` because the victron device pushes the data periodically (once per second). The pipsolar component [...]

Could You advice me how to apply this filter ? Just small example of Your code - I'm using ESP8266 that is sending data to my mqtt HA server (esp is not in local network).

Terrorr, 2022-05-04:
> PS. You can attach multiple victron devices to a single ESP (using this component multiple times) and add a template sensor to sum all power sensors to get the total power.

I tried to connect two victrons with esp8266 nodemcu v3, RX it's only working on GPIO13 (at GPIO3 it didn't receive any data...)

Terrorr, 2022-05-04:
My small HA server with 128GB memory won't  handle to long ;-) 

[image]

syssi, 2022-05-04:
https://github.com/KinDR007/VictronMPPT-ESPHOME/issues/18

```
sensor:
  - platform: victron
    victron_id: victron0
    panel_voltage:
      name: "${name} panel voltage"
      filters:
        - throttle: 60s
    panel_power:
      name: "${name} panel power"
      filters:
        - throttle: 60s
```

Terrorr, 2022-05-05:
> #18
> 
> ```
> sensor:
>   - platform: victron
>     victron_id: victron0
>     panel_voltage:
>       name: "${name} panel voltage"
>       filters:
>         - throttle: 60s
>     panel_power:
>       name: "${name} panel power"
>       filters:
>         - throttle: 60s
> ```

Working perfect! Thanks !

bangadrum, 2022-05-20:
This solution works great for the numeric sensors, but there doesn't seem to be a means to the same for the text sensors (charging mode/error/warning)

Any ideas?

syssi, 2022-05-20:
You are right. Let's introduce a global `throttle` setting to skip readings.

syssi, 2022-05-20:
I've prepared a pull request to introduce a `throttle` parameter. It can be tested using

```
external_components:
  - source: github://syssi/esphome-victron-vedirect@add-throttle
```

Please add the new setting to the `victron` component:

```
victron:
  id: victron0
  uart_id: uart_bus
  throttle: 10s
```

This affects all ESPHome entities because status message will be discarded.

PR #47 (2022-05-21): Introduce throttle parameter
Fixes: #46
commit 547b1fa3 (2022-05-21): Introduce throttle parameter

Fixes: #46
closed 2022-05-21 by KinDR007