KinDR007/VictronMPPT-ESPHOME #48: Throttle breaks sensors

cserem, 2022-05-29:
Hello!

The commit https://github.com/KinDR007/VictronMPPT-ESPHOME/commit/be1c30bfd2e9d4e97b3d63ff97f87f0c2425782b breaks the update of sensors.

I have tried setting the throttle parameter multiple ways, even not setting it, but it doesn't work.

Reverting to https://github.com/KinDR007/VictronMPPT-ESPHOME/commit/835dbadd25b4e7004d43fdbc263d83c0caa42282  fixes the issue.

Could you please investigate?

Thanks 
M

syssi, 2022-05-29:
Sorry for the mess. I will care about a fix. Could you help testing?

syssi, 2022-05-29:
Okay, I know what's going on. My approach is completely wrong.

cserem, 2022-05-29:
Sure thing, just write here and  I will test as soon as possible!

syssi, 2022-05-29:
I've pushed a fix. Please give it a try by using this feature branch:

```
external_components:
  - source: github://syssi/esphome-victron-vedirect@fix-throtte
```

cserem, 2022-05-29:
This seems to fix the issue!

Thank you!

syssi, 2022-05-29:
Could you test the throttle (f.e. `1s` vs. `10s`)?

syssi, 2022-05-29:
I was able to test it on my workbench. This time it should work as intended.

closed 2022-05-29 by KinDR007

cserem, 2022-05-29:
Yes, throttle parameter works as intended, if set. If unconfigured it defaults to 1s properly.

PR #50 (2022-05-29): Fix throttling
This PR fixes a bug introduced by #47. This time it's well tested. ;-) 

Fixes: #48