KinDR007/VictronMPPT-ESPHOME #49: Introduce tags (versioning?)

cserem, 2022-05-29:
Hello!

ESPHome - to my knowledge - doesn't offer an option to refernce a commit as github source.
Because of this, rollback is a real pain: git clone/git checkout manually, then copy  the files to the appropriate folder.
Please evaluate the idea of addig tags to commits, so they can be referenced in the config.yaml for easy "version" switching.

Thanks
M

syssi, 2022-05-29:
I would be happy about Semantic Versioning (https://semver.org/).

cserem, 2022-05-29:
Yes, but wouldn't that require extra effort to classify a change as major/minor/patch?

Wouldn't a date based tag be simpler? yyyymmdd+an extra digit if multiple commits get merged per day?