KinDR007/VictronMPPT-ESPHOME #73: Victron multi RS 48V 6000VA 450V MPPT

cristof48, 2022-06-12:
hello, 
I have received the new hybrid inverter from Victron: Multi RS 48V with 450V MPPT.
I have catch the VE Direct RS232 data and I have the file

[victron (2).odt](https://github.com/KinDR007/VictronMPPT-ESPHOME/files/8886511/victron.2.odt)
could you give me an example how to make esphome file ?
best regards

syssi, 2022-06-12:
This configuration should work for you: https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/phoenix-inverter-esp8266-example.yaml

Please follow these instructions: https://github.com/KinDR007/VictronMPPT-ESPHOME#installation

cristof48, 2022-06-14:
hello,
 thanks for your reply,
it start working , but I have a yellow message into esphome logs (see picture)
can you help me about it ?

[image]

syssi, 2022-06-14:
Could you provide the config yaml of your node? Could you check the wire (Data & GND) between the inverter and the ESP. It looks like the connection is bad. I would expect one message per second.

cristof48, 2022-06-14:
[victron.txt](https://github.com/KinDR007/VictronMPPT-ESPHOME/files/8897957/victron.txt)
the wire between esp32 and multi RS solar is good.

syssi, 2022-06-14:
Does it change something if you set the throttle to `0s`?

syssi, 2022-06-14:
Could you provide another screenshot? I assume your interpretation is wrong. This is the background: Every ve.direct devices publishes a bunch of readings / a message every second. If you use `throttle: 10s` 9 of 10 messages are discarded and one message is published (the green sensor:113 lines at the log).

If you change to `throttle` from `10s` to `0s` every message is published. Therefore you will see a lot of more `sensor:113` lines and may be the number of warnings is still the same.

cristof48, 2022-06-14:
you mean screenshoot from esphome logs ?
can we record the esphome logs ?

syssi, 2022-06-14:
Yes. Screenshot or copy&paste from the esphome logs. I assume you can pipe/tee the output to a file.

syssi, 2022-06-14:
Good job! I have understood now.

1. Your "Multi RS 48V with 450V MPPT" is a bit special.
2. Your first post contains a full status frame

```
PID	0xA442
SER#	HQ2206WVU9D
FWE	109FF
CS	6
ERR	0
AC_OUT_I	10
AC_OUT_V	23001
H19	2551
H20	217
H21	1289
H22	264
H23	1404
HSDS	5
I	-3200
PPV	0
V	56650
VPV	190090
Checksum	⸮:A0922001C0004
:A052200EB00000039
:A8DED00221699
:A0422009500000090
:A8DED0021169A
:ABBED003C4A1D
:ACBEC003C4A0E
```

This is the part of the message this current implementation cannot handle:

```
:A0922001C0004
:A052200EB00000039
:A8DED00221699
:A0422009500000090
:A8DED0021169A
:ABBED003C4A1D
:ACBEC003C4A0E
```

This component/implementation expects the following format (for every received line):

```
key<tab>value

# f.e.
PID    0xA442
```

The special lines starting with a `:` doesn't contain a `\t`. This confuses the parser and results in the warning.

syssi, 2022-06-14:
We could modify the parser to ignore/skip all lines starting with a ":".

syssi, 2022-06-14:
Did you connect RX, TX and GND to the device or just RX and GND? It looks like you have triggered the ve.direct hex protocol (by accident).

cristof48, 2022-06-14:
hi, A great thanks for your help.
Only RX is connected.
 also, I want to show you what I did with HA / Grafana for my 3 batteries 'lifepo4 and lithium ion" with ANT bms
[image]

syssi, 2022-06-14:
Wow. Very nice interface and a lot of energy. Do you use my ANT BMS component?

syssi, 2022-06-14:
Before we filter out the "ve.direct hex lines" could you ask the Victron support why your devices uses two protocols (ascii & hex) at the same time?

This is the normal behavior in my opinion:

1. A ve.direct device transmits ASCII lines periodically
2. If you send HEX requests to the victron device the ASCII ve.direct protocol is disabled and the device responds in HEX.
3. If you don't send HEX requests for ~5 minutes to device falls back into the ve.direct ASCII mode again

cristof48, 2022-06-14:
yes I use 3 x ANT BMS, one for each batteries, the 3 batteries are working simultaneus in // mode

cristof48, 2022-06-14:
I have a contact with a guy in netherland. He is working for Victron.
He asked the firmware team about it, this is what they said.
" Their implementation should be able to handle hex protocol as well. They can ignore its content but they have to be aware of it.
These products always send hex announces.
start of hex frame is ':' and end of hex frame is a newline.
the only position in the text frame that can contain these characters is checksum.
So they should be able to skip the hex content. It is described in the reference implementation."

syssi, 2022-06-14:
In this case lets handle this as adviced. I will prepare a pull request

cristof48, 2022-06-14:
hello, it's working, no more yellow warning.
there is a few sensor not working, no "blue" feedback in esphome logs
-device mode id
-ac out apparant power
-warning code
-tracking mode id
-load curent

syssi, 2022-06-14:
You device doesn't provide data for the "not working sensors". If you check the response / status message of your inverter this is all:

| Key      | Value        | Senor name          |
| -------- | ------------ | --------------------|
| PID      | 0xA442       | device_type         |
| SER#     | HQ2206WVU9D  | serial_number       |
| FWE      | 109FF        | |
| CS       | 6            | charging_mode_id, charging_mode |
| ERR      | 0            | error_code, error   |
| AC_OUT_I | 10           | ac_out_current      |
| AC_OUT_V | 23001        | ac_out_voltage      |
| H19      | 2551         | yield_total         |
| H20      | 217          | yield_today         |
| H21      | 1289         | max_power_today     |
| H22      | 264          | yield_yesterday     |
| H23      | 1404         | max_power_yesterday |
| HSDS     | 5            | day_number          |
| I        | -3200        | battery_current     |
| PPV      | 0            | panel_power         |
| V        | 56650        | battery_voltage     |
| VPV      | 190090       | panel_voltage       |

cristof48, 2022-06-14:
many many thanks for this code and all your help.
Merci beaucoup pour ce code et pour l'aide que tu m'as apporté.

syssi, 2022-06-14:
You are welcome!

PR #74 (2022-06-15): Add Victron Multi RS support
Fixes: #73
closed 2022-06-15 by KinDR007