KinDR007/VictronMPPT-ESPHOME #88: Serial output voltage and compatibility with ESP not tolerant to 5V

hugokernel, 2022-06-30:
Hi,

Apparently, the TX output of Victron devices can be in 3V or 5V, personally, I checked mine and it is in 5V thus not compatible with a direct connection on an ESP32.

It is therefore necessary to use a voltage divider bridge in order not to destroy the ESP.

That's why I think a warning should be added in the Readme.

syssi, 2022-06-30:
Good point! We should point to the Q&A too: https://www.victronenergy.com/live/vedirect_protocol:faq#q4is_the_vedirect_interface_33_or_5v

closed 2022-07-23 by KinDR007

T123S, 2022-10-07:
@hugokernel (or someone else with the 5v problem) can you maybe add a picture with the Victron device, the logic level converter and the esp, so we see it all together? Thx in advance, Theo

hugokernel, 2022-10-17:
@T123S You can follow this example (note that what is important is the ratios of the resistors, the values can be different as long as the ratio is the same).
You can try via: https://www.digikey.fr/fr/resources/conversion-calculators/conversion-calculator-voltage-divider

[image]

yoannchaudet, 2022-12-04:
Is a voltage divider really a valid solution? I am very novice when it comes to electronics so I could be wrong but I have a feeling, this should not work for a digital signal (but be ok for an analog one).

A [logic converter](https://forum.arduino.cc/t/5v-digital-pin-data-on-3-3v-board/967561/2) may be the most reliable way to do it. Will report back in a few weeks!

yoannchaudet, 2022-12-04:
My mistake it does work indeed! I may have used too low of a set of resistors earlier (1k/2k ohm). With 5.1k/10k I get data flowing just fine. 🎉

ssieb, 2023-02-10:
The ESP is officially 5V tolerant (on GPIO pins), so this isn't necessary.

commit 88732fcc (2022-07-23): Add a warning about different logic levels (3.3V vs. 5V) (Closes: #88) (#91)