KinDR007/VictronMPPT-ESPHOME #89: hybride inverter and mppt 150/35

cristof48, 2022-07-12:
hello, since you helped me with the multi rs solar, every thing is OK. I have now a new external victron mppt 150/35.
I wonder if I can use the same ESP32 for multi and mppt ? (serial 16&17) and serial 1&3)
can you help me ?

syssi, 2022-07-12:
Could you edit your post and add a new line containing ``` at front and at the end of the code block?

syssi, 2022-07-12:
Your setup will look like this as soon we are ready: https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/smartsolar-mppt-esp8266-example-multiple-uarts.yaml

cristof48, 2022-07-12:
``` 
substitutions:
  name: phoenix-inverter
  external_components_source: github://syssi/esphome-victron-vedirect@discard-vedirect-hex-traffic

esphome:
  name: ${name}
  platform: ESP32
  board: esp32dev

external_components:
 - source: ${external_components_source}
   refresh: 0s

wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password
  
ota:
api:

logger:
  baud_rate: 0
  #esp8266_store_log_strings_in_flash: false

#mqtt:
#broker: !secret mqtt_host
#username: !secret mqtt_username
#password: !secret mqtt_password
#id: mqtt_client

uart:
  id: uart0
  tx_pin: GPIO 17  # Not connected! The communication is read-only
  rx_pin: GPIO 16 # Connect this this GPIO and GND to the MPPT charger
  baud_rate: 19200
  rx_buffer_size: 512

victron:
  uart_id: uart0
  id: victron0
  throttle: 1s

sensor:
  - platform: victron
    victron_id: victron0
    charging_mode_id:
      name: "${name} charging mode id"
    error_code:
      name: "${name} error code"
    yield_total:
      name: "${name} yield total"
    yield_today:
      name: "${name} yield today"
    max_power_today:
      name: "${name} max power today"
    yield_yesterday:
      name: "${name} yield yesterday"
    max_power_yesterday:
      name: "${name} max power yesterday"
    day_number:
      name: "${name} day number"
    battery_current:
      name: "${name} battery current"
    panel_power:
      name: "${name} panel power"
    battery_voltage:
      name: "${name} battery voltage"
    panel_voltage:
      name: "${name} panel voltage"
    ac_out_current:
      name: "${name} ac out current"
      id: ac_current
    ac_out_voltage:
      name: "${name} ac out voltage"
      id: ac_voltage
      
  - platform: template
    name: "${name} ac out power"
    lambda: |-
       return id(ac_voltage).state * id(ac_current).state;
    update_interval: 1s   
    unit_of_measurement: "W"
    accuracy_decimals: 1
    icon: mdi:power 
    
text_sensor:
  - platform: victron
    victron_id: victron0
    alarm_reason:
      name: "${name} alarm reason"
    # TODO: "OR"
    charging_mode:
      name: "${name} charging mode"
    firmware_version:
      name: "${name} firmware version"
    device_type:
      name: "${name} device type"
    serial_number:
      name: "${name} serial number"
    device_mode:
      name: "${name} device mode"
    warning:
      name: "${name} warning"

binary_sensor:
  - platform: victron
    victron_id: victron0
    relay_state:
      name: "${name} relay state"
 ```

syssi, 2022-07-12:
```
substitutions:
 name: phoenix-inverter
 external_components_source: github://KinDR007/VictronMPPT-ESPHOME@main

esphome:
 name: ${name}
 platform: ESP32
 board: esp32dev

external_components:
- source: ${external_components_source}
  refresh: 0s

wifi:
 ssid: !secret esphome_ssid
 password: !secret esphome_password
 
ota:
api:

logger:
 baud_rate: 0
 #esp8266_store_log_strings_in_flash: false

#mqtt:
#broker: !secret mqtt_host
#username: !secret mqtt_username
#password: !secret mqtt_password
#id: mqtt_client

uart:
  - id: uart0
    tx_pin: GPIO17
    rx_pin: GPIO16
    baud_rate: 19200
    rx_buffer_size: 512
  - id: uart1
    tx_pin: GPIO4
    rx_pin: GPIO5
    baud_rate: 19200
    rx_buffer_size: 256

victron:
  - id: victron_multi
    uart_id: uart0
    throttle: 1s
  - id: victron_mppt
    uart_id: uart1
    throttle: 1s

sensor:
 - platform: victron
   victron_id: victron_multi
   charging_mode_id:
     name: "${name} charging mode id"
   error_code:
     name: "${name} error code"
   yield_total:
     name: "${name} yield total"
   yield_today:
     name: "${name} yield today"
   max_power_today:
     name: "${name} max power today"
   yield_yesterday:
     name: "${name} yield yesterday"
   max_power_yesterday:
     name: "${name} max power yesterday"
   day_number:
     name: "${name} day number"
   battery_current:
     name: "${name} battery current"
   panel_power:
     name: "${name} panel power"
   battery_voltage:
     name: "${name} battery voltage"
   panel_voltage:
     name: "${name} panel voltage"
   ac_out_current:
     name: "${name} ac out current"
     id: ac_current
   ac_out_voltage:
     name: "${name} ac out voltage"
     id: ac_voltage

  - platform: victron
    victron_id: victron_mppt
    max_power_yesterday:
      name: "${name} max power yesterday"
    max_power_today:
      name: "${name} max power today"
    yield_total:
      name: "${name} yield total"
    yield_yesterday:
      name: "${name} yield yesterday"
    yield_today:
      name: "${name} yield today"
    panel_voltage:
      name: "${name} panel voltage"
    panel_power:
      name: "${name} panel power"
    battery_current:
      name: "${name} panel current"
    battery_voltage:
      name: "${name} battery voltage"
    day_number:
      name: "${name} day number"
    charging_mode_id:
      name: "${name} charging mode id"
    error_code:
      name: "${name} error code"
    tracking_mode_id:
      name: "${name} tracking mode id"
    load_current:
      name: "${name} load current"

 - platform: template
   name: "${name} ac out power"
   lambda: |-
      return id(ac_voltage).state * id(ac_current).state;
   update_interval: 1s   
   unit_of_measurement: "W"
   accuracy_decimals: 1
   icon: mdi:power 
   
text_sensor:
 - platform: victron
   victron_id: victron_multi
   alarm_reason:
     name: "${name} alarm reason"
   # TODO: "OR"
   charging_mode:
     name: "${name} charging mode"
   firmware_version:
     name: "${name} firmware version"
   device_type:
     name: "${name} device type"
   serial_number:
     name: "${name} serial number"
   device_mode:
     name: "${name} device mode"
   warning:
     name: "${name} warning"

  - platform: victron
    victron_id: victron_mppt
    charging_mode:
      name: "${name} charging mode"
    error:
      name: "${name} error"
    tracking_mode:
      name: "${name} tracking mode"
    firmware_version:
      name: "${name} firmware version"
    device_type:
      name: "${name} device type"
    serial_number:
      name: "${name} serial number"

binary_sensor:
 - platform: victron
   victron_id: victron_multi
   relay_state:
     name: "${name} relay state"

  - platform: victron
    victron_id: victron_mppt
    load_state:
      name: "${name} load state"
    relay_state:
      name: "${name} relay state"
```

Please attach the second device to GPIO5.

cristof48, 2022-07-12:
I have a board ready with gpio 16&17 and gpio 1&3
I will use gpio 16 and 3 . OK ?

cristof48, 2022-07-12:
there is a probleme : 
[image]

syssi, 2022-07-12:
Line 52 isn't properly indented.

syssi, 2022-07-12:
Using `GPIO3` as `rx_pin` is also fine.

cristof48, 2022-07-12:
still the same :
[image]
[image]
[image]

syssi, 2022-07-12:
I've attached a fixed version of the example (properly indented and lint issues fixed).

[example.yaml.txt](https://github.com/KinDR007/VictronMPPT-ESPHOME/files/9096887/example.yaml.txt)

cristof48, 2022-07-13:
hello, 
I have this esphome msg :
[image]

syssi, 2022-07-13:
This is strange. Is your ESPHome installation up-to-date? Which operation system / distribution and version do you use?

cristof48, 2022-07-13:
with the first yalm (only multi rs solar) it's ok, I can compil it with esphome.
[image]

syssi, 2022-07-13:
Your ESPHome version is outdated. Some parts of the MPPT implementation (the `off_reason_bitmask`) uses some helpers functions of recent ESPHome versions. Please update ESPHome or remove the `off_reason_bitmask` sensor as workaround.

cristof48, 2022-07-13:
ok, I have updated Esphome, and it's OK I have compiled the file.

cristof48, 2022-07-14:
hello Sebastian,
I use gpio 16 for multi rs, and it's ok.
1 can't read anything in gpio 03 and I am sure as the data connected to GPIO 03 are ok
[image]

cristof48, 2022-07-15:
I have changed gpio 01&03 for gpio 18&19.
It's working.
I will try today multi and mppt simultaneously

syssi, 2022-07-15:
I'v no idea why `GPIO3` did not work out of the box. Good luck!

syssi, 2022-07-20:
Feel free to close this issue if it's solved!

closed 2022-07-23 by KinDR007

cristof48, 2022-07-23:
hello, it's working well, but I had to change entity name as they can't be same for multi and mppt.
A big thanks.