KinDR007/VictronMPPT-ESPHOME #97: Maximum sensors rached?

Sebik47, 2022-09-01:
I have about 10 numeric sensors (throttled to 2s) and 2 text sensors that i wanted to send over to home assistant.

Via log output i can see that a maximum of 5 sensors or text_sensors is sent every 2 seconds (only exception are the text_sensors because there is no way to throttle those?).

The other sensor values are not sent (they do not show up in the log output) and are shown as unknown in home assistant.

Nearly every time i install (even when making no changes to the config) the 5 values that are sent, change.

I already have simplified the config by leaving out some filters to get kWh instead of Wh for the yield. Still no change.

I also switched from a ESP8266 to a ESP32 because i thought the processor of the 8266 was too slow but still no change.

Here is my config:

esphome:
  name: victron-smart-solar-250-100

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxx"

ota:
  password: "xxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 10.0.0.199
    gateway: 10.0.0.1
    subnet: 255.255.255.0

external_components:
  - source: github://KinDR007/VictronMPPT-ESPHOME@main

uart:
  id: uart0
  tx_pin: 17  # Not used! The communication is read-only
  rx_pin: 16
  baud_rate: 19200
  rx_buffer_size: 256

victron:
  id: victron0
  uart_id: uart0

sensor:
  - platform: victron
    victron_id: victron0
    battery_voltage:
      name: "SS Victron Batterie Spannung"
      filters:
        - throttle: 2s
      accuracy_decimals: 1
    max_power_yesterday:
      name: "SS Victron Max Leistung Gestern"
      filters:
        - throttle: 2s
      accuracy_decimals: 0
    max_power_today:
      name: "SS Victron Max Leistung Heute"
      accuracy_decimals: 0
      filters:
        - throttle: 2s
    yield_total:
      name: "SS Victron Ausbeute Total"
      filters:
        - throttle: 2s
    yield_yesterday:
      name: "SS Victron Ausbeute Gestern"
      filters:
        - throttle: 2s
    yield_today:
      name: "SS Victron Ausbeute Heute"
      filters:
        - throttle: 2s
    panel_voltage:
      name: "SS Victron Panel Spannung"
      accuracy_decimals: 1
      filters:
        - throttle: 2s
    panel_power:
      name: "SS Victron Panel Leistung"
      filters:
        - throttle: 2s
    battery_current:
      name: "SS Victron Batterie Strom"
      filters:
        - throttle: 2s
      accuracy_decimals: 1
#    charging_mode_id:
#      name: "SS Victron Lademodus ID"
#      filters:
#       - throttle: 2s
#    error_code:
#      name: "SS Victron ERROR ID"
#      filters:
#        - throttle: 2s

text_sensor:
  - platform: victron
    victron_id: victron0
    charging_mode:
      name: "SS Victron Lade Modus"
    error:
      name: "SS Victron ERROR"

LOG OUTPUTS: 

[10:38:54][D][sensor:127]: 'SS Victron Ausbeute Total': Sending state 22690.00000 kWh with 0 decimals of accuracy
[10:38:54][D][sensor:127]: 'SS Victron Ausbeute Heute': Sending state 1430.00000 kWh with 0 decimals of accuracy
[10:38:54][D][sensor:127]: 'SS Victron Max Leistung Heute': Sending state 1844.00000 W with 0 decimals of accuracy
[10:38:54][D][sensor:127]: 'SS Victron Ausbeute Gestern': Sending state 650.00000 kWh with 0 decimals of accuracy
[10:38:54][D][sensor:127]: 'SS Victron Max Leistung Gestern': Sending state 174.00000 W with 0 decimals of accuracy
[10:38:56][D][sensor:127]: 'SS Victron Ausbeute Total': Sending state 22690.00000 kWh with 0 decimals of accuracy
[10:38:56][D][sensor:127]: 'SS Victron Ausbeute Heute': Sending state 1430.00000 kWh with 0 decimals of accuracy
[10:38:56][D][sensor:127]: 'SS Victron Max Leistung Heute': Sending state 1844.00000 W with 0 decimals of accuracy
[10:38:56][D][sensor:127]: 'SS Victron Ausbeute Gestern': Sending state 650.00000 kWh with 0 decimals of accuracy
[10:38:56][D][sensor:127]: 'SS Victron Max Leistung Gestern': Sending state 174.00000 W with 0 decimals of accuracy
[10:38:58][D][sensor:127]: 'SS Victron Ausbeute Total': Sending state 22690.00000 kWh with 0 decimals of accuracy
[10:38:58][D][sensor:127]: 'SS Victron Ausbeute Heute': Sending state 1430.00000 kWh with 0 decimals of accuracy
[10:38:58][D][sensor:127]: 'SS Victron Max Leistung Heute': Sending state 1844.00000 W with 0 decimals of accuracy
[10:38:58][D][sensor:127]: 'SS Victron Ausbeute Gestern': Sending state 650.00000 kWh with 0 decimals of accuracy
[10:38:58][D][sensor:127]: 'SS Victron Max Leistung Gestern': Sending state 174.00000 W with 0 decimals of accuracy
[10:39:01][D][sensor:127]: 'SS Victron Ausbeute Total': Sending state 22690.00000 kWh with 0 decimals of accuracy
[10:39:01][D][sensor:127]: 'SS Victron Ausbeute Heute': Sending state 1430.00000 kWh with 0 decimals of accuracy
[10:39:01][D][sensor:127]: 'SS Victron Max Leistung Heute': Sending state 1844.00000 W with 0 decimals of accuracy
[10:39:01][D][sensor:127]: 'SS Victron Ausbeute Gestern': Sending state 650.00000 kWh with 0 decimals of accuracy
[10:39:01][D][sensor:127]: 'SS Victron Max Leistung Gestern': Sending state 174.00000 W with 0 decimals of accuracy
[10:39:03][D][sensor:127]: 'SS Victron Ausbeute Total': Sending state 22690.00000 kWh with 0 decimals of accuracy
[10:39:03][D][sensor:127]: 'SS Victron Ausbeute Heute': Sending state 1430.00000 kWh with 0 decimals of accuracy
[10:39:03][D][sensor:127]: 'SS Victron Max Leistung Heute': Sending state 1844.00000 W with 0 decimals of accuracy
[10:39:03][D][sensor:127]: 'SS Victron Ausbeute Gestern': Sending state 650.00000 kWh with 0 decimals of accuracy
[10:39:03][D][sensor:127]: 'SS Victron Max Leistung Gestern': Sending state 174.00000 W with 0 decimals of accuracy
[10:39:05][D][sensor:127]: 'SS Victron Ausbeute Total': Sending state 22690.00000 kWh with 0 decimals of accuracy
[10:39:05][D][sensor:127]: 'SS Victron Ausbeute Heute': Sending state 1430.00000 kWh with 0 decimals of accuracy
[10:39:05][D][sensor:127]: 'SS Victron Max Leistung Heute': Sending state 1844.00000 W with 0 decimals of accuracy
[10:39:05][D][sensor:127]: 'SS Victron Ausbeute Gestern': Sending state 650.00000 kWh with 0 decimals of accuracy
[10:39:05][D][sensor:127]: 'SS Victron Max Leistung Gestern': Sending state 174.00000 W with 0 decimals of accuracy

After a new install:

[10:52:47][D][sensor:127]: 'SS Victron Batterie Spannung': Sending state 26.42000 V with 1 decimals of accuracy
[10:52:47][D][sensor:127]: 'SS Victron Batterie Strom': Sending state 61.80000 A with 1 decimals of accuracy
[10:52:47][D][sensor:127]: 'SS Victron Panel Spannung': Sending state 164.17999 V with 1 decimals of accuracy
[10:52:47][D][sensor:127]: 'SS Victron Panel Leistung': Sending state 1705.00000 W with 0 decimals of accuracy
[10:52:47][D][text_sensor:067]: 'SS Victron Lade Modus': Sending state 'Bulk'
[10:52:48][D][text_sensor:067]: 'SS Victron Lade Modus': Sending state 'Bulk'
[10:52:49][D][sensor:127]: 'SS Victron Batterie Spannung': Sending state 26.42000 V with 1 decimals of accuracy
[10:52:49][D][sensor:127]: 'SS Victron Batterie Strom': Sending state 61.70000 A with 1 decimals of accuracy
[10:52:49][D][sensor:127]: 'SS Victron Panel Spannung': Sending state 164.23000 V with 1 decimals of accuracy
[10:52:49][D][sensor:127]: 'SS Victron Panel Leistung': Sending state 1702.00000 W with 0 decimals of accuracy
[10:52:49][D][text_sensor:067]: 'SS Victron Lade Modus': Sending state 'Bulk'
[10:52:51][D][text_sensor:067]: 'SS Victron Lade Modus': Sending state 'Bulk'
[10:52:52][D][sensor:127]: 'SS Victron Batterie Spannung': Sending state 26.42000 V with 1 decimals of accuracy
[10:52:52][D][sensor:127]: 'SS Victron Batterie Strom': Sending state 61.60000 A with 1 decimals of accuracy
[10:52:52][D][sensor:127]: 'SS Victron Panel Spannung': Sending state 164.25000 V with 1 decimals of accuracy
[10:52:52][D][sensor:127]: 'SS Victron Panel Leistung': Sending state 1701.00000 W with 0 decimals of accuracy
[10:52:52][D][text_sensor:067]: 'SS Victron Lade Modus': Sending state 'Bulk'
[10:52:53][D][text_sensor:067]: 'SS Victron Lade Modus': Sending state 'Bulk'
[10:52:54][D][sensor:127]: 'SS Victron Batterie Spannung': Sending state 26.42000 V with 1 decimals of accuracy
[10:52:54][D][sensor:127]: 'SS Victron Batterie Strom': Sending state 61.60000 A with 1 decimals of accuracy
[10:52:54][D][sensor:127]: 'SS Victron Panel Spannung': Sending state 163.89999 V with 1 decimals of accuracy
[10:52:54][D][sensor:127]: 'SS Victron Panel Leistung': Sending state 1701.00000 W with 0 decimals of accuracy
[10:52:54][D][text_sensor:067]: 'SS Victron Lade Modus': Sending state 'Bulk'
[10:52:55][D][text_sensor:067]: 'SS Victron Lade Modus': Sending state 'Bulk'

closed 2022-09-01 by Sebik47

syssi, 2022-09-01:
How did you solve your issue?

Sebik47, 2022-09-01:
Set the throttle to 6 seconds for the whole uart0.

The standard code only sends 5 values, when you set the throttle time higher the controller (or the code of this program) has more time and can send another batch (the remaining 5) of values. I think the code could be optimised to give a higher update rate but 6 seconds is fine for me at this time.

Also, when you throttle some of the values with a different throttle value it somehow breaks the code and after some time the values stop refreshing. I think thats the other issue that is still open.