KinDR007/VictronMPPT-ESPHOME #98: Only able to use 2 UARTs on esp32

cainef, 2022-09-10:
Hi,
First up, I want to thank you for creating this.
I'm hoping someone here can point me in the right direction.

I am using an esp32 board (https://www.aliexpress.com/item/1005001929935550.html) to connect to a 100/20, 100/30 and a smart shunt.

Unfortunately I can only seem to get two UARTs working at the same time. 

In the code below only uart0 and uart1 will work - uart2 does not come up in the logs at all. If I comment out uart1 uart2 will work fine.

I have tried disabling logging, which does not seem to have any affect. My understanding (and I could be wrong here) is that the esp32 has 3 uarts, but the rx and tx pins for one is assigned to the USB port, although this should be able changed by specifying different pins in the code as below.

Am I doing anything obviously wrong here or should I look to add another esp32 for one of the devices? 

Regards

`esphome:
  name: victron

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "ZPZT0ltwBx0DF0GNuaPqWb1jcBwIx02gk76CF1miG0M="

ota:
  password: "5e7c5c128ad3af4b15acab47b891ce8d"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  manual_ip:
    static_ip: 172.16.0.31
    gateway: 172.16.0.1
    subnet: 255.255.255.0
    dns1: 172.16.0.1

substitutions:
  name: victron
  device0: shunt
  device1: mppt30
  device2: mppt20

external_components:
  - source: github://KinDR007/VictronMPPT-ESPHOME@main

uart:
  - id: uart0
    tx_pin: GPIO1  # Not connected! The communication is read-only
    rx_pin: GPIO3  # Connect this this GPIO and GND to the first MPPT charger
    baud_rate: 19200
    rx_buffer_size: 256
  - id: uart1
    tx_pin: GPIO25  # Not connected! The communication is read-only
    rx_pin: GPIO26  # Connect this this GPIO and GND to the second MPPT charger
    baud_rate: 19200
    rx_buffer_size: 256
  - id: uart2
    tx_pin: GPIO17  # Not connected! The communication is read-only
    rx_pin: GPIO16  # Connect this this GPIO and GND to the second MPPT charger
    baud_rate: 19200
    rx_buffer_size: 256

victron:
  - id: shunt
    uart_id: uart0
    throttle: 5s
  - id: mppt30
    uart_id: uart1
    throttle: 5s
  - id: mppt20
    uart_id: uart2
    throttle: 5s

sensor:
  - platform: victron
    victron_id: shunt
    battery_voltage:
      name: "${device0} battery voltage"
    battery_current:
      name: "${device0} battery current"
    instantaneous_power:
      name: "${device0} instantaneous power"
    consumed_amp_hours:
      name: "${device0} consumed amp hours"
    state_of_charge:
      name: "${device0} state of charge"
    time_to_go:
      name: "${device0} time to go"
    cumulative_amp_hours_drawn:
      name: "${device0} cumulative amp hours drawn"

  - platform: victron
    victron_id: mppt30
    max_power_yesterday:
      name: "${device1} max power yesterday"
    max_power_today:
      name: "${device1} max power today"
    yield_total:
      name: "${device1} yield total"
    yield_yesterday:
      name: "${device1} yield yesterday"
    yield_today:
      name: "${device1} yield today"
    panel_voltage:
      name: "${device1} panel voltage"
    panel_power:
      name: "${device1} panel power"
    battery_current:
      name: "${device1} panel current"
    battery_voltage:
      name: "${device1} battery voltage"
    day_number:
      name: "${device1} day number"
    charging_mode_id:
      name: "${device1} charging mode id"
    error_code:
      name: "${device1} error code"
    tracking_mode_id:
      name: "${device1} tracking mode id"
    load_current:
      name: "${device1} load current"
      
  - platform: victron
    victron_id: mppt20
    max_power_yesterday:
      name: "${device2} max power yesterday"
    max_power_today:
      name: "${device2} max power today"
    yield_total:
      name: "${device2} yield total"
    yield_yesterday:
      name: "${device2} yield yesterday"
    yield_today:
      name: "${device2} yield today"
    panel_voltage:
      name: "${device2} panel voltage"
    panel_power:
      name: "${device2} panel power"
    battery_current:
      name: "${device2} panel current"
    battery_voltage:
      name: "${device2} battery voltage"
    day_number:
      name: "${device2} day number"
    charging_mode_id:
      name: "${device2} charging mode id"
    error_code:
      name: "${device2} error code"
    tracking_mode_id:
      name: "${device2} tracking mode id"
    load_current:
      name: "${device2} load current"

text_sensor:
  - platform: victron
    victron_id: shunt
    alarm_condition_active:
      name: "${device0} alarm condition active"
    alarm_reason:
      name: "${device0} alarm reason"
    model_description:
      name: "${device0} model description"
    firmware_version:
      name: "${device0} firmware version"
    device_type:
      name: "${device0} device type"
    serial_number:
      name: "${device0} serial number"
    dc_monitor_mode:
      name: "${device0} dc monitor mode"

  - platform: victron
    victron_id: mppt30
    charging_mode:
      name: "${device1} charging mode"
    error:
      name: "${device1} error"
    tracking_mode:
      name: "${device1} tracking mode"
    firmware_version:
      name: "${device1} firmware version"
    device_type:
      name: "${device1} device type"
    serial_number:
      name: "${device1} serial number"
    model_description:
      name: "${device1} model description"
      
  - platform: victron
    victron_id: mppt20
    charging_mode:
      name: "${device2} charging mode"
    error:
      name: "${device2} error"
    tracking_mode:
      name: "${device2} tracking mode"
    firmware_version:
      name: "${device2} firmware version"
    device_type:
      name: "${device2} device type"
    serial_number:
      name: "${device2} serial number"
    model_description:
      name: "${device2} model description"`

syssi, 2022-09-10:
Which ESPHome version do you use?

cainef, 2022-09-10:
v2022.6.3 running in a container in Home Assistant

cainef, 2022-09-10:
closing issue as changing the pins to the following fixed the issue;

```
uart:
  - id: uart0
    baud_rate: 19200
    tx_pin: GPIO1
    rx_pin: GPIO3
  - id: uart1
    baud_rate: 19200
    tx_pin: GPIO14
    rx_pin: GPIO4
  - id: uart2
    baud_rate: 19200
    tx_pin: GPIO32
    rx_pin: GPIO33
```

closed 2022-09-10 by cainef