KinDR007/VictronMPPT-ESPHOME #99: battery_temperature not working

kkaminski7, 2022-09-17:
Hi
I have "Victron Smartsolar 100/20" mppt with "Smart battery sense" connected. Everything seems to work fine so far except "battery_temperature". Is it working?
Maybe I have defined it wrong in esphome, below my config for it:
```
    battery_temperature:
      name: "Battery Temperature"
      device_class: temperature
      state_class: measurement
      unit_of_measurement: "°C"
```

syssi, 2022-09-18:
Could you enable the debug output/mode of the `uart` component and provide a log of your Esphome node? It should include a complete status frame of your victron device. I would like to understand which measurements are available and which aren't part of the status frame.

kkaminski7, 2022-09-18:
I have set debug according to debug-esp32-example.yaml
```
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

Downloaded logs attached.
[logs_esphome-web-da590e_logs.txt](https://github.com/KinDR007/VictronMPPT-ESPHOME/files/9594375/logs_esphome-web-da590e_logs.txt)

syssi, 2022-09-18:
This is the response of your Victron device:

```
PID     0xA060
FW     159
SER#   HQ21254RHYW
V      13230
I      -20
VPV    16590
PPV    5
CS     3
MPPT   2
OR     0x00000000
ERR    0
LOAD   ON
IL     400
H19    2427
H20    13
H21    92
H22    22
H23    125
HSDS   122
Checksum       �
```

Do you see the temperature you are looking for here?

syssi, 2022-09-18:
Battery temperature (key `T`) isn't available. It looks like your are looking for a measurement your Victron device doesn't provide (at the ve.direct ascii telegram).

kkaminski7, 2022-09-18:
I see. Thanks for your help.

closed 2022-09-18 by kkaminski7