KinDR007/VictronMPPT-ESPHOME #101: Installation Error

bemilamedia, 2022-11-05:
hi,
I have a problem to install it on my Wemos D1 Mini v4.
I'm using the ESPHome Integration for HomeAssistant.

When I use D8, D7 i got the error message:
`INFO Reading configuration /config/esphome/esphome-web-e37f4b.yaml...
Failed config

uart: [source /config/esphome/esphome-web-e37f4b.yaml:30]
  - id: uart0
    
    Cannot resolve pin name 'D8' for board esp01_1m.
    tx_pin: D8
    
    Cannot resolve pin name 'D7' for board esp01_1m.
    rx_pin: D7
    baud_rate: 19200
    rx_buffer_size: 256`

When I delete the "D" and I use
[image]
I got the message:
`INFO Reading configuration /config/esphome/esphome-web-e37f4b.yaml...
Failed config

uart: [source /config/esphome/esphome-web-e37f4b.yaml:30]
  - id: uart0
    
    This pin cannot be used on ESP8266s and is already used by the flash interface (function: Flash Data 1).
    tx_pin: 8
    
    This pin cannot be used on ESP8266s and is already used by the flash interface (function: Flash Data 0).
    rx_pin: 7
    baud_rate: 19200
    rx_buffer_size: 256`

I think I'm just a little bit to stupid to setup and use the correct way of ESPHome-Integration in HomeAssistant.
I have treid it with "wirelessly", "Plugedin to this computer", "manuell download"

Is there a little tutorial how to setup this correct?

Kind regards
Dominic

syssi, 2022-11-05:
If you want to access the GPIOs using the `D1...Dx` labels you have to use the correct board:

```
esp8266:
  board: d1_mini
```

The board setting `esp01_1m` means something like: This is a raw esp8266 chip without any labels / dev board attached.

Instead of using `D7` you could access the GPIO directly by `GPIO13`:

```
uart:
  id: uart0
  baud_rate: 19200
  tx_pin: GPIO15
  rx_pin: GPIO13
```

bemilamedia, 2022-11-05:
@syssi
thanks for your assist.

Setting up the right board and GPIO helps and now I can install the firmware.

On the printed PCB are no informations about any Dx or GPIOs.
the backside looks like the picture in your first link: [https://arduino-projekte.info/wp-content/uploads/2022/02/Wemos-Lolin-D1-mini-V4.0-Back-840x630.webp](https://arduino-projekte.info/wp-content/uploads/2022/02/Wemos-Lolin-D1-mini-V4.0-Back-840x630.webp)

syssi, 2022-11-05:
The numbers at your PCB are the GPIOs. You have to prefix the numbers with `GPIO`.

bemilamedia, 2022-11-05:
> The numbers at your PCB are the GPIOs. You have to prefix the numbers with `GPIO`.

But where can I find the right GPIOs for Tx,Rx. On the PCB only "TX" /"RX" printed without any numbers.

syssi, 2022-11-05:
GPIO1 and GPIO3.

syssi, 2022-11-06:
Please close the issue if it's solved!

closed 2022-11-06 by bemilamedia