KinDR007/VictronMPPT-ESPHOME #102: Secret 'mqtt_host' not defined

nuvolare12, 2022-11-08:
Why is there this error?
INFO Reading configuration /config/esphome/test.yaml...
ERROR Error while reading config: Invalid YAML syntax:

Secret 'mqtt_host' not defined
  in "/config/esphome/test.yaml", line 26, column 11:
      broker: !secret mqtt_host
              ^

kowi-k, 2022-11-09:
Probably because you haven't setup your secrets.yaml. 

https://esphome.io/guides/faq.html#tips-for-using-esphome

syssi, 2022-11-09:
You have to define 5 different secrets

```
# secrets.yaml
wifi_ssid: YOUR_WLAN_SSID
wifi_password: YOUR_WIFI_PASSWORD
mqtt_host: YOUR_MQTT_BROKER
mqtt_username: YOUR_MQTT_USER
mqtt_password: YOUR_MQTT_PASSWORD
```

nuvolare12, 2022-11-09:
Hello, no still same error .
I did a new file in config/esphome/secrets.yaml  and i saved this code

`mqtt:
  mqtt_host: "mosquitto"
  mqtt_username: "********"
  mqtt_password: "********"
  mqtt_client: "mqtt_client"`

the original code is this and the lineonf the error is 26

`25 mqtt:
26  broker: !secret mqtt_host
27 username: !secret mqtt_username
28  password: !secret mqtt_password
29 id: mqtt_client`

I can't find the problem

syssi, 2022-11-09:
It looks like the format of your `secrets.yaml` is wrong. The file is a key-value store. Each line contains a `KEY: VALUE`. No spaces, no indents etc.

nuvolare12, 2022-11-09:
```
# Your Wi-Fi SSID and password
wifi_ssid:"**********"
wifi_password:"**********"
#################################MQTT########################
mqtt_host:"Mosquitto"
mqtt_username:"********"
mqtt_password:"*********"
mqtt_client:"mqtt_client"
```

It is now correct?

syssi, 2022-11-09:
No. This format should work:

```
# Your Wi-Fi SSID and password
wifi_ssid: "**********"
wifi_password: "**********"
#################################MQTT########################
mqtt_host: "Mosquitto"
mqtt_username: "********"
mqtt_password: "*********"
mqtt_client: "mqtt_client"
```

nuvolare12, 2022-11-09:
yes, now it's working perfectly. But now i have this error
````
ERROR Error resolving IP address of victron-smartshunt.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname
```

But the esp is online, why there is this error?

kowi-k, 2022-11-10:
Apparently mDNS is not working. Try to reach the ESP with its IP directly.

syssi, 2022-11-10:
Not working means: mDNS advertisements doesn't arrive at your host. Please check your networking or switch to static IP addresses

nuvolare12, 2022-11-10:
Hello, i tryed all day. Now the esp is working fine. I don't know what was the reason, but definitelly i connect trought the usb in the pc, i'have installed the code, and now it is working fine.
But now what is the problem? seem like every tie there is something wrong...

`INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from victron-smartshunt/debug
ERROR Cannot connect to MQTT broker: [Errno -5] No address associated with hostname`

but in the code in the secret is
`mqtt_host: "mosquitto"
mqtt_username: "**********"
mqtt_password: "**********"
mqtt_client: "mqtt_client"`

nuvolare12, 2022-11-11:
i found the problem . The mqtt_host: is 127.0.0.1

syssi, 2022-11-11:
Is your setup up and running now? Any open questions?

nuvolare12, 2022-11-11:
Yes, i think is running, just where i have to find the sensor? Because i'm looking for it but i doesn't find in the list, have do some think else? I just have the Smart shunt connected to the esp8266. Thank you

syssi, 2022-11-11:
About what list we are talking about? Do you use Home Assistant?

nuvolare12, 2022-11-11:
Yes, i'm using home Assistant, and i thought to find the list of the sensor in the home Assistant like indicator, or like entity to add. Is it this right or i have to do other to it like Shelly? Thank you

syssi, 2022-11-11:
Please remove the `mqtt` component from your YAML and you the `api` component instead. Go to the "integration page" and add the a new integration called "ESPHome". Click on the ESPHome integration tile and add the new node using the IP address.

nuvolare12, 2022-11-11:
if i remouve the mqq component from yaml what i have to punt inside the api?
Because if i don't do anything, just remouve the mqtt the sistem give me an error like this
`ERROR No remote or local logging method configured (api/mqtt/logger)`

syssi, 2022-11-11:
The updated YAML should look like this:

```
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
logger:
api:
```

nuvolare12, 2022-11-11:
the code that i pasted in yaml, from the link https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/smartshunt-esp8266-example.yaml

is

`https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/smartshunt-esp8266-example.yaml`

nuvolare12, 2022-11-11:
> 

Sorry, but what i have to put inside the
Ita, logger and api? Nobody speak about this, so i don't know what to do. Thank you

syssi, 2022-11-11:
Nothing. Just loading the components is sufficient.

nuvolare12, 2022-11-11:
perfetc, now it's working fine. Sorry for the trouble. Thank you so much for  your hlp

syssi, 2022-11-12:
Please close the issue if it's resolved!

closed 2022-11-28 by nuvolare12