KinDR007/VictronMPPT-ESPHOME #104: Use VCC to power the Wemos on the 5V-pin?

snipah, 2022-11-29:
Brief question: Can I use the 5V from the VCC to power the Wemos on the 5V-Pin or is the USB-cable required?

KinDR007, 2022-11-30:
yes, the VIN pin can be used to power the esp board .. where do you want to power it from?  if from battery storage, the best option is to use stepdown and connect only one cable to the victron ve.direct port, namely D7 (gnd to ve.direct port NC!) look this forum https://community.home-assistant.io/t/victron-energy-vedirect-bluetooth/20726/109

kowi-k, 2022-11-30:
Does the 5v pin provide enough current to power the ESP?

snipah, 2022-11-30:
> yes, the VIN pin can be used to power the esp board .. where do you want to power it from? if from battery storage, the best option is to use stepdown and connect only one wire to the victron ve.direct port, namely D7 (gnd to ve.direct port NC!) look this forum https://community.home-assistant.io/t/victron-energy-vedirect-bluetooth/20726/109

...read the linked thread and read that  ve.direct port has current limit (15-20mA) and thus cannot power the ESP. I will order a stepdown-converter and use this to power the ESP, but where to get the 12V input ideally? Connect to the Victron output?

closed 2022-12-04 by snipah