KinDR007/VictronMPPT-ESPHOME #105: Switch for load output possible?

snipah, 2022-12-18:
As the load output is configurable/ switchable via the Victron App, I want to ask if this is achievable via MQTT as well? It would be great to have control over the load output via Home Assistant.

syssi, 2022-12-18:
This implementation uses the read-only ve.direct ASCII protocol right now. I assume we have to switch to the binary version to control the device.

SeByDocKy, 2023-01-02:
> This implementation uses the read-only ve.direct ASCII protocol right now. I assume we have to switch to the binary version to control the device.

And even the BLE version ;)

syssi, 2023-06-14:
Ah, I understand the concept now. The RX pin is the unused one in a read-only setup.

Git-ChrisSc, 2023-06-14:
https://www.victronenergy.com/media/pg/Manual_BlueSolar_MPPT_75-10_up_to_100-20/de/configuration-and-settings.html#UUID-b6cabb55-8d14-4fc9-190f-c0856aab60d3

Here is an Manual for Bluesolar

Git-ChrisSc, 2023-06-15:
I can confirm, that this way is working.

If you put 5V on RX Pin and config RX for switching the Load-Output in the Bluetooth-App.

After this, you have to build a "switch-component" in ESPHOME

syssi, 2023-06-16:
Could you try to drive the `load output` using a GPIO with 3.3V TTL:

```
switch:
  - platform: gpio
    name: "${name} load output"
    pin:
      number: GPIO14
      mode:
        pulldown: true
```

Git-ChrisSc, 2023-06-21:
Hi, 

Oh, i can test it after my vacation - because my "Jerrycan-Battery" is actuall in use and allready solderd with iron.

But: Why with 3V? For stability (in longterm use) I use a cheap logic level converter, because i wouldn't use 5V on the ESP32 Pins.
https://www.amazon.de/gp/product/B07BZJ73LV/

i using not the TTL as Output, i have solderd a "Jumperbridge" on the platin for XOR.
[image]

On the HV-Side i solderd the VCC from Victron and on HG is the Victron GND connected.
H1 and H4 connected to RX und TX Pins on the Victron.

syssi, 2023-06-21:
Let's ignore the voltage level of the UART communication for now (off-topic). If we focus on "how to control the load output of a victron device using the RX pin" we have to apply some voltage to this pin. If we simply attach a GPIO to the RX pin the voltage level will be 3.3V because the ESP is a 3.3V device. The RX pin probably expects 5V here (on some victron devices). I would like to know the lower voltage level is sufficient to drive the load output.

Git-ChrisSc, 2023-06-21:
`"UART communication for now (off-topic)"`
I think that i get missunderstand, because my english isnt good. My intention wasn't to discuss about the voltaglevel by UART.

Some facts for the focus: 
When i apply 3.3V on the RX Pin the Victron will switch the load output to on. 
(after right config in VictronApp on SmartSolar MPPT _**1_)

In the previous post, I only just wanted to warn that 5V is not a good idea for long-term use on the RX of the ESP, and that you could switch everything via a logic-level converter that doesn't just do TTL.

So the Answere of the question above is:
1. You can switch the load output with an ESP(and ESPHome) and there is no need of this lib: But you have to config the Victron in the Right-Way _**1_.
2. Switching the Load Output need no more additional Hardware: But for reading Victron it's safer to use a Logik-Level-Converter and now you can also use it for switching the Output with right 5V

_**1_ And as an additional Answer:
I use it and tested it in the Last 5 days. Every nigth.
In the App you have to configure 
1. under menue-point: Battery -> Operation-Mode: Always on
2. under menue-point: RX port -> RX port function: Load output on/off normal 
.... i have NOT tested it with some other options as battery-safer or something else

syssi, 2023-06-21:
It took some time to understand your idea. :-) Long story short: The TX channel of a (two channel) level shifter can be used to drive the load output using 5V.

zechnkaas, 2024-03-25:
just a quick not from me:
i do have a 75/10 and a 75/15 in a setup with Rx pin set to "Load output on/off normal" and putting Rx to GND in the victron. This turns off the load on closed and turn back on after releasing the relais. (it takes a few seconds until the victron actually switches the load)