KinDR007/VictronMPPT-ESPHOME #108: Invalid data being reported.

mrnoname22, 2023-03-01:
Im having issues with incorrect data being reported. Example would be Voltage. Im running a 12v batt and see reported values of 0v and 30v. Its reporting these vaules a couple times an hour. I see it across other sensors as well. 

Im wondering what the best place to start troubleshooting this would be.  I was initially using an esp8266, switched to an esp32, then switched to a WT32-ETH01 to eliminate wifi as being an issue.

Currently running WT32-ETH01 with the following config.

substitutions:
  name: esp32-eth
  device0: mppt
  device1: shunt
  external_components_source: github://KinDR007/VictronMPPT-ESPHOME@main
  accuracy: "2"

esphome:
  name: esp32-eth
  friendly_name: esp32-eth
  platform: ESP32
  board: esp-wrover-kit

external_components:
  - source: ${external_components_source}
    refresh: 0s

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "****"

ota:
  password: "******"

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
  domain: .tnw.lan

time:
  - platform: homeassistant

uart:
  - id: uart0
    tx_pin: GPIO14  # Not connected! The communication is read-only
    rx_pin: GPIO36  # Connect this this GPIO and GND to the first MPPT charger
    baud_rate: 19200
    rx_buffer_size: 256
  - id: uart1
    tx_pin: GPIO16  # Not connected! The communication is read-only
    rx_pin: GPIO39  # Connect this this GPIO and GND to the second MPPT charger
    baud_rate: 19200
    rx_buffer_size: 256

victron:
  - id: victron0
    uart_id: uart0
    throttle: 1s
  - id: victron1
    uart_id: uart1
    throttle: 1s

sensor:
  - platform: victron
    victron_id: victron0
    panel_voltage:
      name: "${device0} Panel voltage"
      id: pv
    battery_voltage:
      name: "${device0} Battery voltage"
      id: bv
    battery_current:
      name: "${device0} Battery current"
      id: bc
    max_power_yesterday:
      name: "${device0} Max power yesterday"
      id: max_power_yesterday
    max_power_today:
      name: "${device0} Max power today"
      id: max_power_today
      accuracy_decimals: ${accuracy}
    yield_total:
      name: "${device0} Yield total"
      id: yield_total
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
      unit_of_measurement: kWh
      accuracy_decimals: ${accuracy}
    yield_yesterday:
      name: "${device0} Yield yesterday"
      id: yield_yesterday
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
      unit_of_measurement: kWh
      accuracy_decimals: ${accuracy}
    yield_today:
      name: "${device0} Yield today"
      id: yt
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
      unit_of_measurement: kWh
      accuracy_decimals: ${accuracy}
    panel_power:
      name: "${device0} Panel power"
      id: panel_power
    day_number:
      name: "${device0} Day sequence number"
      id: day_number
      icon: "mdi:calendar-today"
    charging_mode_id:
      name: "${device0} Charger mode id"
    error_code:
      name: "${device0} Error code"
    tracking_mode_id:
      name: "${device0} Tracker mode id"

  - platform: uptime
    name: "${name} Uptime"
    id: uptime_s
    update_interval: 60s

  - platform: victron
    victron_id: victron1
    battery_voltage:
      name: "${device1} battery voltage"
    battery_current:
      name: "${device1} battery current"
    battery_temperature:
      name: "${device1} battery temperature"
    instantaneous_power:
      name: "${device1} instantaneous power"
    consumed_amp_hours:
      name: "${device1} consumed amp hours"
    state_of_charge:
      name: "${device1} state of charge"
    time_to_go:
      name: "${device1} time to go"
    depth_of_the_deepest_discharge:
      name: "${device1} depth of the deepest discharge"
    depth_of_the_last_discharge:
      name: "${device1} depth of the last discharge"
    depth_of_the_average_discharge:
      name: "${device1} depth of the average discharge"
    number_of_charge_cycles:
      name: "${device1} number of charge cycles"
    number_of_full_discharges:
      name: "${device1} number of full discharges"
    cumulative_amp_hours_drawn:
      name: "${device1} cumulative amp hours drawn"
    min_battery_voltage:
      name: "${device1} min battery voltage"
    max_battery_voltage:
      name: "${device1} max battery voltage"
    last_full_charge:
      name: "${device1} last full charge"
    number_of_automatic_synchronizations:
      name: "${device1} number of automatic synchronizations"
    number_of_low_main_voltage_alarms:
      name: "${device1} number of low main voltage alarms"
    number_of_high_main_voltage_alarms:
      name: "${device1} number of high main voltage alarms"
    amount_of_discharged_energy:
      name: "${device1} amount of discharged energy"
    amount_of_charged_energy:
      name: "${device1} amount of charged energy"
    dc_monitor_mode_id:
      name: "${device1} dc monitor mode id"

  - platform: template   # FVE percent
    name: "${device0} FVE percent"  # 500Wp max power of my FVE grid
    lambda: |-
        if (((((id(panel_power).state) / 500) * 100) >= 120) && ((((id(panel_power).state) / 500) * 100) < 0)) {
          return NAN;
        }
        return (((id(panel_power).state) / 500) * 100);
    icon: "mdi:gauge"
    unit_of_measurement: "%"
    device_class: "power"
    accuracy_decimals: ${accuracy}
    update_interval: 30s

  - platform: template   # median generated day energy
    name: "${device0} Day median generated "  # 500Wp max power of my FVE grid
    lambda: |-
        return (id(yield_total).state) / (id(day_number).state);
    update_interval: 60s
    icon: "mdi:gauge"
    unit_of_measurement: "kWh"
    device_class: "power"
    accuracy_decimals: 3

text_sensor:
  - platform: victron
    victron_id: victron0
    charging_mode:
      name: "${device0} Charging Mode"
    error:
      name: "${device0} Error"
    tracking_mode:
      name: "${device0} Tracking Mode"
    device_type:
      name: "${device0} Device Type"

  - platform: victron
    victron_id: victron1
    alarm_condition_active:
      name: "${device1} alarm condition active"
    alarm_reason:
      name: "${device1} alarm reason"
    model_description:
      name: "${device1} model description"
    device_type:
      name: "${device1} device type"
    dc_monitor_mode:
      name: "${device1} dc monitor mode"

switch:
  - platform: restart
    icon: mdi:reload-alert
    name: "${name} Restart"

binary_sensor:
  - platform: victron
    victron_id: victron0
    relay_state:
      name: "${device0} relay state"

syssi, 2023-03-01:
Please enable the `debug` output of the `uart` component by:

```
uart:
  id: uart0
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

And provide some logs including the mentioned issue.

canuckray, 2023-03-01:
I recall seeing some spurious values from batt voltage when I first deployed, so I added a lambda filter to the sensor value as shown below.  You could tweak your max and min values for your setup.

```
sensor:
  - platform: victron
    victron_id: victron0
    panel_voltage:
      name: "Array Voltage"
      id: pv
      accuracy_decimals: ${accuracy}
    battery_voltage:
      name: "Battery Voltage"
      id: bv
      accuracy_decimals: ${accuracy}
      filters:
        - lambda: |-
            float MIN_VALUE = 10.0;
            float MAX_VALUE = 100.0;
            if (MIN_VALUE <= x && x <= MAX_VALUE) return x;
            else return {};
```