KinDR007/VictronMPPT-ESPHOME #110: ESPHone update to uart compiling error

mynarrowboat, 2023-04-30:
Hi,
I'm not sure if this is universal now after updating ESPHome, but in case it helps someone...On trying to install the code via ESPHome from the sample code for a Victron (100/30) MPPT controller to a esp8266 d1_mini, flagged a compiling uart error on that line (two errors flagged). From another thread (https://community.home-assistant.io/t/esphome-update-and-uart2/562633/16)  changes to ESPHome and uarts used internally, uart0 can no longer be used. Changing 
"uart:
  id: uart0"  from uart0 to uart_0 and again 
"victron:
  uart_id: uart0" to uart_0 seems to fix the issue for me.

I'm using and updating ESPHome within Home Assistant. I have a Victron MPPT 100/30 controller and your code works great. Thank you so much for your work.

gabcraftia, 2023-04-30:
Hi, I also always run the latest version of esphome. Now i tried your fix because i was having the same issue. But i still get no reading from my victron controller. And i don't know what i did wrong. I connected the gnd to gnd and the TX pin of the controller to the D8 pin. Is there any way to troubleshoot it? And check if there is a connection between the 2?

Kind regards,

mynarrowboat, 2023-04-30:
> *DuckDuckGo* removed 1 tracker.
> More →

Yours compiled and uploaded, and you just don't receive data you mean? TX
from Victron to ESP Pin D7 on the sketch.

MNB

gabcraftia, 2023-05-02:
Yes that is the problem but it i think it just needs a level shift converter. from 5v to 3.3V ordered some and i will see what it does.

Garfs-Homassistant, 2023-05-02:
eAusweis

> Yes that is the problem but it i think it just needs a level shift converter. from 5v to 3.3V ordered some and i will see what it does.

Die Daten kommen auch ohne Level Shift, aber auf Dauer ist das nicht gut. Eventuell RX und TX vertauscht? Ging mir so.

gabcraftia, 2023-05-02:
Guten tag, Ich habe nur diesem video gefolgt. https://www.youtube.com/watch?v=D08sjKWBW7o&ab_channel=KPeyanski 
Ab 3:33 kansst du sehen das pin eins und drei verbinded sind das habe ich auch gemacht. eins to GND und drei to pin D7. wenn ich in die logs siehe ist da keine data von der victron. Entschuldigung fur meiner deutsch das ist nicht perfect :)

gabcraftia, 2023-05-02:
Ich habe das nur getested mit der batterie angeslossen nicht mit paneelen.

Garfs-Homassistant, 2023-05-02:
Ok, da lief es bei mir mit dem esp8266 noch fehlerfrei. Hast Du mal einen anderen esp versucht?

gabcraftia, 2023-05-02:
ja das haben ich versucht. aber kein unterschied.

closed 2023-05-02 by KinDR007