KinDR007/VictronMPPT-ESPHOME #113: missing panel current

roessi, 2023-05-18:
hello,
First of all thx for the cool project!!

I noticed something.
The Battery current is mapped as panel current in the config of the ESP.
`panel_current:
      name: "${name} battery current"`
and the battery current is missing.

I tried to remap it.
`battery_current:
      name: "${name} battery current"
panel_current:
      name: "${name} panel current"`
but the expression panel_current: is a invalid option.

Please can you help me.
i think the interface delivers the data, because the battery and panel current is displayed on the vicrton mobile phone app.

KinDR007, 2023-05-19:
Hi , mppt report panely power and voltage but not panel current 
You must calculate if you wish I=P/U

Git-ChrisSc, 2023-06-14:
Hi,

here is my solution for this:

 - platform: template # Panel_current calculation 
    name: "Panel Current"
    id: panel_current
    unit_of_measurement: "A"
    lambda: |-
     float panel_current = id(panel_power).state / id(panel_voltage).state;
     return panel_current;

closed 2023-08-14 by KinDR007