KinDR007/VictronMPPT-ESPHOME #123: Battery Temperature not pulling in

uncappedshady, 2023-07-25:
First of all I would just like to say thank you so much for releasing this code in order to pull in my stats into ESPHome for my mppt controller. I am not sure if this is how the code currently works or not but I get all the required values apart from battery temperature.

Currently I have a 75/15 mppt Victron smartsolar controller and have purchased the additional smart battery sense module in order to detect the battery voltage and temperature. 

In the Victron App I am seeing the battery voltage and temperature but in ESPhome I am not seeing the temperature value.

Is this something I would need to add to the code loaded to my D1 mini ESP8266?

Would greatly appreciate the assistance.

PolyPV81, 2023-10-22:
I have the same issue.
But on a 100/20 smartsolar.

i added:

 battery_temperature:

      name: "battery_temperature"

      id: battery_temperature

only, but it does not work.

Did i wrong, or would that be the right way?

Maybe it does not work with the mppt controllers, only with the BMV70/71?

See: https://www.victronenergy.com/upload/documents/VE.Direct-Protocol-3.33.pdf page 6?

Perfunctory85, 2024-02-11:
I have the same issue.
I connected smartsense and can read the temp in the Victronapp, but not over ESP. 

Any idea, if this will be possible?