KinDR007/VictronMPPT-ESPHOME #124: Unable to use multiple uarts

MisterX-RC, 2023-07-30:
I tried to use the example code to read multiple uarts, but the second one won’t read anything. I’m getting som error’s on the log output, bur I don’t know for sure there are related:

07:58:17][W][component:204]: Component esphome.coroutine took a long time for an operation (0.24 s).
[07:58:17][W][component:205]: Components should block for at most 20-30ms.

If I switch the first and second uart pin around I’m able to read the other pin, so the first uart alway’s works and the second declared uart fail’s, but both pin’s should be able to read data! 

I’m running the Wemos D1 mini V4.0 with the ESP 8266

syssi, 2023-07-30:
Could you provide a photo of your ESP and your `uart` configuration? I would like to make sure you haven't soldered the second UART to the wrong pin.

MisterX-RC, 2023-07-30:
I made a special adapter board for the wemos with a logic level converter for the uart’s. 

I checked this by switching the uart pin’s in software, so pin 5&7. This resulted in the software reading the other MPPT. (I got 2 MPPT’s connected) 

I’m playing with the possibility that maybe the logic level converter is the problem, but I’m reading fine from one uart, and both lanes got there own level converter, so there should be no problem there. 

I’m unable to copy my uart config ATM, but it’s a straight copy of the example from the file. 

[image]

MisterX-RC, 2023-08-06:
Well, I owe you an apology. 

In my design I installed 2 logic level shifters to be able to just plug a Victron device in and not worry about the 5V of 3.3V logic level and have it automatically shift to the 3.3V logic level of the Wemos D1 mini V4 I'm using. 

Last night I did some testing and found out that the logic level converters worked fine with the 3 MPPT 75\10's I have, but I also have 2 MPPT 100/30 and 1 75\15. 
I got 3 wemos total, all of them have a 75/10 on port 1, and another on port 2. And I don't know why, but the 100/30 and the 75/15's won't get thru. because this happened with all 3 the Wemos this caused me to believe it's a problem with the second UART. And yesterday I took the TX wire and pushed it straight onto the connector of the wemos, and Tataaa, there's the data of 2 MPPT's. 

So in the end I soldered a bypass wire straight from the connector to the Wemos for the 100/30's and the 75/15, and now all works fine!

Sorry to bother you with my no issue, and keep up the good work

closed 2023-08-06 by MisterX-RC

syssi, 2023-08-06:
Do you know the official FAQ of victron? It mentions some devices comes with a 3.3V logic level and some with 5V.

MisterX-RC, 2023-08-06:
Yes I know, that's why I implemented the logic level shifter, to be able to plug any Victron device in without being bothered to check the logic level. That was the plan, but somehow this didn't work out as planned