KinDR007/VictronMPPT-ESPHOME #125: Occasional inaccurate values

JshStadler, 2023-08-07:
I have a Victron SmartShunt connected to an ESP32 to pull the battery stats into Home Assistant.

Occasionally (usually once a day, or every few days) I get a value spikes down to exactly 9.1% SOC for the battery or 2.7V battery voltage (both doesn't happen at the same time, and the spike seems to only happen while the battery is at 91%, or 27V). This messes up the statistics a bit. 

Has anyone else experienced this? Is there a way to ignore an extreme outlier value like this?

syssi, 2023-08-07:
You could apply a `quantile` filter to the affected sensors: https://esphome.io/components/sensor/index.html

JshStadler, 2023-08-07:
@syssi 
Thanks, quantile looks promising, I've setup the filter to ignore the specific values for now (as this should cause the least amount of overhead/delay), if I see any more spikes, I'll setup the quantile filter.

closed 2023-08-07 by JshStadler