KinDR007/VictronMPPT-ESPHOME #131: MPPT 100/20 wrong entity class

lucasimons, 2023-11-04:
This 3 sensor : total energy prpduction, yesterday production and today production have wrong device class ....they have power class instead of energy

````
Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:691
Integration: Sensore ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 3 novembre 2023 alle ore 22:09:15 (3 occurrences)
Last logged: 3 novembre 2023 alle ore 23:55:38

Entity sensor.energia_totale_victron (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'kWh' which is not a valid unit for the device class ('power') it is using; expected one of ['kW', 'W']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
Entity sensor.energia_fv2_ieri (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'kWh' which is not a valid unit for the device class ('power') it is using; expected one of ['kW', 'W']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
Entity sensor.energia_fv2_oggi (<class 'homeassistant.components.esphome.sensor.EsphomeSensor'>) is using native unit of measurement 'kWh' which is not a valid unit for the device class ('power') it is using; expected one of ['kW', 'W']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+esphome%22
```

KinDR007, 2023-11-06:
> This 3 sensor : total energy prpduction, yesterday production and today production have wrong device class ....they have power class instead of energy
>
>
>
> ````
>
> Logger: [...]

Hi,

Can you please send me your full sketch ?

lucasimons, 2023-11-06:
```
esphome:
  name: sottoscala-esp8266-1
  friendly_name: sottoscala-esp8266-1

esp8266:
  board: d1_mini
  
external_components:

#  - source: github://syssi/esphome-pipsolar@2424mse1
#   refresh: 0s
 - source: github://KinDR007/VictronMPPT-ESPHOME@main
   refresh: 0s
 - source: github://syssi/esphome-ant-bms@main
   refresh: 0s

# Enable logging
logger:
  baud_rate: 0

    
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sottoscala-Esp8266-1"
    password: "TfcwMNR8M5jZ"

captive_portal:

uart:
#victron
- id: uart_bus
  tx_pin: D8 # not connected !! 
  rx_pin: D7 # d7 use olny this pin and GND to MPPT chrager
  baud_rate: 19200 
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_buffer_size: 256
 #ant bms 
- id: uart_antbms
  baud_rate: 19200
  rx_buffer_size: 384
  tx_pin: GPIO4
  rx_pin: GPIO5

ant_bms:
  - id: bms0
    uart_id: uart_antbms
    password: "12345678"
    update_interval: 5s
    supports_new_commands: false

  
victron:
  id: victron0
  uart_id: uart_bus  
  throttle: 6s

sensor:
  - platform: victron
    victron_id: victron0
    panel_voltage:
      name: "Tensione Fotovoltaico 2" 
      id: pv

    battery_voltage:
      name: "Batteria Tensione"  
      id: bv

    battery_current:
      name: "Batteria Corrente" 
      id: bc
      
    load_current:
      name: "Corrente carichi DC" 
      id: load_current
      filters:
       - sliding_window_moving_average:
          window_size: 4
          send_every: 4
    max_power_yesterday:  
      name: "Massima potenza Ieri" 
      id: max_power_yesterday
     
    max_power_today:  
      name: "Massima potenza Oggi" 
      id: max_power_today
      accuracy_decimals: 2
       
    yield_total:
      name: "Energia Totale Victron"   
      id: yield_total
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001

      unit_of_measurement: kWh
      accuracy_decimals: 2
    yield_yesterday:
      name: "Energia FV2 Ieri"
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001

      unit_of_measurement: kWh
      accuracy_decimals: 2
    yield_today:
      name: "Energia FV2 Oggi"  
      id: yt
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
      
      unit_of_measurement: kWh
      accuracy_decimals: 2
    panel_power:
      name: "Potenza Fotovoltaico 2"  
      id: panel_power
 
    day_number:
      name: "Day sequence number"  
      id: day_number
      icon: "mdi:calendar-today"

      
  - platform: ant_bms
    current:
      name: "bms1 current"
      
    soc:
      name: "bms1 soc"
      
    capacity_remaining:
      name: "bms1 capacity remaining"
      
    battery_cycle_capacity:
      name: "bms1 battery cycle capacity"
      
    total_voltage:
      name: "bms1 total voltage"
      
    total_runtime:
      name: "bms1 total runtime"
    average_cell_voltage:
      name: "bms1 average cell voltage"
      
    power:
      name: "bms1 power"
     
    min_cell_voltage:
      name: "bms1 min cell voltage"
      
    max_cell_voltage:
      name: "bms1 max cell voltage"
      
    min_voltage_cell:
      name: "bms1 min voltage cell"
      
    max_voltage_cell:
      name: "bms1 max voltage cell"
      
    temperature_1:
      name: "bms1 temperature 1"
      
    temperature_2:
      name: "bms1 temperature 2"
      
    temperature_3:
      name: "bms1 temperature 3"
      
    temperature_4:
      name: "bms1 temperature 4"
      
    cell_voltage_1:
      name: "bms1 cell voltage 1"
      
    cell_voltage_2:
      name: "bms1 cell voltage 2"
      
    cell_voltage_3:
      name: "bms1 cell voltage 3" 
      
    cell_voltage_4:
      name: "bms1 cell voltage 4"
      
    cell_voltage_5:
      name: "bms1 cell voltage 5"
      
    cell_voltage_6:
      name: "bms1 cell voltage 6"
      
    cell_voltage_7:
      name: "bms1 cell voltage 7"
      
    cell_voltage_8:
      name: "bms1 cell voltage 8"
      
    charge_mosfet_status_code:
      name: "bms1 charge mosfet status code"
    discharge_mosfet_status_code:
      name: "bms1 discharge mosfet status code"
    balancer_status_code:
      name: "bms1 balancer status code"

  - platform: wifi_signal
    name: "WiFi Signal Sensor Sottoscala 2"
    update_interval: 5s

  - platform: uptime
    name: "Sottoscala 2 ESP32 Uptime"
    id: uptime_s
    update_interval: 5s    
text_sensor:
 # - platform: victron
 #   victron_id: victron0
 #   charging_mode:
 #     name: "Charging Mode"
 #   error:
 #     name: "Error"
 #   tracking_mode:
 #     name: "Tracking Mode"

  - platform: ant_bms
    charge_mosfet_status:
      name: "bms1 charge mosfet status"
    discharge_mosfet_status:
      name: "bms1 discharge mosfet status"
    balancer_status:
      name: "bms1 balancer status"
    total_runtime_formatted:
      name: "bms1 total runtime formatted"

switch:

  - platform: ant_bms
    charging:
      name: "bms1 charging"
    discharging:
      name: "bms1 discharging"

    
 
        
  - platform: gpio
    name: "Rel1"
    pin: GPIO14
    

   
binary_sensor:
    
  - platform: gpio
    pin: GPIO0
    name: "Botola"
    device_class: door

    

```

PR #141 (2023-11-25): Fix device class of energy sensors (Closes: #131)
Closes: #131
closed 2023-11-25 by syssi