KinDR007/VictronMPPT-ESPHOME #132: Blocking Esphome

swifty99, 2023-11-05:
Hi,

I like this component, however there are timing issues:
```
[21:53:00][W][component:214]: Component esphome.coroutine took a long time for an operation (0.22 s).
[21:53:00][W][component:215]: Components should block for at most 20-30ms.
```

I tracked this down to the victron platform. Excluding it stops giving these warnings.
```
esphome version
Version: 2023.10.6
```
on a ESP32:
```
esp32:
  board: az-delivery-devkit-v4
  #board:   nodemcu-32s
  framework:
    type: arduino
```

can you reproduce this?

KinDR007, 2023-11-06:
> Hi,
> 
> 
> 
> I like this component, however there are timing issues:
> 
> ```
> 
> [21:53:00][W][component:214]: Component esphome.coroutine took a long time for an operation (0.22 s).
> 
> [21:53:00][W][component:215]: Components should block for at most 20-30ms.
> 
> ```
> 
> 
> 
> I tracked this down to the victron platform. Excluding it stops giving these warnings.
> 
> ```
> 
> esphome version
> 
> Version: 2023.10.6
> 
> ```
> 
> on a ESP32:
> 
> ```
> 
> esp32:
> 
>   board: az-delivery-devkit-v4
> 
>   #board:   nodemcu-32s
> 
>   framework:
> 
>     type: arduino
> 
> ```
> 
> 
> 
> when disconnectin the serial, the warnings disappear.
> 
> Looking at the code this might be a problem:
> 
> ```
> 
>   last_transmission_ = now;
> 
>   while (available()) {
> 
>     uint8_t c;
> 
>     read_byte(&c);
> 
>     if (state_ == 0) {
> 
> ```
> 
> while available could take some time. 
> 
> 
> 
> more information here:
> 
> https://github.com/esphome/issues/issues/4717
> 
> 
> 
> maybe reading the buffer state could help:
> 
> https://arduino.stackexchange.com/questions/9666/how-to-avoid-blocking-while-loop-reading-serial
> 
> 
> 
> cheers
> 
> Jan

Hi Jan,

Can you please send me your full sketch ?

swifty99, 2023-11-06:
sure.

```
substitutions:
  plug_name: victron
  ip_adress: "192.168.178.170"
  name: "victron"

esphome:
  name: ${plug_name}
 # esp8266_restore_from_flash: true

# device_base.yaml
preferences:
  flash_write_interval: 10min
# covers generic basics

wifi:
  networks:
  - ssid: "Bit4Thing_wz"
    password: !secret wifi_pw_home
  - ssid: "Bit_Pumpe"
    password: !secret wifi_pw_home 
  - ssid: "Bit4Thing"
    password: !secret wifi_pw_4thing 
  - ssid: "BitMobil"
    password: !secret wifi_password_mobile 

  ap:
    ssid: "${plug_name}.a"
    password: !secret web_server_password

logger:
  level: DEBUG
  logs:
    adc: ERROR

api:
  reboot_timeout: 0h

ota:

external_components:
  - source: github://KinDR007/VictronMPPT-ESPHOME@main

web_server:
  port: 80

esp32:
  board: az-delivery-devkit-v4
  #board:   nodemcu-32s
  framework:
    type: arduino

uart:
  id: uart0v
  rx_pin: 33 # Connect this this GPIO and GND to the MPPT charger
  baud_rate: 19200
  rx_buffer_size: 256

victron:
  uart_id: uart0v
  id: victron0
  throttle: 10s

binary_sensor:
  - platform: victron
    victron_id: victron0
    load_state:
      name: "${name} load state"
[... 221 lines cut ...]
[18:10:08][W][component:214]: Component esphome.coroutine took a long time for an operation (0.18 s).
[18:10:08][W][component:215]: Components should block for at most 20-30ms.
[18:10:19][D][sensor:094]: 'victron battery voltage': Sending state 26.84000 V with 3 decimals of accuracy
[18:10:19][D][sensor:094]: 'victron battery current': Sending state -0.02000 A with 3 decimals of accuracy
[18:10:19][D][sensor:094]: 'victron panel voltage': Sending state 0.01000 V with 3 decimals of accuracy
[18:10:19][D][sensor:094]: 'victron panel power': Sending state 0.00000 W with 0 decimals of accuracy
[18:10:19][D][sensor:094]: 'victron charging mode id': Sending state 0.00000  with 0 decimals of accuracy
[18:10:19][D][text_sensor:064]: 'victron charging mode': Sending state 'Off'
[18:10:19][D][sensor:094]: 'victron tracking mode id': Sending state 0.00000  with 0 decimals of accuracy
[18:10:19][D][text_sensor:064]: 'victron tracking mode': Sending state 'Off'
[18:10:19][D][sensor:094]: 'victron error code': Sending state 0.00000  with 0 decimals of accuracy
[18:10:19][D][text_sensor:064]: 'victron error': Sending state 'No error'
[18:10:19][D][sensor:094]: 'victron load current': Sending state 0.00000 A with 3 decimals of accuracy
[18:10:19][D][sensor:094]: 'victron yield total': Sending state 15250.00000 Wh with 0 decimals of accuracy
[18:10:19][D][sensor:094]: 'victron yield today': Sending state 0.00000 Wh with 0 decimals of accuracy
[18:10:19][D][sensor:094]: 'victron max power today': Sending state 0.00000 W with 0 decimals of accuracy
[18:10:19][D][sensor:094]: 'victron yield yesterday': Sending state 610.00000 Wh with 0 decimals of accuracy
[18:10:19][D][sensor:094]: 'victron max power yesterday': Sending state 118.00000 W with 0 decimals of accuracy
[18:10:19][D][sensor:094]: 'victron day number': Sending state 134.00000  with 0 decimals of accuracy
[18:10:19][W][component:214]: Component esphome.coroutine took a long time for an operation (0.17 s).
[18:10:19][W][component:215]: Components should block for at most 20-30ms.
[18:10:30][D][sensor:094]: 'victron battery voltage': Sending state 26.84000 V with 3 decimals of accuracy
[18:10:30][D][sensor:094]: 'victron battery current': Sending state -0.02000 A with 3

```

lucasimons, 2023-11-07:
I have a similar issue, when i have installed esphome 10.6 the victron component report this warning, it si strange because sometimes it works fine but for exemple today, most the time not works... I have tried to reboot but it not works

```
[22:24:43][W][component:215]: Components should block for at most 20-30ms.
[22:24:44][W][victron:093]: Last transmission too long ago
```

[image]

this is my config:
```
esphome:
  name: sottoscala-esp8266-1
  friendly_name: sottoscala-esp8266-1

esp8266:
  board: d1_mini
  
external_components:

#  - source: github://syssi/esphome-pipsolar@2424mse1
#   refresh: 0s
 - source: github://KinDR007/VictronMPPT-ESPHOME@main
   refresh: 0s
 - source: github://syssi/esphome-ant-bms@main
   refresh: 0s

# Enable logging
logger:
  baud_rate: 0

uart:
#victron
- id: uart_bus
  tx_pin: D8 # not connected !! 
  rx_pin: D7 # d7 use olny this pin and GND to MPPT chrager
  baud_rate: 19200 
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_buffer_size: 256
 #ant bms 
- id: uart_antbms
  baud_rate: 19200
  rx_buffer_size: 384
  tx_pin: GPIO4
  rx_pin: GPIO5

ant_bms:
  - id: bms0
    uart_id: uart_antbms
    password: "12345678"
    update_interval: 5s
    supports_new_commands: false

  
victron:
  id: victron0
  uart_id: uart_bus  
  throttle: 6s

sensor:
  - platform: victron
    victron_id: victron0
    panel_voltage:
      name: "Tensione Fotovoltaico 2" 
      id: pv

    battery_voltage:
      name: "Batteria Tensione"  
      id: bv

    battery_current:
      name: "Batteria Corrente" 
      id: bc
      
    load_current:
      name: "Corrente carichi DC" 
      id: load_current
      filters:
       - sliding_window_moving_average:
          window_size: 4
          send_every: 4
    max_power_yesterday:  
      name: "Massima potenza Ieri" 
      id: max_power_yesterday
     
    max_power_today:  
      name: "Massima potenza Oggi" 
      id: max_power_today
      accuracy_decimals: 2
       
    yield_total:
      name: "Energia Totale Victron"   
      id: yield_total
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001

      unit_of_measurement: kWh
      accuracy_decimals: 2
    yield_yesterday:
      name: "Energia FV2 Ieri"
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001

      unit_of_measurement: kWh
      accuracy_decimals: 2
    yield_today:
      name: "Energia FV2 Oggi"  
      id: yt
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
      
      unit_of_measurement: kWh
      accuracy_decimals: 2
    panel_power:
      name: "Potenza Fotovoltaico 2"  
      id: panel_power
 
    day_number:
      name: "Day sequence number"  
      id: day_number
      icon: "mdi:calendar-today"

      
  - platform: ant_bms
    current:
      name: "bms1 current"
      
    soc:
      name: "bms1 soc"
      
    capacity_remaining:
      name: "bms1 capacity remaining"
      
    battery_cycle_capacity:
      name: "bms1 battery cycle capacity"
      
    total_voltage:
      name: "bms1 total voltage"
      
    total_runtime:
      name: "bms1 total runtime"
    average_cell_voltage:
      name: "bms1 average cell voltage"
      
    power:
      name: "bms1 power"
     
    min_cell_voltage:
      name: "bms1 min cell voltage"
      
    max_cell_voltage:
      name: "bms1 max cell voltage"
      
    min_voltage_cell:
      name: "bms1 min voltage cell"
      
    max_voltage_cell:
      name: "bms1 max voltage cell"
      
    temperature_1:
      name: "bms1 temperature 1"
      
    temperature_2:
      name: "bms1 temperature 2"
      
    temperature_3:
      name: "bms1 temperature 3"
      
    temperature_4:
      name: "bms1 temperature 4"
      
    cell_voltage_1:
      name: "bms1 cell voltage 1"
      
    cell_voltage_2:
      name: "bms1 cell voltage 2"
      
    cell_voltage_3:
      name: "bms1 cell voltage 3" 
      
    cell_voltage_4:
      name: "bms1 cell voltage 4"
      
    cell_voltage_5:
      name: "bms1 cell voltage 5"
      
    cell_voltage_6:
      name: "bms1 cell voltage 6"
      
    cell_voltage_7:
      name: "bms1 cell voltage 7"
      
    cell_voltage_8:
      name: "bms1 cell voltage 8"
      
    charge_mosfet_status_code:
      name: "bms1 charge mosfet status code"
    discharge_mosfet_status_code:
      name: "bms1 discharge mosfet status code"
    balancer_status_code:
      name: "bms1 balancer status code"

  - platform: wifi_signal
    name: "WiFi Signal Sensor Sottoscala 2"
    update_interval: 5s

  - platform: uptime
    name: "Sottoscala 2 ESP32 Uptime"
    id: uptime_s
    update_interval: 5s    
text_sensor:
 # - platform: victron
 #   victron_id: victron0
 #   charging_mode:
 #     name: "Charging Mode"
 #   error:
 #     name: "Error"
 #   tracking_mode:
 #     name: "Tracking Mode"

  - platform: ant_bms
    charge_mosfet_status:
      name: "bms1 charge mosfet status"
    discharge_mosfet_status:
      name: "bms1 discharge mosfet status"
    balancer_status:
      name: "bms1 balancer status"
    total_runtime_formatted:
      name: "bms1 total runtime formatted"

switch:

  - platform: ant_bms
    charging:
      name: "bms1 charging"
    discharging:
      name: "bms1 discharging"

    
 
        
  - platform: gpio
    name: "Rel1"
    pin: GPIO14
    
  - platform: restart
    name: "sottoscala-esp8266-1 Restart"
   
binary_sensor:
    
  - platform: gpio
    pin: GPIO0
    name: "Botola"
    device_class: door

    
```

swifty99, 2023-11-11:
one more remark,

there is quite a bit of a timeout until available() gets false.
Chatgpt recommended this approach, which seems ok:

```
void setup() {
  Serial.begin(9600); // or the desired baud rate
}

void loop() {
  // Your other loop activities can go here

  // Check if there are data available in the serial buffer
  if (Serial.available() > 0) {
    // Read data incrementally without blocking the loop
    while (Serial.available() > 0) {
      char c = Serial.read();
      // Process the read character as needed
      // Here, you might store the data in an array or process it in some way
    }

    // Add additional activities after reading the data here
  }

  // Your other loop activities can go here
}
```

This code uses Serial.available() to check if there is data available in the serial buffer. If data is available, it enters a loop to read the data incrementally without blocking the loop() function. 

rebduilding the whole string will need to be done after no data is reveived for a while.

PR #133 (2023-11-18): stop blocking ESPHome
Updated to rely on the ESPHome event loop instead of waiting to read the entire serial buffer at a time.

It seems to work in the sense that it no longer throws warnings about blocking esphome too long, however now it randomly throws errors like: `[D][victron:1024]: Unhandled property: HS 0`. which seems like something is getting corrupted, which wasn't before. Maybe increasing the uart buffer would help.

fixes #132
closed 2023-11-18 by KinDR007