KinDR007/VictronMPPT-ESPHOME #134: Erroneous readings

igorparsadanov, 2023-11-19:
I have had Victron integration working perfectly for last few months.  But noticed that recently (can't pinpoint which version, but currently running ESPHome 2023.11.2) I started getting erroneous readings.  Such as battery voltage being 0.0V, when its not in reality and checking via bluetooth app the battery is reading correctly.  When looking at the logs I get unhandled property errors such as the one listed here:

[13:06:37][D][victron:1024]: Unhandled property: VPV
H19 3515
[13:08:13][D][victron:1024]: Unhandled property: CVPV 29170

I thought maybe my ESP8266 is misbehaving so I tried another but same results.  I have two different Victron MPPT controllers and both appear to be doing this.  Did something in new ESPHome versions change and is breaking this now?

lucasimons, 2023-11-19:
I have a similar issue 
https://github.com/KinDR007/VictronMPPT-ESPHOME/issues/132#issuecomment-1800182212

The load power output of victorn mppt 100/20 works only when the sun is available 

[image]

This is before the update and it works fine

[image]

joevk5ei, 2023-11-20:
Im also getting similar errors since updating to Firmware: 2023.11.2

jsp2021, 2023-11-20:
Same problems, everything works as it should since "Firmware: 2023.11.2"

syssi, 2023-11-20:
@KinDR007 could you please tag the commit `cb90f577bf68a071f6dadf20bbe854392ff1c7f5` as version `1.0.0`:

```
git tag 1.0.0 cb90f577bf68a071f6dadf20bbe854392ff1c7f5
git push --tags
```

It looks like the new async serial handling doesn't work properly and the people should be able to go back to the previous implementation using the version 1.0.0 tag:

```
substitutions:
  name: victron-smartshunt
  external_components_source: github://KinDR007/VictronMPPT-ESPHOME@1.0.0
```

syssi, 2023-11-20:
IMO this issue isn't ESPHome core related. This project / implementation was modified recently.

wozz, 2023-11-20:
likely due to #133 - I like the idea of tagging commits with versions to allow for testing before wider release. alternatively, that commit could just be reverted

I have a branch that I think fixes the corrupted values: https://github.com/wozz/VictronMPPT-ESPHOME/tree/validate_checksum

however, I'm not really sure what the root cause is - my latest change just validates the checksum before publishing the values - which should catch the corrupted values, but I'm not sure why they're corrupted to begin with, when they weren't previously. So there might be a bigger issue.

wozz, 2023-11-20:
> I have a similar issue [#132 (comment)](https://github.com/KinDR007/VictronMPPT-ESPHOME/issues/132#issuecomment-1800182212)
> 
> The load power output of victorn mppt 100/20 works only when the sun is available
> 
> [image]
> 
> This is before the update and it works fine
> 
> [image]

this might be a separate issue? these graphs show your issue occurring prior to #133 being merged

lucasimons, 2023-11-20:
> this might be a separate issue? these graphs show your issue occurring prior to #133 being merged

This is my log file

```
[14:33:21][D][victron:1024]: Unhandled property: \x8fHSDS 86
[14:33:21][D][victron:1024]: Unhandled property: \x8eChecksum\x8d\xf2
PID 0xA060
[14:33:22][D][victron:1024]: Unhandled property: \xe819 1\xbb2212
[14:33:22][D][victron:1024]: Unhandled property: H2\xb0 24
```

[logs_sottoscala-esp8266-1_logs (1).txt](https://github.com/KinDR007/VictronMPPT-ESPHOME/files/13415403/logs_sottoscala-esp8266-1_logs.1.txt)

wozz, 2023-11-20:
> This is my log file
>
> ```
> [14:33:21][D][victron:1024]: Unhandled property: \x8fHSDS 86
> [14:33:21][D][victron:1024]: Unhandled property: \x8eChecksum\x8d\xf2
> PID 0xA060
> [14:33:22][D][victron:1024]: [...]

The unhandled property issue is the same - but your issue where load values disappear overnight seems different (and started prior to #133 being merged)

lucasimons, 2023-11-20:
I started having problems at the beginning of November I think it was version 10.6 of esphome, I don't know what changes were made. However, as soon as it's ready I can test this github://KinDR007/VictronMPPT-ESPHOME@1.0.0

wozz, 2023-11-20:
I assume you can revert to previous state even without a tag by referencing the commit directly:

`github://KinDR007/VictronMPPT-ESPHOME@cb90f577bf68a071f6dadf20bbe854392ff1c7f5`

syssi, 2023-11-20:
Only branches or tags are supported here: https://esphome.io/components/external_components.html#example-of-git-repositories

igorparsadanov, 2023-11-20:
Just installed this branch: https://github.com/wozz/VictronMPPT-ESPHOME/tree/validate_checksum

Looks like now every update has invalid checksum and doesn't make it into the database
[12:59:51][W][victron:155]: Received invalid checksum, dropping frame: recv 129, calc 221

igorparsadanov, 2023-11-20:
For now I installed from this fork that appears to be the last commit maybe?  Its back to working:
external_components_source: github://syssi/esphome-victron-vedirect

wozz, 2023-11-20:
I've been running my branch since early this morning and I have had a few instances of mismatched checksums, but definitely not every message. I've also had no erroneous/corrupted values reported (the corrupted frames are properly dropped).

What buffer size do you have set for your uart config? And which victron device are you using?

igorparsadanov, 2023-11-20:
My Uart buffer is set to 256 and I'm running MPPT 100/20 48V.  Every frame appeared to be corrupt and then dropped due to bad checksum.

joevk5ei, 2023-11-21:
> For now I installed from this fork that appears to be the last commit maybe? Its back to working: external_components_source: github://syssi/esphome-victron-vedirect

That fork is also working for me with no errors, thanks

closed 2023-11-21 by syssi

lucasimons, 2023-11-21:
I still have the same issue ...

I have tryed this two branchs:

github://KinDR007/VictronMPPT-ESPHOME@main
github://syssi/esphome-victron-vedirect

[logs_sottoscala-esp8266-1_logs (2).txt](https://github.com/KinDR007/VictronMPPT-ESPHOME/files/13428782/logs_sottoscala-esp8266-1_logs.2.txt)

wozz, 2023-11-21:
I don't see any issues in your logs

lucasimons, 2023-11-21:
I see this error 
the last trasmission is too long

PR #138 (2023-11-21): Revert "update (#133)"
This reverts commit 3d6d2457afe66bb5e74362a1f83fad5bc1e9b243.

Reverts #133
Fixes #134
Reopens #132

Supersedes #135