KinDR007/VictronMPPT-ESPHOME #140: MPPT 100/20 strange issue no solar

lucasimons, 2023-11-24:
Hi, I updated to the latest release but I continue to have this problem or when there is no sun I have this error, 

```
[15:03:25][D][victron:1026]: Unhandled property: PRt\xe3\xc160H\xc8U(\xb2\x8a\x8fSER#B*ɱ06V\xe6V\xd5UHȕ26\xbf10
\xfb\xe3K\xa6\x82j
V\xf8V 39680
\x8fERR \xb8
[15:03:25][D][victron:1026]: Unhandled property: L*%O\xef
I\xec 140\xb8\xec\x8c?\xcaJ12\xbb770
[15:03:25][D][victron:1026]: Unhandled property: \x8fH20B7\xa2\x8fH21 436
```
in fact during the night and in the morning all the Victron sensors don't work then when the sun is high enough in the sky it starts to go up until it descends under the mountains (blue arrow) and the error appears

[image]

[logs_sottoscala-esp8266-1_logs (3).txt](https://github.com/KinDR007/VictronMPPT-ESPHOME/files/13459932/logs_sottoscala-esp8266-1_logs.3.txt)

https://github.com/KinDR007/VictronMPPT-ESPHOME/issues/134#issuecomment-1817959026

closed 2023-12-12 by lucasimons

syssi, 2023-12-12:
How did you resolve the issue?

lucasimons, 2023-12-12:
> How did you resolve the issue?

I have changed the board from esp8266 d1mini supply by a dc dc isolated converter, to a esp32 UNO board supply by 12v so i removed the gnd cable to the victron MPPT, now i use only the tx pin .....   I had used this second configuration since 2019 to 2022,  but it start to disconnect from wifi i replace the esp32 board but it doesn t work (now i have found the mistake, i have 3 of 6 esp32 UNO board, bought from Ali,  that have a very bad wifi module, it works fine when i do the test in my lab, but when i connected it to the real Victron MPPT it doen t works because the wifi strength wasn t enough ), so i switched to  d1mini at the start of 2022 and now i come back to the esp32 UNO board with good wifi module and it works fine but It drove me crazy :)