KinDR007/VictronMPPT-ESPHOME #142: 2 MPPT 2 UART on ESPS2mini

cogito44, 2023-12-10:
Hello,
I made a program with two UART inputs for two MPPTs, but when I compile the program, I have no error, but once it arrives on the ESP, it no longer communicates in Wi-Fi.

I have to flash it with the cable with only one UART version.

```
uart:
  - id: uart_0
    tx_pin: GPIO2  # Not used! The communication is read-only
    rx_pin: GPIO1
    baud_rate: 19200
    rx_buffer_size: 256

  - id: uart_1
    tx_pin: GPIO6  # Not used! The communication is read-only
    rx_pin: GPIO4
    baud_rate: 19200
    rx_buffer_size: 256

victron:
  - id: victron0
    uart_id: uart_0

  - id: victron1
    uart_id: uart_1

sensor:
  - platform: victron
    victron_id: victron0
    panel_voltage:
      name: "Votage PV"
      id: Votage_PV
    battery_voltage:
      name: "Voltage batterie"
      id: Voltage_batterie
    battery_current:
      name: "Intensité Batterie"
      id : intensite_batterie
    panel_power :
      name: "Puissance"
      id: Puissance_MPPT
    yield_total:
      name: "Production totale"
      id: Production_totale
    yield_today:
      name: "Production d'aujourd'hui"
      id: Production_aujourd_hui
    yield_yesterday:
      name: "Production d'hier"
      id: Production_hier

  - platform: victron
    victron_id: victron1
    panel_voltage:
      name: "Votage PV 2"
      id: Votage_PV1
    battery_voltage:
      name: "Voltage batterie 2"
      id: Voltage_batterie1
    battery_current:
      name: "Intensité Batterie 2"
      id : intensite_batterie1
    panel_power :
      name: "Puissance"
      id: Puissance_MPPT1
    yield_total:
      name: "Production totale 2"
      id: Production_totale1
    yield_today:
      name: "Production d'aujourd'hui 2"
      id: Production_aujourd_hui1
    yield_yesterday:
      name: "Production d'hier 2"
      id: Production_hier1

text_sensor:
    - platform: victron
      victron_id: victron0
      charging_mode:
        name: "charging_mode"
        id: charging_mode
      error:
        name: "Error MPPT"
        id: error
        

    - platform: victron
      victron_id: victron1
      charging_mode:
        name: "charging_mode 2"
        id: charging_mode1
      error:
        name: "Error MPPT 2"
        id: error1
```

It works very well, with a single MPPT.

Is it possible to have two UART configurations?
Thank you for your excellent work.

zechnkaas, 2024-01-01:
Hi

Yes I do have multiple victron devices on one esp. I think you used the wrong gpio pins. Google for "Esp32S2mini pinout". Try as pins the following:
GPIO17,16,15,27

rs3hc, 2024-01-02:
Hey guys

I´m up to read out 2 MPPT 100/20 with one ESP32 Wroom.
How must i wire it up?
Are the VE.direct wires wired up parallel to the same pin on the ESP or must i use two seperate pins??

@KinDR007 its really great to see this tool, but a wiring image for two devices would make the Infos quite easier.
Also the values **uart_id** and **victron_id** are (for me) not clear.... which mppt Tracker is 1 or 2 or 0 ??

Can you please add a few more lines of explanation?

syssi, 2024-01-02:
One GPIO / RX line / `uart` component per device is required.

rs3hc, 2024-01-02:
> One GPIO / RX line / `uart` component per device is required.

ok i will try on friday :+1:

KinDR007, 2024-01-03:
> Hello, I made a program with two UART inputs for two MPPTs, but when I compile the program, I have no error, but once it arrives on the ESP, it no longer communicates in Wi-Fi.
> 
> I have to flash it with the cable with only one UART version.
> 
> ```
> uart:
>   - id: uart_0
>     tx_pin: GPIO2  # Not used! The communication is read-only
>     rx_pin: GPIO1
>     baud_rate: 19200
>     rx_buffer_size: 256
>  ..........
> ```
> 
> It works very well, with a single MPPT.
> 
> Is it possible to have two UART configurations? Thank you for your excellent work.

Hi ,
here is example for two uarts 
[smartsolar-mppt-esp8266-example-multiple-uarts.yaml](https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/smartsolar-mppt-esp8266-example-multiple-uarts.yaml)

The wiring is the same as for two individual devices, only the pins are different for each UART interface. I'm not sure now for ESP32 S2 is already supported in ESPHOME without additional settings .. see here [esp-idf-framework](https://esphome.io/components/esp32.html#esp-idf-framework)

rs3hc, 2024-01-03:
> Hi , here is example for two uarts [...]

Thanks for your information.
I use an Esp32 Wroom-E or ESP8266, which is more like the standard one.
Does it matter which pins you use or are there special ones for use with UART?

syssi, 2024-01-03:
The ESP32 is equipped with a GPIO matrix and supports three hardware UARTs. Because of the GPIO matrix the hardware UARTs can be assigned to (nearly) any GPIO.

The ESP8266 supports a single hardware UART on some specific GPIOs (GPIO1/GPIO3 is used in general). There is no GPIO matrix here so the hardware features of the GPIOs cannot be changed. If you assign the `uart` component to GPIOs not supporting a hardware  UART a software implementation called SoftwareSerial ist used to emulate a UART. Missing some bytes every now and then is pretty common/normal here. You should prefer hardware UARTs whenever possible.

rs3hc, 2024-01-03:
@KinDR007  
> Hi , here is example for two uarts [...]

I got a ESP8266-12E so the wiring should be as follows with reference to your example config?

Can you confirm this?

[image]

KinDR007, 2024-01-03:
I can confirm this but you need to test if everything works as it should.
The power supply is implemented from what source? Will the battery connected to the mppt be used? Or is the power supply separate?

Přeloženo pomocí DeepL (https://www.deepl.com/app/?utm_source=ios&utm_medium=app&utm_campaign=share-translation)

rs3hc, 2024-01-03:
> I can confirm this but you need to test if everything works as it should. The power supply is implemented from what source? Will the battery connected to the mppt be used? Or is the power supply separate?

I think I will use a normal AC USB charger.

But maybe I can use a 60v (20-70V) to 5V stepdown to use the battery as supply.

KinDR007, 2024-01-04:
If you use stepdown with battery connected to mppt so do not connect gnd between esp and ve.direct port .connect only tx pin on ve.direct to rx pin of esp nothing more

rs3hc, 2024-01-19:
finally i got it working with a esp8266.

Great job man

closed 2024-01-20 by KinDR007