KinDR007/VictronMPPT-ESPHOME #143: ERROR: victron:093 BAD_INDICATOR errno=11

Miaoucat0, 2024-01-01:
hello.
i am trying to make this work and apparently no matter what i do it just doesnt. tried with and without the ground wire connected to the ESP32. i tried to use the RX00 pin and RX02... i saw a couple of comments on this YT video (watch?v=D08sjKWBW7o) mentioning the same problem.

if somebody could please help me with this problem.

[HARDWARE]
-ESP32 with 30pins
-Victron MPPT 75-15
-wires and voltage divider using 1K in series with the TX pin, and a 2K resistor to ground
GND connected to both Victron and ESP32 board. TX pin from Victron is connected to pin 3(RX) of ESP
-ESP32 powered thru USB with a "cigar lighter" type power supply on the solar system.

esphome log screenshot:
[image]
2nd source for image: https://i.imgur.com/qyW0yBc.jpg

my YAML file:

`substitutions:
  name: victron-mppt
  external_components_source: github://KinDR007/VictronMPPT-ESPHOME@main

esphome:
  name: ${name}
  friendly_name: esp32-victron

external_components:
  - source: ${external_components_source}
    refresh: 0s

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "REDACTED"

ota:

web_server:
  port: 80

#mqtt:
#  broker: 
#  username: 
#  password: 
#  id: mqtt_client

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esp32-test"
    password: "44444444"

captive_portal:

uart:
  id: uart_0
  tx_pin: 1  # Not connected! The communication is read-only
  rx_pin: 3  # Connect this this GPIO and GND to the MPPT charger
  baud_rate: 19200
  rx_buffer_size: 256

victron:
  uart_id: uart_0
  id: victron0
  throttle: 10s

sensor:
  - platform: victron
    victron_id: victron0
    max_power_yesterday:
      name: "${name} max power yesterday"
    max_power_today:
      name: "${name} max power today"
    yield_total:
      name: "${name} yield total"
    yield_yesterday:
      name: "${name} yield yesterday"
    yield_today:
      name: "${name} yield today"
    panel_voltage:
      name: "${name} panel voltage"
    panel_power:
      name: "${name} panel power"
    battery_current:
      name: "${name} panel current"
    battery_voltage:
      name: "${name} battery voltage"
    day_number:
      name: "${name} day number"
    charging_mode_id:
      name: "${name} charging mode id"
    error_code:
      name: "${name} error code"
    tracking_mode_id:
      name: "${name} tracking mode id"
    load_current:
      name: "${name} load current"

text_sensor:
  - platform: victron
    victron_id: victron0
    charging_mode:
      name: "${name} charging mode"
    error:
      name: "${name} error"
    tracking_mode:
      name: "${name} tracking mode"
    firmware_version:
      name: "${name} firmware version"
    device_type:
      name: "${name} device type"
    serial_number:
      name: "${name} serial number"

binary_sensor:
  - platform: victron
    victron_id: victron0
    load_state:
      name: "${name} load state"
    relay_state:
      name: "${name} relay state"`

board used:
https://i.imgur.com/mNFT1L0.jpeg

here is the special DEBUG firmware LOG:

'INFO Upload took 5.24 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.0.42 using esphome API
INFO Successfully connected to victron-debug @ 192.168.0.42 in 7.216s
INFO Successful handshake with victron-debug @ 192.168.0.42 in 0.035s
[13:51:42][I][app:102]: ESPHome version 2023.12.5 compiled on Jan  1 2024, 13:47:46
[13:51:42][C][wifi:573]: WiFi:
[13:51:42][C][wifi:405]:   Local MAC: REDACTED
[13:51:42][C][wifi:410]:   SSID: [redacted]
[13:51:42][C][wifi:411]:   IP Address: 192.168.0.42
[13:51:42][C][wifi:413]:   BSSID: [redacted]
[13:51:42][C][wifi:414]:   Hostname: 'victron-debug'
[13:51:42][C][wifi:416]:   Signal strength: -52 dB ▂▄▆█
[13:51:42][C][wifi:420]:   Channel: 11
[13:51:42][C][wifi:421]:   Subnet: 255.255.255.0
[13:51:42][C][wifi:422]:   Gateway: 192.168.0.1
[13:51:42][C][wifi:423]:   DNS1: 192.168.0.1
[13:51:42][C][wifi:424]:   DNS2: 0.0.0.0
[13:51:42][C][logger:443]: Logger:
[13:51:42][C][logger:444]:   Level: DEBUG
[13:51:42][C][logger:445]:   Log Baud Rate: 115200
[13:51:42][C][logger:447]:   Hardware UART: UART0
[13:51:42][C][uart.idf:139]: UART Bus 1:
[13:51:42][C][uart.idf:140]:   TX Pin: GPIO16
[13:51:42][C][uart.idf:141]:   RX Pin: GPIO17
[13:51:42][C][uart.idf:143]:   RX Buffer Size: 256
[13:51:42][C][uart.idf:145]:   Baud Rate: 19200 baud
[13:51:42][C][uart.idf:146]:   Data Bits: 8
[13:51:42][C][uart.idf:147]:   Parity: NONE
[13:51:42][C][uart.idf:148]:   Stop bits: 1
[13:51:42][C][victron:032]: Victron:
[13:51:42][C][victron:063]:   Firmware Version 'victron-debug firmware version'
[13:51:42][C][victron:065]:   Device Type 'victron-debug device type'
[13:51:42][C][victron:084]:   Model Description 'victron-debug model description'
[13:51:42][C][mdns:115]: mDNS:
[13:51:42][C][mdns:116]:   Hostname: victron-debug
[13:51:42][C][ota:097]: Over-The-Air Updates:
[13:51:42][C][ota:098]:   Address: victron-debug.local:3232
[13:51:42][C][api:139]: API Server:
[13:51:42][C][api:140]:   Address: victron-debug.local:6053
[13:51:42][C][api:144]:   Using noise encryption: NO
[13:51:45][D][api:102]: Accepted 192.168.0.35
[13:51:45][W][api.connection:107]: : Reading failed: BAD_INDICATOR errno=11
[13:51:51][D][api:102]: Accepted 192.168.0.35
[13:51:51][W][api.connection:107]: : Reading failed: BAD_INDICATOR errno=11'

KinDR007, 2024-01-01:
Hi,
first of all try 

Check vedirect port with multimeter (between pin and gnd) to select right pin with tx uart on his side.

After select right pin try next change :

UART 2 pins. Labeled as rx2 (GPIO16)and tx2 (GPIO17) without resistor divider! Uart port on esp32 its mostly compatible with 5v 

´´´yaml
uart:
id: uart_0
tx_pin: GPIO17
rx_pin: GPIO16
´´´

Miaoucat0, 2024-01-01:
Thank you very much for the extended reply!
i will try the things you pointed me too and give you a reply of the results in a day or so.

Miaoucat0, 2024-01-01:
AHH! it is working! following your instructions it is now working!

- first i tested the voltage between the two pin of the Victron device. TX and GND is around **4-7V pulsating**.
- i changed the code to use the 2nd set of RX-TX pin. RX2 as you said GPIO16
- **i made another wire without the voltage divider**

Thank you very much for your help!!

[image]
[image]

closed 2024-01-01 by Miaoucat0

KinDR007, 2024-01-02:
Great ,
You are welcome