KinDR007/VictronMPPT-ESPHOME #144: Support for BlueSmart IP22 Charger?

nobthenob, 2024-01-14:
Hi there,

Thank you very much for providing this code to read data from Victron devices via ve.direct !

Beside some MPPT solar charger I also use an AC charger Victron BlueSmart IP22, which I would like to integrate to my HA environment.
I found this article that describes a hidden ve.direct port inside the Victron BlueSmart IP22 Charger:
[https://community.victronenergy.com/questions/226131/bluesmart-ip22-charger-and-vedirect.html](https://community.victronenergy.com/questions/226131/bluesmart-ip22-charger-and-vedirect.html)

I managed to follow the description and connected an ESP8266 D1 Mini to the port and was able to read some data and generated a debug log with your sample code [debug-esp8266-example.yaml](https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/debug-esp8266-example.yaml)

However, there is some information missing (e.g. type, model, serial number), when I tried to read the sensors with your component.
Unfortunately I am not able to use this information to add support for the IP22 charger to your code.
I have uploaded the log file if you are interested:
[debuglog_victron-BlueSmart_IP22_Charger.txt](https://github.com/KinDR007/VictronMPPT-ESPHOME/files/13932026/debuglog_victron-BlueSmart_IP22_Charger.txt)

Would be great if you could include support for the IP22 as well.

Once again, thank you for doing all this!
With best regards
Nob

KinDR007, 2024-02-21:
is supported if I'm not mistaken

[image]

icefest, 2024-06-05:
Ney Nob, I have a blue smart too, and added code for mine with #158
I've found another person online who looked like they had a similar one as you, so it might work properly for you now. 
If it does, then we can close this issue.

syssi, 2024-06-05:
BlueSmart IP22 frame extracted from https://github.com/KinDR007/VictronMPPT-ESPHOME/files/13932026/debuglog_victron-BlueSmart_IP22_Charger.txt

```
\r\n
PID\t0xA334\r\n
FWE\t0342FF\r\n
SER#\tHQ2438CPW6H\r\n
V\t26890\r\n
I\t3000\r\n
T\t---\r\n
ERR\t0\r\n
CS\t3\r\n
HC#\t378060\r\n
Checksum\t\r:A0920000022\n
:A0E200021FC\n
:A0A2000380B0000DE\n
:A0920000022\n
:A0E200021FC\n
:A0920000022\n
:A0E200021FC\n
:A0920000022\n
:A0E200021FC\n
:A0220007C0AA3\n
:A0920000022\n
:A0E200021FC\n
:A0A2000380B0000DE\n
:A0920000022\n
:A0E200021FC\n
:A0920000022\n
:A0E200021FC\n
:A8DED007F0A48\n
:AD5ED007F0A00\n
:A16EE007C0AC1\n
:A200100A702000081\n
:A30EC00032C\n
:A31EC000129E3A49834A300FF4203007FFFFFFFFF22E2979989A30014040000B7FFFFFFFF24\n
:A32EC0001C5BEBD9A60A00061010000C6FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7FFFFFFFFFBE\n
:A20EC00ECEDFFFFFFFFFFFF8DED000C22E297998CED000C22E297993EEC000FC5BEBD9AE6\n
:A0920000022\n
:A0E200021FC\n
:A0220007C0AA3\n
:A1320001C0C0000F0\n
:A0A2000200B0000F6\n
:A0920000022\n
:A0E200021FC\n
:A0920000022\n
:A0E200021FC\n
\r\n
```

nobthenob, 2024-06-09:
Hi Sebastian,

I can confirm that it is now recognizing my Victron charger IP22 24/12 (PID 0xA334) better, at least I now get the device type as "Blue Smart IP22 Charger 24|12" and the serial number. Therefore I guess we can close issue #144.

Thank for very much for your work to add support for the IP22 device - Much appreciated! :)

Meanwhile I have another Blue Smart IP22 device, but different model: 24|8. 
This has a maximum charge current of 8A and shows the PID 0xA332 in the log.

Just in case you want to add this model too, I will open an issue and provide the log there.

thanks again and best regards

closed 2024-06-09 by nobthenob

nobthenob, 2024-06-09:
Just as additional information here is a screenshot showing the Charger Produkt Details in the VictronConnect App.

[image]