KinDR007/VictronMPPT-ESPHOME #145: Errors with SmartSolar MQTT 100/50

Flori261, 2024-01-16:
Hello everyone,
I have tried to integrate my SmartSolar MPPT 100/50 with an ESP8266. However, I only receive the Panel Power value. 

I have installed the debug program. Here is the LOG

Do you see a possibility to implement this. Does the software need to be adapted or am I doing something wrong?

[logs_victron-solar10050-esp_logs.txt](https://github.com/KinDR007/VictronMPPT-ESPHOME/files/13955704/logs_victron-solar10050-esp_logs.txt)

KinDR007, 2024-01-16:
probably same as here 

https://github.com/KinDR007/VictronMPPT-ESPHOME/issues/143

Flori261, 2024-01-17:
Hey thanks for the reply,

I use a nodemcuv2 Amica without an second Rx tx pin. What pin should I use instead of D7?

Will a swap to a WeMos d1 mini help?

Flori261, 2024-01-17:
Change to WeMos d1 mini was not successful

Flori261, 2024-01-17:
Changed the wiring and the software on the nodemcu to talk on gpio3 (Rx) Not working at all.

KinDR007, 2024-01-20:
Please send foto of wiring (of mppt and esp side)and yaml for esphome

KinDR007, 2024-01-20:
And plase check settings of mppt "normal communication must be selected [image]

closed 2024-05-05 by KinDR007