KinDR007/VictronMPPT-ESPHOME #146: Missing value for "Panel current" for SmartSolar 100/30

Cs0nly, 2024-02-06:
Hello KinDR007,
first of all, thank you for this great project and especially for sharing your work and not least thanks to all involved participants.

It works great for me with Home Assistant all values are shown well, but one thing drives me crazy.
**I'm not be able to get a value for **"Panel current"**. It always shows "unbekannt (unknown)"**

I tried all what came in my mind but without a result.

Are you able to help me in this case with any idea?
All hints would be very appreciated.

**My Config:**
Victron SmartSolar 100/30
Solar Battery BIG 280 Ah 12V
Homeassistant current Version based NUC
Homeassistant current Version based RaspBerry Pi3
source 30V / 5 A via Power Supply
2 x ESP32 D1 flashed with 
 - smartsolar-mppt-esp8266-example.yaml
 - smartsolar-mppt-esp8266-example-advanced.yaml

Btw. in the file smartsolar-mppt-esp8266-example.yaml I think there is a small error in line 60,61. The "charging current" of the battery is displayed as "Panel Current", if I'm not wrong. 
....
60    battery_current:
61     name: "${name} panel current"
....

Thanks a lot in advance
Christian

[image]

KinDR007, 2024-02-07:
Hi ,you right it is battery current not panels curent.

the mppt charger reports only the power and voltage of solar panels . If you need current, you have to create a template sensor and calculate it according to Ohm's law

Cs0nly, 2024-02-07:
many thanks you for your fast response.
Ok, I understand but I've seen lot of pictures in here, which are shown the value. And cause it is provided in the BT VictronConnect app I thought the mistake is on my side.
Does this mean that it depends only my SmartSolar 100/30 or is this ment for all SmarSolars?

But thanks again this prevents me to waste more time on this 😊

Cs0nly, 2024-02-07:
Thank you
Christian

closed 2024-02-07 by Cs0nly