KinDR007/VictronMPPT-ESPHOME #151: Max Power Yesterday - Shows Incorrect Value

Roving-Ronin, 2024-03-13:
Hi @KinDR007  / Jiri.

Could you please assist, just looking at the 'Max Power Yesterday' and noticed that it doesn't work correctly. Instead of showing the maximum power for yesterday, it is showing a lower figure from just before midnight each day.

Tried to find the library / formula that defines this sensor in the integration, but doesn't appear visible in the files of components/victron ?

See below, where it is showing 274W for max yesterday, but in fact it was 399W.

[image]

KinDR007, 2024-03-13:
Hi, what value do you see in the victron app? the component only reads what mppt sends it, so please check whether mppt really does not report those values ​​like that

Roving-Ronin, 2024-03-13:
Unfortunately I don't have access to the units, they are at a remote site.  Are they a reading the Victron provides itself, or does your integration take another output and filter it to show the highest yesterday (hence why I was trying to find it in the source code)?

Just looking now at the 'Max Power Today', but looking at the sensor data for yesterday and I noticed that it is also showing 274W (like the Max Power Yesterday reading, even though the Max Power Yesterday graph shows 399W).

[image]

KinDR007, 2024-03-14:
The component does not calculate it, it only receives the value from mppt

closed 2024-05-05 by KinDR007