KinDR007/VictronMPPT-ESPHOME #152: [W][victron:093] Last transmission too long ago

supp0815, 2024-03-14:
Hey there,

I don't think it's a software issue. I had it running a year ago. System was down for a while. Today I connected everything including a new D1 mini. Data from TX on the mppt to D7 (GPIO13). When connected I don't get any victron data and instead the "[W][victron:093] Last transmission too long ago" message in the logs. Removing the data cable, message disappears. 

System is, 12V battery goes to a linear actuator and 5V regulator, from the regulator to D1 mini. D1 controls the relays and a data line on D7 to the MPPT 100/30.  

Any ideas what's missing?

[image]

KinDR007, 2024-03-14:
What mppt ? Blue or smart? How is the "ve.direct" port set? How sure are you about the wiring ? Any photos of wiring ?

supp0815, 2024-03-14:
> What mppt ? Blue or smart? How is the "ve.direct" port set? How sure are you about the wiring ? Any photos of wiring ? 

It's the Victron Energy BlueSolar MPPT. The wiring should be fine, as is said I was using it for a while and it was working with the same code.

[image]

D7 to MPPT

[image]

supp0815, 2024-03-16:
Alright, I think it was the D1 mini. I tried another ESP and connected that one separately straight to the victron and powered by the 12V battery (using a regulator). Works fine, no issues.

closed 2024-03-16 by KinDR007