KinDR007/VictronMPPT-ESPHOME #155: Some values are not displayed Victron MPPT 150/45 and 100/20

DerKlotz74, 2024-04-06:
Hello, 
I have the problem that some values are not supplied by the Victron. Please help. What is needed?

`sensor:
  - platform: victron
    victron_id: victron150
    max_power_yesterday:
      name: "${device0} max power yesterday"
    max_power_today:
      name: "${device0} max power today"  
    panel_voltage:
      name: "${device0} panel voltage"
    panel_power:
      name: "${device0} panel power"
    battery_current:
      name: "${device0} panel current"
    load_current:
      name: "${device0} load current"
    battery_voltage:
      name: "${device0} battery voltage"
    min_battery_voltage:
      name: "${device0} min battery voltage" 
    max_battery_voltage:
      name: "${device0} max battery voltage" 
    number_of_charge_cycles:  
      name: "${device0} number of charge circles"
    number_of_full_discharges:
      name: "${device0} number of full discharges"
    amount_of_discharged_energy:
      name: "${device0} amount of discharged energy"
    amount_of_charged_energy:
      name: "${device0} amount of charged energy"  
`

[image]
[image]

DerKlotz74, 2024-04-06:
[image]

DerKlotz74, 2024-04-06:
Have an ESP32

syssi, 2024-06-04:
Please enable the debug output of the `uart` component and provide some logs.

```
uart:
  id: uart_0
  baud_rate: 19200
  tx_pin: ${tx_pin}
  rx_pin: ${rx_pin}
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

I assume your Victron device doesn't provide the missing measurements above and you have to simply remove the sensors from the YAML to get rid of the useless sensors without measurements.

KinDR007, 2024-06-20:
Hi , 
sorry for delayed response. 
All values with N/A corespond for smartshunt or BVM these values is not available for MPPT's

closed 2024-06-20 by KinDR007