KinDR007/VictronMPPT-ESPHOME #156: WROOM ESP32 CH340C Pinout

tbgoose, 2024-05-05:
Hi all

Trying this with a wroom-32, and I can't get my head around what D7 and D8 are on this board.

Looks like D7 may be GPIO13, but I can't figure out what D8 is at all, or what alternative pin would be ok to use. I have this wroom board with CH340C - https://www.aliexpress.com/item/1005006336429842.html

I think I have my head around the rest, here is my prepared yaml.

```
substitutions:
  name: victron_solar_controller
  external_components_source: github://KinDR007/VictronMPPT-ESPHOME@main

esphome:
  name: ${name}
  comment: Front Gate ESP32 controller
  area: Outside

external_components:
  - source: ${external_components_source}
    refresh: 0s

esp32:
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

uart:
  id: uart_0
  tx_pin: D8  # Not used! The communication is read-only
  rx_pin: GPIO13
  baud_rate: 19200
  rx_buffer_size: 256

victron:
  uart_id: uart_0
  id: victron0
  throttle: 30s

sensor:
  - platform: victron
    victron_id: victron0
    max_power_yesterday:
      name: "${name} max power yesterday"
    max_power_today:
      name: "${name} max power today"
    yield_total:
      name: "${name} yield total"
    yield_yesterday:
      name: "${name} yield yesterday"
    yield_today:
      name: "${name} yield today"
    panel_voltage:
      name: "${name} panel voltage"
    panel_power:
      name: "${name} panel power"
    battery_current:
      name: "${name} panel current"
    battery_voltage:
      name: "${name} battery voltage"
    day_number:
      name: "${name} day number"
    charging_mode_id:
      name: "${name} charging mode id"
    error_code:
      name: "${name} error code"
    tracking_mode_id:
      name: "${name} tracking mode id"
    load_current:
      name: "${name} load current"

text_sensor:
  - platform: victron
    victron_id: victron0
    charging_mode:
      name: "${name} charging mode"
    error:
      name: "${name} error"
    tracking_mode:
      name: "${name} tracking mode"
    firmware_version:
      name: "${name} firmware version"
    device_type:
      name: "${name} device type"
    serial_number:
      name: "${name} serial number"

binary_sensor:
  - platform: victron
    victron_id: victron0
    load_state:
      name: "${name} load state"
    relay_state:
      name: "${name} relay state"

KinDR007, 2024-05-05:
hi ,
use these pins :
**GPIO16 as (RX)**
GPIO17 as (TX) (default NC)
it is classic wiring for esp32 chips and UART 2

```
uart:
  id: uart_0
  tx_pin: GPIO17  # Not used! The communication is read-only
  rx_pin: GPIO16
  baud_rate: 19200
  rx_buffer_size: 256
  ```

tbgoose, 2024-05-06:
Thankyou

I can't seem to get this to work either. At my desk it connects to wifi and I can visit it via localip or see logs in ESPHome. But there is obviously no data.

When I take it to the victron and plug in the white RX cable on the JST cable to GPIO 16, it won't ever seem to jump back on the wifi. Tested both with ground connected or unplugged.

```
INFO ESPHome 2024.4.2
INFO Reading configuration /config/esphome/esphome-web-41ff18.yaml...
INFO Updating https://github.com/KinDR007/VictronMPPT-ESPHOME.git@main
WARNING 'victron_solar_controller': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Starting log output from 192.168.1.148 using esphome API
WARNING Can't connect to ESPHome API for victron_solar_controller @ ip: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='ip', port=6053))]: [Errno 113] Connect call failed ('ip', 6053) (SocketAPIError)
INFO Trying to connect to victron_solar_controller @ ip in the background
```

My gate controller is right at the limit of the property so thinking that it's a little too far away even though there is wifi there. I use a sonoff controller to actually monitor and control the gate at the same location. It complains occasionally but it does connect and stays connected for the most part.

Perhaps the wireless antenna on the esp board is a bit weaker. Thoughts on if you think this might be the problem or if I have done something else wrong. It is night time now but I can take some photos showing wiring etc tomorrow.

Should I look for a board with external antenna (if such a thing exists).

crankswagon, 2024-05-19:
Those MPPT TX pins are 5vs, the ESP32-WROOM can only take a max of 3.6v` on its GPIO pins. You are probably overvolting the ESP32 board and it just crashes completely rather than dropping off of wifi.

You will need a voltage divider if you want to connect to GPIO16; check out the diagram in this repo here https://github.com/RalfJL/VE.Direct2MQTT/wiki/Hardware

I use a ESP32-WROOM with a BMV700, its TX pin is 3.3v as per Victron documentation here https://www.victronenergy.com/live/vedirect_protocol:faq#q4is_the_vedirect_interface_33_or_5v