KinDR007/VictronMPPT-ESPHOME #157: Adding another VE.Direct device (charger 12/25)

icefest, 2024-06-04:
I have a **Blue Smart IP65 Charger 12/25**
Device ID 0xA30A

It has a VE.Direct port that I can interrogate to get some information, but I was hoping to help to get it fully supported. 

Log of device:
[logs_airquality_logs (1).txt](https://github.com/user-attachments/files/15542696/logs_airquality_logs.1.txt)

I'm glad to do more debugging to help this work.

syssi, 2024-06-04:
Good job! These are some frames from your log:

```
PID\t0xA30A\r\n
FWE\t0325FF\r\n
SER#\tHQ2232GGMEH\r\n
V\t14200\r\n
I\t1300\r\n
ERR\t0\r\n
CS\t4\r\n
HC#\t136663\r\n
Checksum\t\x10
:A8CED003C05000091\n
:A200100750C0000A9\n
\r\n

PID\t0xA30A\r\n
FWE\t0325FF\r\n
SER#\tHQ2232GGMEH\r\n
V\t14200\r\n
I\t1300\r\n
ERR\t0\r\n
CS\t4\r\n
HC#\t136663\r\n
Checksum\t\x10
:A8CED0028050000A5\n
:ADBED00520829\n
:A200100760C0000A8\n
\r\n

PID\t0xA30A\r\n
FWE\t0325FF\r\n
SER#\tHQ2232GGMEH\r\n
V\t14200\r\n
I\t1300\r\n
ERR\t0\r\n
CS\t4\r\n
HC#\t136663\r\n
Checksum\t\x10
:A8FED000800C7\n
:A8CED0016030000B9\n
:AD7ED0008007F\n
:ADBED00480833\n
:A200100770C0000A7\n
\r\n

PID\t0xA30A\r\n
FWE\t0325FF\r\n
SER#\tHQ2232GGMEH\r\n
V\t14200\r\n
I\t800\r\n
ERR\t0\r\n
CS\t4\r\n
HC#\t136663\r\n
Checksum\t<
:A8CED0020030000AF\n
:ADBED00520829\n
:A200100780C0000A6\n
\r\n

PID\t0xA30A\r\n
FWE\t0325FF\r\n
SER#\tHQ2232GGMEH\r\n
V\t14200\r\n
I\t800\r\n
ERR\t0\r\n
CS\t4\r\n
HC#\t136663\r\n
Checksum\t<
:A8FED000E00C1\n
:A8CED006E0500005F\n
:AD7ED000E0079\n
:A200100790C0000A5\n
\r\n
```

syssi, 2024-06-04:
Your device publishes measurement of these keys:

- PID
- FWE
- SER#
- V
- I
- ERR
- CS
- HC

Let's check what's already supported by the component and what's missing.

syssi, 2024-06-04:
PID: https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/components/victron/victron.cpp#L968-L970
FWE: https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/components/victron/victron.cpp#L968-L970
SER#: https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/components/victron/victron.cpp#L973-L976
V: https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/components/victron/victron.cpp#L645-L648
I: https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/components/victron/victron.cpp#L693-L697
ERR: https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/components/victron/victron.cpp#L925-L930
CS: https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/components/victron/victron.cpp#L932-L937
HC: Not implemented yet.

PID 0xA30A: Missing here: https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/components/victron/victron.cpp#L343-L620

TODOs:

1. Write a example YAML including the mentioned sensors above.
2. Add 0xA30A to the `device_type_text` method to return your proper device model
3. Download the Victron VE.Direct ASCII protocol description and get an idea what's `HC` is about and implement a new sensor.

icefest, 2024-06-04:
Thanks! 

This is the yaml:

```

substitutions:
  name: bluesmart-charger
  external_components_source: github://KinDR007/VictronMPPT-ESPHOME@main

esphome:
  name: ${name}

esp32:
  board: esp32dev
  framework:
    type: arduino

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

ota:
# api:

logger:
  baud_rate: 0

mqtt:
  broker: !secret mqtt_host
  username: !secret mqtt_username
  password: !secret mqtt_password
  id: mqtt_client

uart:
  id: uart_0
  tx_pin: D8  # Not connected! The communication is read-only
  rx_pin: D7  # Connect this this GPIO and GND to the charger
  baud_rate: 19200
  rx_buffer_size: 256

victron:
  uart_id: uart_0
  id: victron0
  throttle: 10s

sensor:
  - platform: victron
    victron_id: victron0
    battery_voltage:
      name: "charger output voltage"
    battery_current:
      name: "charger current"
    error_code:
      name: "charger error code"
    charging_mode_id:
      name: "charging mode id"

text_sensor:
  - platform: victron
    victron_id: victron0
    device_type:
      name: "${name} device type"
    firmware_version
      name: "${name} Firmware Version"
    serial_number:
      name: "${name} serial number"
    error:
      name: "${name} error"
      entity_category: DIAGNOSTIC
    charging_mode:
      name: "${name} Charging mode"

```

I've made a fork with the added device IDs

`HC` is the hardware revision

syssi, 2024-06-04:
Do you like to prepare a contribution / pull request?

icefest, 2024-06-04:
I can't figure out why i'm getting rubbish answers for the hardware version: 
[image]

icefest, 2024-06-04:
I've tried the pull request!

syssi, 2024-06-04:
Thanks for your contribution!