KinDR007/VictronMPPT-ESPHOME #160: Support for BlueSmart IP22 Charger model 24/8?

nobthenob, 2024-06-09:
Hello,

with reference to [#144](https://github.com/KinDR007/VictronMPPT-ESPHOME/pull/144) and I would like to open another issue for the IP22 range of chargers.

Meanwhile I have another Blue Smart IP22 device, but different model: 24|8.
This has a maximum charge current of 8A and shows the PID 0xA332 in the log, see below.
I also add a screenshot of the Product Information page as shown in the VictronConnect App.

As there have been some questions rearding how to interpret the firmware tag "FWE\t0342FF\r\n":
It seems at least in my case the number "0342" match the firmware shown in the app.

@syssi : If you would find time to add this, thanks a lot. If I can help with more info, let me know.

nob

nobthenob, 2024-06-09:
[12:34:23][D][uart_debug:158]: <<< "\r\n"
[12:34:23][D][uart_debug:158]: <<< "PID\t0xA332\r\n"
[12:34:23][D][uart_debug:158]: <<< "FWE\t0342FF\r\n"
[12:34:23][D][uart_debug:158]: <<< "SER#\tHQ2249TNFVM\r\n"
[12:34:23][D][uart_debug:158]: <<< "V\t26870\r\n"
[12:34:23][D][uart_debug:158]: <<< "I\t2000\r\n"
[12:34:23][D][uart_debug:158]: <<< "T\t---\r\n"
[12:34:23][D][uart_debug:158]: <<< "ERR\t0\r\n"
[12:34:23][D][uart_debug:158]: <<< "CS\t3\r\n"
[12:34:23][D][uart_debug:158]: <<< "HC#\t57825\r\n"
[12:34:23][D][uart_debug:158]: <<< "Checksum\t\x18:A0920000022\n"
[12:34:23][D][uart_debug:158]: <<< ":A0E200021FC\n"
[12:34:23][D][uart_debug:158]: <<< ":A0920000022\n"
[12:34:23][D][uart_debug:158]: <<< ":A0E200021FC\n"
[12:34:23][D][uart_debug:158]: <<< ":A132000326900007D\n"
[12:34:23][D][uart_debug:158]: <<< ":A0220007F0AA0\n"
[12:34:23][D][uart_debug:158]: <<< ":A0920000022\n"
[12:34:23][D][uart_debug:158]: <<< ":A0E200021FC\n"
[12:34:23][D][uart_debug:158]: <<< ":A0A20001E6600009D\n"
[12:34:23][D][uart_debug:158]: <<< ":A0920000022\n"
[12:34:23][D][uart_debug:158]: <<< ":A0E200021FC\n"
[12:34:23][D][uart_debug:158]: <<< ":A0920000022\n"
[12:34:23][D][uart_debug:158]: <<< ":A0E200021FC\n"
[12:34:23][D][uart_debug:158]: <<< ":A0920000022\n"
[12:34:23][D][uart_debug:158]: <<< ":A0E200021FC\n"
[12:34:23][D][uart_debug:158]: <<< ":A0220007F0AA0\n"
[12:34:23][D][uart_debug:158]: <<< ":A8DED00800A47\n"
[12:34:23][D][uart_debug:158]: <<< ":AD5ED00800AFF\n"
[12:34:23][D][uart_debug:158]: <<< ":A16EE007F0ABE\n"
[12:34:23][D][uart_debug:158]: <<< ":A20010077000000B3\n"
[12:34:23][D][uart_debug:158]: <<< ":A30EC00052A\n"
[12:34:23][D][uart_debug:158]: <<< ":A31EC000122D5569A32A300FF4203007FFFFFFFFFC5BEBD9A60A00063010000B9FFFFFFFFBF\n"
[12:34:24][D][uart_debug:158]: <<< ":A32EC000122E2979989A30016040000BCFFFFFFFFB332EB9A60A00063010000BDFFFFFFFF73\n"
[12:34:24][D][uart_debug:158]: <<< ":A33EC000129E3A49834A300FF420300CCFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF7FFFFFFFFF90\n"
[12:34:24][D][uart_debug:158]: <<< ":A20EC00ECEDFFFFFFFFFFFF8DED000C22E297998CED000C22E297993EEC000FB332EB9A56\n"
[12:34:24][D][uart_debug:158]: <<< ":A0A20001E6600009D\n"
[12:34:24][D][uart_debug:158]: <<< ":A132000AA69000005\n"
[12:34:24][D][uart_debug:158]: <<< ":A0920000022\n"
[12:34:24][D][uart_debug:158]: <<< ":A0E200021FC\n"
[12:34:24][D][uart_debug:158]: <<< ":A0920000022\n"
[12:34:24][D][uart_debug:158]: <<< ":A0E200021FC\n"
[12:34:24][D][uart_debug:158]: <<< ":A0920000022\n"
[12:34:24][D][uart_debug:158]: <<< ":A0E200021FC\n"
[12:34:24][D][uart_debug:158]: <<< ":A0220007F0AA0\n"
[12:34:24][D][uart_debug:158]: <<< ":A0920000022\n"
[12:34:24][D][uart_debug:158]: <<< ":A0E200021FC\n"
[12:34:24][D][uart_debug:158]: <<< "\r\n"

nobthenob, 2024-06-09:
[image]

icefest, 2024-06-09:
Interesting! I can add the model, do you have any idea how to interpret the HC#?
firmware is working for me, have you added the sensor to your yaml file?
```
text_sensor:
  - platform: victron
    victron_id: victron0
    firmware_version
      name: "${name} Firmware Version"
```

closed 2024-06-09 by syssi

syssi, 2024-06-09:
Using this [YAML](https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/bluesmart-charger-esp32-example.yaml) the frame

```
[12:34:23][D][uart_debug:158]: <<< "PID\t0xA332\r\n"
[12:34:23][D][uart_debug:158]: <<< "FWE\t0342FF\r\n"
[12:34:23][D][uart_debug:158]: <<< "SER#\tHQ2249TNFVM\r\n"
[12:34:23][D][uart_debug:158]: <<< "V\t26870\r\n"
[12:34:23][D][uart_debug:158]: <<< "I\t2000\r\n"
[12:34:23][D][uart_debug:158]: <<< "T\t---\r\n"
[12:34:23][D][uart_debug:158]: <<< "ERR\t0\r\n"
[12:34:23][D][uart_debug:158]: <<< "CS\t3\r\n"
[12:34:23][D][uart_debug:158]: <<< "HC#\t57825\r\n"
[12:34:23][D][uart_debug:158]: <<< "Checksum\t\x18:A0920000022\n"
```

is decoded as:

```
[16:43:13][D][sensor:093]: 'bluesmart-charger output voltage': Sending state 14.20000 V with 3 decimals of accuracy
[16:43:14][D][sensor:093]: 'bluesmart-charger current': Sending state 1.00000 A with 3 decimals of accuracy
[16:43:14][D][sensor:093]: 'bluesmart-charger error code': Sending state 0.00000  with 0 decimals of accuracy
[16:43:14][D][text_sensor:064]: 'bluesmart-charger error': Sending state 'No error'
[16:43:14][D][sensor:093]: 'bluesmart-charger mode id': Sending state 4.00000  with 0 decimals of accuracy
[16:43:14][D][text_sensor:064]: 'bluesmart-charger Charging mode': Sending state 'Absorption'
```

The firmware version (`FWE`), hardware revision (`HC#`) and serial number (`SER#`) sensor does publish the state just once (because of the usage of the `publish_state_once_()` method). I guess it's published too early sometimes.

nobthenob, 2024-06-09:
> Interesting! I can add the model, do you have any idea how to interpret the HC#? firmware is working for me, have you added the sensor to your yaml file?
> 
> ```
> text_sensor:
>   - platform: victron
>     victron_id: victron0
>     firmware_version
>       name: "${name} Firmware Version"
> ```

Thank you for the quick reaction!
Yes, the firmware sensors are already present :)
I am using a D1 Mini ESP32 to connect to two IP22 devices, one 24|12 version and one 24|8.
Will report, once the Code is available.

Unfortunately regarding the HC# code, I have no idea :(
nob

nobthenob, 2024-06-09:
Ah, this was overlapping.
@syssi : So would it help if I try publish state triggered by a timer ?

syssi, 2024-06-09:
Proof-of-concept:

```
--- a/components/victron/victron.cpp
+++ b/components/victron/victron.cpp
@@ -1062,8 +1062,8 @@ void VictronComponent::publish_state_once_(text_sensor::TextSensor *text_sensor,
   if (text_sensor == nullptr)
     return;
 
-  if (text_sensor->has_state())
-    return;
+//  if (text_sensor->has_state())
+//    return;
 
   text_sensor->publish_state(state);
 }
```

```
[17:32:54][D][text_sensor:064]: 'bluesmart-charger device type': Sending state 'Blue Smart IP65 Charger 12|25'
[17:32:54][D][text_sensor:064]: 'bluesmart-charger serial number': Sending state 'HQ2232GGMEH'
[17:32:54][D][sensor:093]: 'bluesmart-charger output voltage': Sending state 14.19000 V with 3 decimals of accuracy
[17:32:54][D][sensor:093]: 'bluesmart-charger current': Sending state 1.00000 A with 3 decimals of accuracy
[17:32:54][D][sensor:093]: 'bluesmart-charger error code': Sending state 0.00000  with 0 decimals of accuracy
[17:32:54][D][text_sensor:064]: 'bluesmart-charger error': Sending state 'No error'
[17:32:54][D][sensor:093]: 'bluesmart-charger mode id': Sending state 4.00000  with 0 decimals of accuracy
[17:32:54][D][text_sensor:064]: 'bluesmart-charger Charging mode': Sending state 'Absorption'
[17:32:54][D][text_sensor:064]: 'bluesmart-charger hardware number': Sending state '136663'
```

syssi, 2024-06-09:
Oh, there is a small issue with the firmware version text sensor of the YAML. The `firmware_version_24bit` sensor must be used to decode `FWE` properly:

```diff
diff --git a/bluesmart-charger-esp32-example.yaml b/bluesmart-charger-esp32-example.yaml
index a682b3c..8142e8a 100644
--- a/bluesmart-charger-esp32-example.yaml
+++ b/bluesmart-charger-esp32-example.yaml
     victron_id: victron0
     device_type:
       name: "${name} device type"
-    firmware_version:
+    firmware_version_24bit:
      name: "${name} Firmware Version"
     serial_number:
       name: "${name} serial number"
     hardware_revision:
```

```
[17:36:09][D][text_sensor:064]: 'bluesmart-charger firmware version': Sending state '03.25-official'
```

```

icefest, 2024-06-09:
I was wondering what that bit was for! Seemed to work done without it for
me :(
I'm sorry I'm bad at coding:(

syssi, 2024-06-09:
@icefest No problem. FYI this is the difference:

https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/components/victron/victron.cpp#L951-L968

The `firmware_version` sensor exposes the reading of key `FW`. The `firmware_version_24bit` sensor does decode the `FWE` content.

syssi, 2024-06-09:
Hmm.. it looks like `publish_state_once_` does a good job. No race conditions here. The values are published once on startup:

```
bluesmart-charger/sensor/bluesmart-charger_output_voltage/state 14.200
bluesmart-charger/sensor/bluesmart-charger_current/state 1.000
bluesmart-charger/sensor/bluesmart-charger_mode_id/state 4
bluesmart-charger/sensor/bluesmart-charger_error_code/state 0
bluesmart-charger/sensor/bluesmart-charger_charging_mode/state Absorption
bluesmart-charger/sensor/bluesmart-charger_error/state No error
bluesmart-charger/sensor/bluesmart-charger_firmware_version/state 03.25-official
bluesmart-charger/sensor/bluesmart-charger_device_type/state Blue Smart IP65 Charger 12|25
bluesmart-charger/sensor/bluesmart-charger_serial_number/state HQ2232GGMEH
bluesmart-charger/sensor/bluesmart-charger_hardware_number/state 136663
bluesmart-charger/sensor/bluesmart-charger_output_voltage/state 14.190
bluesmart-charger/sensor/bluesmart-charger_current/state 1.000
bluesmart-charger/sensor/bluesmart-charger_error_code/state 0
bluesmart-charger/sensor/bluesmart-charger_error/state No error
bluesmart-charger/sensor/bluesmart-charger_mode_id/state 4
bluesmart-charger/sensor/bluesmart-charger_charging_mode/state Absorption
bluesmart-charger/sensor/bluesmart-charger_output_voltage/state 14.190
bluesmart-charger/sensor/bluesmart-charger_current/state 1.000
bluesmart-charger/sensor/bluesmart-charger_error_code/state 0
bluesmart-charger/sensor/bluesmart-charger_error/state No error
bluesmart-charger/sensor/bluesmart-charger_mode_id/state 4
bluesmart-charger/sensor/bluesmart-charger_charging_mode/state Absorption
bluesmart-charger/sensor/bluesmart-charger_output_voltage/state 14.200
bluesmart-charger/sensor/bluesmart-charger_current/state 1.000
bluesmart-charger/sensor/bluesmart-charger_error_code/state 0
bluesmart-charger/sensor/bluesmart-charger_error/state No error
bluesmart-charger/sensor/bluesmart-charger_mode_id/state 4
bluesmart-charger/sensor/bluesmart-charger_charging_mode/state Absorption
```

syssi, 2024-06-09:
@nobthenob Do you use `mqtt` or the native `api` component?

nobthenob, 2024-06-09:
Ah, this was overlapping.
@syssi : So would it help if I try publish state triggered by a timer ?

> @nobthenob Do you use `mqtt` or the native `api` component?

I use the native API.
However, I have a question regarding the publish state once procedure:
In my case the ESP is always running but the chargers are switched on/off on the AC side by some Tasmota plugs, because I think the IP22 has no command to stop charging.
That means, when the IP22 is started, the firmware version (and the other two) were probably already attempted to read!?
If my understanding is correct, would it be possible to read in the those values again, when there is first valid data coming from the charger?

Interestingly, I get the serial numbers of my devices w/o a problem.

Hope that helps.

syssi, 2024-06-09:
It's a bit different: This project/component doesn't know about the state (charger on/off) of your charger. If a UART message from your charger arrives it will be decoded instantly. If there is no traffic there is nothing to decode and the sensor states won't be updated anymore.

There are some sensor states  which doesn't change over time (the serial number is static, the hardware revision is static and so on) so the will be decoded just once "per lifetime". If you ESP reboots and receives the `serial number` the first time the values gets stored and published once. If your ESP runs 7 days in a row and you turn on/off the charger multiple times this doesn't have any effect here.

syssi, 2024-06-09:
@nobthenob Could you provide some screenshots of the sensor state? For me it's working properly and it looks like I don't understand the issue at all. What's not working as expected?

nobthenob, 2024-06-09:
Here is the screenshot of the sensor states.
I get the device_info sind serial_number, but not the firmware_version.
[image]

syssi, 2024-06-09:
@nobthenob Could you provide your YAML configuration? I would like to check the `Firmware Version` section.

nobthenob, 2024-06-09:
Here is my YAML code:

[victron_IP22_ve_direct_control.yaml.txt](https://github.com/user-attachments/files/15753886/victron_IP22_ve_direct_control.yaml.txt)

syssi, 2024-06-09:
Please change these lines from:

``` 
# ...

    firmware_version:         # "FWE\t0342FF\r\n" -> unknown ??
      name: "IP22 8A Firmware Version"

# ...

    firmware_version:         # "FWE\t0342FF\r\n" -> unknown ??
      name: "IP22 12A Firmware Version"
```

to 

``` 
# ...

    firmware_version_24bit:         # "FWE\t0342FF\r\n" -> unknown ??
      name: "IP22 8A Firmware Version"

# ...

    firmware_version_24bit:         # "FWE\t0342FF\r\n" -> unknown ??
      name: "IP22 12A Firmware Version"

```

This will fix the issue.

nobthenob, 2024-06-09:
Thank you, this has fixed it. 
It shows now "03.42-official" for both of my IP22 charger models.

Will this be a temporary workaround as the official [project page](https://github.com/KinDR007/VictronMPPT-ESPHOME) shows "firmware_version" as name of the related text sensor?

syssi, 2024-06-09:
I will update the docs. It's just incomplete. This is the background: https://github.com/KinDR007/VictronMPPT-ESPHOME/issues/160#issuecomment-2156661070

nobthenob, 2024-06-09:
> I will update the docs. It's just incomplete. This is the background: [#160 (comment)](https://github.com/KinDR007/VictronMPPT-ESPHOME/issues/160#issuecomment-2156661070)

ok, thx :)

aiolos, 2024-08-17:
> Here is my YAML code:
> 
> [victron_IP22_ve_direct_control.yaml.txt](https://github.com/user-attachments/files/15753886/victron_IP22_ve_direct_control.yaml.txt)

@nobthenob unrelated to this issue, but this configuration is great. It's nice to be able to control the charger a bit. Did you already find a way to disable charging?
It might also be worth to add it as example configuration to this repo, so people can get started with controlling the charger. (I want to add this configuration (adapted to my own situation) also to my ESPHome configuration repo).

nobthenob, 2024-08-18:
@aiolos 
Hi Henri,
Unfortunately, I did not find a way to disable charging via the ve.direct interface.
So far I used Tasmota Wifi plugs to switch the chargers on/off. I tried to find a compromize regarding the control logic in order to reduce the amount of switch action, don't know, if it will effect the overall lifetime of the victron IP22 chargers.

Feel free to add the code to your repo.
Please add a reference to https://community.victronenergy.com/questions/226131/bluesmart-ip22-charger-and-vedirect.html (see [#144 ](https://github.com/KinDR007/VictronMPPT-ESPHOME/pull/144)), which was the original source of information for all this.

Best regards
nob

aiolos, 2024-08-18:
Hi @nobthenob (and others)

Just FYI, I added the code to my repo: https://github.com/aiolos/ESPHome-config/tree/main/victron-charger-control
Including references.

PR #161 (2024-06-09): Add support for 0xA332 - Blue Smart IP22 charger 24|8 
Should partially fix #160 
 FWE works for me, cant replicate error