KinDR007/VictronMPPT-ESPHOME #169: ESPhome firmware 2024.6.1

willkpalmer, 2024-06-20:
Hi there, 

I'm trying to update my ESPhome object to 2024.6.1 and I get the following message when validating my code

_loose object 9d71066bd8746627abd58a095e6bea522b34bb5b (stored in .git/objects/9d/71066bd8746627abd58a095e6bea522b34bb5b) is corrupt.
    source: github://KinDR007/VictronMPPT-ESPHOME@main_

I'm inexperienced with both ESPHome and Git, this is my first ESP8266 project so I'm not sure where the problem lies.

Thanks!

KinDR007, 2024-06-20:
Hi,
please try clean build (respectively clean and rebuild ) 

[image]

Roving-Ronin, 2024-06-21:
@willkpalmer 
The latest release of ESPHome is 2024.6.0  (not 2024.6.1 yet) - https://esphome.io/changelog/2024.6.0.html
Assuming your running this latest version, I've just created a pull to update the example ESP device yaml's to support this version, see: https://github.com/KinDR007/VictronMPPT-ESPHOME/pull/170