KinDR007/VictronMPPT-ESPHOME #175: Not receiving any Values from Victron

Tsumiki-Chan, 2024-08-19:
I installed the config on a NodeMcu ESP8266 ESP-12F and connected it according to the Images. However the ESP doesn't seem to get any information from the victron 75-15. What could be the culprit here?

Config:
```yaml
substitutions:
  external_components_source: github://KinDR007/VictronMPPT-ESPHOME@main
  lower_devicename: "victronsolar"
  devicename: "SmartSolar_75_15"
  config_version: "v2024.06.21"
  wifi_fast_connect: "false"
  accuracy: "2"

esphome:
  name: "${lower_devicename}"
  platform: ESP8266
  #board: d1_mini
  board: nodemcuv2
  min_version: 2024.6.0

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  fast_connect: $wifi_fast_connect

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${devicename} Hotspot"
    password: !secret wifi_password

captive_portal:

ota:
  platform: esphome

logger:
  baud_rate: 0
  # level: VERY_VERBOSE
  esp8266_store_log_strings_in_flash: false

api:

web_server:
  port: 80

time:
  - platform: sntp

uart:
  id: uart_bus
  tx_pin: D8  # Not connected! The communication is read-only
  rx_pin: D7  # Connect this this GPIO and GND to the MPPT charger
  baud_rate: 19200
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_buffer_size: 256
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);

victron:
[... 247 lines cut ...]
[19:36:02][C][uptime.sensor:034]:   Type: Seconds
[19:36:02][C][template.sensor:022]: Template Sensor 'Bat S1 '
[19:36:02][C][template.sensor:022]:   Device Class: 'voltage'
[19:36:02][C][template.sensor:022]:   State Class: ''
[19:36:02][C][template.sensor:022]:   Unit of Measurement: 'V'
[19:36:02][C][template.sensor:022]:   Accuracy Decimals: 2
[19:36:02][C][template.sensor:022]:   Icon: 'mdi:battery'
[19:36:02][C][template.sensor:023]:   Update Interval: 10.0s
[19:36:02][D][text_sensor:064]: 'SmartSolar_75_15 Uptime (formatted)': Sending state '0d 00h 00m 18s'
[19:36:02][D][sensor:093]: 'SmartSolar_75_15 Uptime': Sending state 23.37900 s with 0 decimals of accuracy
[19:36:03][D][sensor:093]: 'Bat working percent': Sending state nan % with 2 decimals of accuracy
[19:36:03][D][sensor:093]: 'Wifi Quality': Sending state 42.00000 % with 0 decimals of accuracy
[19:36:03][D][sensor:093]: 'FVE percent': Sending state nan % with 2 decimals of accuracy
[19:36:04][D][sensor:093]: 'Day median generated ': Sending state nan kWh with 3 decimals of accuracy
[19:36:06][D][sensor:093]: 'Bat S1 ': Sending state nan V with 2 decimals of accuracy
[19:36:07][D][text_sensor:064]: 'SmartSolar_75_15 Uptime (formatted)': Sending state '0d 00h 00m 23s'
[19:36:07][D][sensor:093]: 'SmartSolar_75_15 Uptime': Sending state 28.37800 s with 0 decimals of accuracy
[19:36:10][D][sensor:093]: 'SmartSolar_75_15 WiFi Signal Sensor': Sending state -80.00000 dBm with 0 decimals of accuracy
[19:36:12][D][text_sensor:064]: 'SmartSolar_75_15 Uptime (formatted)': Sending state '0d 00h 00m 28s'
[19:36:12][D][sensor:093]: 'SmartSolar_75_15 Uptime': Sending state 33.37800 s with 0 decimals of accuracy```

![20240819_192250](https://github.com/user-attachments/assets/ca3cd656-0930-4f40-807c-2b09e580189c)
![20240819_192332](https://github.com/user-attachments/assets/d9d126b3-1224-48e0-84b6-30448e1f8770)

I also tested the connection from the controller to the end of the wires and it seems connected on D7 and GND all the way through

syssi, 2024-08-19:
Did you see this image:

[image]

It looks like you have to use the other two pins.

KinDR007, 2024-08-19:
Yep ,wrong selected board its a not nodemcu !! Corect bord name its wemos d1

Tsumiki-Chan, 2024-08-19:
> Yep ,wrong selected board its a not nodemcu !! Corect board name its wemos d1

Changed the model to d1_mini (though it was listed as NodeMCU) but still no data. I'm kinda suspecting it's the board itself.

Tsumiki-Chan, 2024-08-20:
It seems like that was a miss. I got another Board, flashed the exact same firmware flashed and tried probing every single pin on the board. All i get are NaN-Values. What really sets me up is that I don't seem to get any debug log from Victron (aka I don't see any hex coming from it.

KinDR007, 2024-08-20:
What version of mppt do you have ?with bluetooth or without? If with bluetooth, check the settings to see if you have normal communication on TX pin[image]

Tsumiki-Chan, 2024-08-20:
> What version of mppt do you have ?with bluetooth or without? If with bluetooth, check the settings to see if you have normal communication on TX [...]

It's a Smart MPPT with BT. TX-Pin is set to Normal communication

Tsumiki-Chan, 2024-08-20:
Oh hell no... after scratching my head for another 2hours I went outside and restarted everything, including the SmartSolar. And out of the blue it works... Just the battery readings are off but that's because i went for the default config. However all the other readings seem to be correct

closed 2024-08-20 by Tsumiki-Chan