KinDR007/VictronMPPT-ESPHOME #178: No value following update 2024.8.3

titis14, 2024-09-10:
Hello,
I updated my ESP to version 2024.8.3 yesterday and since then the values ​​no longer go up and are not seen as "nan".
The ESP works well because it sends me the diagnostic information but also the temperature of the sensor that I added. The Victron module works well too.
When I go to the web page I see this error in red, could this be the cause?
Thanks
[image]
[image]

syssi, 2024-09-11:
Could you remove the web server component from your YAML to free some resources and try again? Does it work now?

titis14, 2024-09-11:
Hello, I removed the following code but still the same
web_server:
port: 80

syssi, 2024-09-14:
Could you try to reduce your YAML step by step to get an idea what's consumping memory and is it still working in general?

titis14, 2024-09-14:
I removed things but still the problem, yet the flash and ram are far from being filled.
What is strange is the fact that everything that is not Victron works. However it is indeed the update that did this because the graphics all stopped at the time of the update.
[image]

`substitutions:
  external_components_source: github://KinDR007/VictronMPPT-ESPHOME@main
  lower_devicename: "photovoltaique_cabane"
  devicename: "photovoltaique_cabane"
  config_version: "v2021.05.30"
  wifi_fast_connect: "false"
  accuracy: "2"

esphome:
  name: "${lower_devicename}"
  platform: ESP8266
  board: d1_mini
  # board: nodemcuv2

external_components:
  - source: ${external_components_source}
    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: none
  fast_connect: $wifi_fast_connect
  manual_ip:
    static_ip: 192.168.1.37
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 1.1.1.1 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${devicename} Hotspot"
    password: !secret wifi_password

captive_portal:
logger:
  baud_rate: 0
  # level: VERY_VERBOSE
  esp8266_store_log_strings_in_flash: false

# Enable Home Assistant API
api:

ota:
  - platform: esphome

time:
  - platform: homeassistant

uart:
  id: uart_bus
  tx_pin: D8  # Not connected! The communication is read-only
  rx_pin: D7  # Connect this this GPIO and GND to the MPPT charger
  baud_rate: 19200
  stop_bits: 1
  data_bits: 8
  parity: NONE
  rx_buffer_size: 256

victron:
  id: victron0
  uart_id: uart_bus
  throttle: 10s

sensor:
  - platform: victron
    victron_id: victron0
    panel_voltage:
      name: "Panel voltage"
      id: pv
    battery_voltage:
      name: "Battery voltage"
      id: bv
    battery_current:
      name: "Battery current"
      id: bc
    load_current:
      name: "Load current"
      id: load_current
    max_power_today:
      name: "Max power today"
      id: max_power_today
      accuracy_decimals: ${accuracy}

    yield_today:
      name: "Yield today"
      id: yt
      filters:
        # Multiplication factor from W to kW is 0.001
        - multiply: 0.001
      unit_of_measurement: kWh
      accuracy_decimals: ${accuracy}
    panel_power:
      name: "Panel power"
      id: panel_power
    day_number:
      name: "Day sequence number"
      id: day_number
      icon: "mdi:calendar-today"
    charging_mode_id:
      name: "Charger mode id"
    error_code:
      name: "Error code"
    tracking_mode_id:
      name: "Tracker mode id"

  - platform: wifi_signal
    name: "${devicename} WiFi Signal Sensor"
    id: rssi_sensor
    update_interval: 15s

  - platform: template
    name: "Wifi Quality"
    unit_of_measurement: "%"
    accuracy_decimals: 0
    icon: "mdi:wifi"
    update_interval: 15s
    lambda: |-
        int quality;
        const int rssi = id(rssi_sensor).state;
        if (rssi <= -100) {
          quality = 0;
        } else if (rssi >= -50) {
          quality = 100;
        } else {
          quality = 2 * (rssi + 100);
        }
        return quality;

  - platform: uptime
    name: "${devicename} Uptime"
    id: uptime_s
    update_interval: 15s

  - platform: template  # for 1 cell of battery
    name: "Bat S1 "     # 7S pack
    lambda: |-
      if ((((id(bv).state) / 7) > 4.22) && (((id(bv).state) / 7) < 3.2)) {
        return NAN;
      }
      return ((id(bv).state) / 7);

    update_interval: 10s
    icon: "mdi:battery"
    unit_of_measurement: V
    device_class: "voltage"
    accuracy_decimals: ${accuracy}
  - platform: template           # Battery working percent
    name: "Bat working percent"  # cycle batt between 3,7v - 4.17v
    lambda: |-
      if (((((((((id(bv).state) / 7 )) - 3.700) / (4.170 - 3.700)) * 100)) >= 115) && ((((((((id(bv).state) / 7 )) - 3.700) / (4.170 - 3.700)) * 100)) <= -25)) {
        return NAN;
      }
      return ((((((id(bv).state) / 7 )) - 3.700) / (4.170 - 3.700)  ) * 100);

    update_interval: 15s
    icon: "mdi:battery"
    unit_of_measurement: "%"
    device_class: "voltage"
    accuracy_decimals: ${accuracy}
    
  - platform: dht
    pin: 
      number: GPIO5
    model: DHT11  
    temperature:
      name: "Temperature Cabane"
      filters: 
        - offset: -5.0
    update_interval: 60s

text_sensor:
  - platform: victron
    victron_id: victron0
    charging_mode:
      name: "Charging Mode"
    error:
      name: "Error"
    tracking_mode:
      name: "Tracking Mode"
    firmware_version:
      name: "Firmware Version"
    device_type:
      name: "Device Type"

  - platform: template
    name: "${devicename} Config Version"
    icon: mdi:information-outline
    lambda: |-
      return {"${config_version}"};
  - platform: version
    name: "${devicename} Esphome Version"
    icon: mdi:information-outline
  - platform: wifi_info
    ip_address:
      name: "${devicename} IP Address"
      icon: mdi:ip
    ssid:
      name: "${devicename} Connected SSID"
      icon: mdi:wifi
    bssid:
      name: "${devicename} Connected BSSID"
    mac_address:
      name: "${devicename} Mac Wifi Address"
  - platform: template
    name: "${devicename} Uptime (formatted)"
    lambda: |-
      uint32_t dur = id(uptime_s).state;
      int dys = 0;
      int hrs = 0;
      int mnts = 0;
      if (dur > 86399) {
        dys = trunc(dur / 86400);
        dur = dur - (dys * 86400);
      }
      if (dur > 3599) {
        hrs = trunc(dur / 3600);
        dur = dur - (hrs * 3600);
      }
      if (dur > 59) {
        mnts = trunc(dur / 60);
        dur = dur - (mnts * 60);
      }
      char buffer[17];
      sprintf(buffer, "%ud %02uh %02um %02us", dys, hrs, mnts, dur);
      return {buffer};
    icon: mdi:clock-start
    update_interval: 5s

switch:
  - platform: restart
    icon: mdi:reload-alert
    name: "${devicename} Restart"`

titis14, 2024-09-15:
If the problem comes from the lack of memory do you think I could redo it with an ESP32-S2? Currently is an ESP8266.

titis14, 2024-10-30:
Hi, After a while my Victron stopped working completely. After an electric restart everything came back. This was a coincidence with the ESP update.

closed 2024-10-30 by titis14