KinDR007/VictronMPPT-ESPHOME #180: Only selected values appearing

Vunaflow, 2024-09-22:
Hello,
I am trying to set up an ESP8266 to monitor my Victron inverter and MPPT. Thanks for this component, this is going to be very helpful !
Right now, I can only see the battery voltage, product information and working mode for the inverter (similar challenges with the MPPT device). The Victron dongle works fine.
How can I get the ESP8266 to correctly pick up information about the current/load and other parameters?

See screenshot and relevant yaml code
[image]

Yaml:
```
substitutions:
  name: esphome-web-6d688a
  friendly_name: Solar ESP2

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  min_version: 2024.6.0
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: dev

esp8266:
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
- platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp8266.yaml@main
  import_full_config: true

# To have a "next url" for improv serial
web_server:

external_components:
  - source: github://KinDR007/VictronMPPT-ESPHOME@main

uart:
  - id: uart_0
    tx_pin: GPIO0  # Not used! The communication is read-only
    rx_pin: GPIO13
    baud_rate: 19200
    rx_buffer_size: 256
  - id: uart_1
    tx_pin: GPIO12  # Not used! The communication is read-only
    rx_pin: GPIO4
    baud_rate: 19200
    rx_buffer_size: 256

victron:
  - id: victron_inverter_12_500
    uart_id: uart_0
  - id: victron_MPPT_100_30
    uart_id: uart_1

sensor:
  - platform: victron
    victron_id: victron_inverter_12_500
    max_power_yesterday:
      name: "Inverter max power yesterday"
    max_power_today:
      name: "Inverter max power today"
    yield_total:
      name: "Inverter yield total"
    yield_yesterday:
      name: "Inverter yield yesterday"
    yield_today:
      name: "Inverter yield today"
    panel_voltage:
      name: "Inverter panel voltage"
    panel_power:
      name: "Inverter panel power"
    battery_current:
      name: "Inverter panel current"
    battery_voltage:
      name: "Inverter battery voltage"
    day_number:
      name: "Inverter day number"
    charging_mode_id:
      name: "Inverter charging mode id"
    error_code:
      name: "Inverter error code"
    tracking_mode_id:
      name: "Inverter tracking mode id"
    load_current:
      name: "Inverter load current"
    
  - platform: victron
    victron_id: victron_MPPT_100_30
    max_power_yesterday:
      name: "MPPT max power yesterday"
    max_power_today:
      name: "MPPT max power today"
    yield_total:
      name: "MPPT yield total"
    yield_yesterday:
      name: "MPPT yield yesterday"
    yield_today:
      name: "MPPT yield today"
    panel_voltage:
      name: "MPPT panel voltage"
    panel_power:
      name: "MPPT panel power"
    battery_current:
      name: "MPPT panel current"
    battery_voltage:
      name: "MPPT battery voltage"
    day_number:
      name: "MPPT day number"
    charging_mode_id:
      name: "MPPT charging mode id"
    error_code:
      name: "$MPPT error code"
    tracking_mode_id:
      name: "MPPT tracking mode id"
    load_current:
      name: "MPPT load current"

text_sensor:
  - platform: victron
    victron_id: victron_MPPT_100_30
    charging_mode:
      name: "MPPT charging mode"
    error:
      name: "MPPT error"
    tracking_mode:
      name: "MPPT tracking mode"
    firmware_version:
      name: "MPPT firmware version"
    device_type:
      name: "MPPT device type"
    serial_number:
      name: "MPPT serial number"
  - platform: victron
    victron_id: victron_inverter_12_500
    charging_mode:
      name: "Inverter charging mode"
    error:
      name: "Inverter error"
    tracking_mode:
      name: "Inverter tracking mode"
    firmware_version:
      name: "Inverter firmware version"
    device_type:
      name: "Inverter device type"
    serial_number:
      name: "Inverter serial number"

binary_sensor:
  - platform: victron
    victron_id: victron_MPPT_100_30
    load_state:
      name: "MPPT load state"
    relay_state:
      name: "MPPT relay state"
  - platform: victron
    victron_id: victron_inverter_12_500
    load_state:
      name: "Inverter load state"
    relay_state:
      name: "Inverter relay state"
```

Vunaflow, 2024-09-22:
[logs_esphome-web-6d688a_logs.txt](https://github.com/user-attachments/files/17088608/logs_esphome-web-6d688a_logs.txt)
Here is the log. It looks like the AC_OUT_I is being communicated as zero, so this is interpreted correctly by the ESP.

Vunaflow, 2024-09-22:
Thanks for the prompt !
I was able to solve all the issues for both the MPPT and inverter by rewriting the YAML starting from the examples. The one issues that is remaining is the output current/load. 
Interestingly, the output current/load is correctly picked up when I set the automatic transfer switch the inverter is connected to to the solar set up manually. But if the transfer switch is in automatic mode, it cannot pick up the output current/load from the inverter at all, even if the grid is down and the inverter is working. The Victron dongle however is showing the output load correctly. I have no idea how the presence of the transfer switch is affecting the serial output of the inverter...

syssi, 2024-09-22:
Is the serial output different in these cases?

closed 2024-11-11 by syssi