KinDR007/VictronMPPT-ESPHOME #182: BlueSolar MPPT 75/15 missing from device list

gaz99gh, 2024-11-10:
I have a BlueSolar MPPT 75/15 connected to a DIY ESPHome board and the "Device Type" is showing as "Unknown"

I have checked the debug logs and the PID for this model is 0xA07D.
`[18:33:41][D][uart_debug:158]: <<< "PID\t0xA07D\r\n"`

This must be a new revision, so can it be added to the device list please.

syssi, 2024-11-10:
Do you like to contribute the improvement? Your pull request will be very similar to this one: https://github.com/KinDR007/VictronMPPT-ESPHOME/pull/181/files

closed 2024-11-11 by syssi

gaz99gh, 2024-11-11:
That was quick, I had just arranged with a friend to show me how to do a PR but maybe next time.

BTW, the VE.Direct-Protocol-3.33 document shows the device name for this PID as BlueSolar MPPT 75/15 rev3.